fatihak/InkyPi #4: Inky Frame support

GreenMan36, 2025-01-02:
You have official support for the Inky Impression, however these are all sold out for me.

I'd like to get an Inky Frame instead, but you don't seem to have official support for it. I'm sure it'll work fine except for the buttons.

It would be neat to get support for those buttons to for example, switch between plugins, or for a plugin to have basic inputs.

- E-Ink Display: Pimoroni Inky Frame, available in 3 sizes: 
  -  [4 Inch Display](https://shop.pimoroni.com/products/inky-frame-4)
  -   [5.7 Inch Display](https://shop.pimoroni.com/products/inky-frame-5-7)
  -   [7.3 Inch Display](https://shop.pimoroni.com/products/inky-frame-7-3)

GreenMan36, 2025-01-02:
Perhaps Inky wHAT would also be neat to support in the future. Currently I can only commit to one piece of hardware.

fatihak, 2025-01-02:
@GreenMan36 The Inky wHAT and Inky Frame displays _should_ work with this project since I'm using the inky python library to connect to and update the display. I haven’t added them to the list of supported devices because I haven’t personally tested the project with those displays.

If you decide to purchase one of those devices, I’ll do my best to provide support and address any issues that arise. Once it’s confirmed to work smoothly, I can add it to the list of supported devices. We can keep this issue open if you decide to purchase one to continue the discussion once you're able to test it.

Additionally, I’m planning to add support for the buttons as well, including customizable action bindings from the UI. I'll add that to the roadmap in the README since others have suggested it as well.

fatihak, 2025-01-07:
I just realized that the Inky Frame displays use a Pico 2 W, which means they rely on MicroPython or CircuitPython. Unfortunately, this would require significant changes to the project, so I won’t be able to support that display at this time. I’m really sorry for the inconvenience if you’ve already purchased the display. If you ordered it specifically for this project, I recommend returning or canceling your order if possible.

bradsour, 2025-03-19:
> Hey i wanted to add this could still be possible .....sort of... can you still produce a known URL of an image. It would be pretty easy for your code to be a web interface on another pi or docker or whatever. But then have the inky frame (aka pico pi models) go to an url to get the image to render?

That would be really cool to run docker container and have it serve up different images and be able to point different inky's to those specific images that can be grabbed over local network. Could have a master inky running the container (or any other device) and then have various displays around the house.

Like a Now playing dashboard (from a headless plexamp address?).

Thinking more it could be cool to be able to tile out different webpages into a broader webpage. So could combine the suggestion above of a media player along with displaying the current weather and other informative tiles like a shopping/task list.

sdh2, 2025-03-23:
I have an Inky wHAT and would be happy to test.

fatihak, 2025-06-23:
Based on feedback Inky wHAT displays already work out of the box with this project, and there are currently no plans to support Inky Frame due to the hardware limitations. Closing this issue.

closed 2025-06-23 by fatihak