fatihak/InkyPi #5: installation error?

ejw71, 2025-01-02:
I am getting this issue when the installer attempts to install the python virtual environment:

Creating python virtual environment. 
	Installing python dependencies.  [|]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/requests/requests-2.32.3-py3-none-any.whl
	Installing python dependencies.  [/]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/urllib3/urllib3-2.3.0-py3-none-any.whl
	Installing python dependencies.  [-] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/certifi/
	Installing python dependencies.  [-] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/httpcore/

Not sure if it's a simple mistake I made as I am reasonably new to coding on the pi.  But also to note that I'm trying to install this on an older Pi zero W (not the pi zero 2 W).

the inkypi service then fails to start and cannot be started manually

Any suggestions?

Jason13L, 2025-01-02:
I was getting the same errors on my zero w (1.1). I later switched to a 3B+ and didn't receive the errors.

fatihak, 2025-01-03:
Hm this might be a network issue, can you check if your pi is connected to wifi? You can try running:
`ping -c 4 google.com`

If thats not the case it might be caused by the swap size on the Pi and you can try increasing it:
`sudo dphys-swapfile swapoff
sudo nano /etc/dphys-swapfile`
In the file, set the CONF_SWAPSIZE value to 1024, then run:
`sudo dphys-swapfile setup
sudo dphys-swapfile swapon`
And verify the swap size by running `free -h`

Jason13L, 2025-01-03:
Confirmed that pi had WiFi, this was after a sudo apt update. I will try increasing the swap file size and report back.

Jason13L, 2025-01-03:
Still getting the error. Upon completing installation in ran a ping and that worked correctly. Here is a copy/pasta. 

Creating python virtual environment.
        Installing python dependencies.  [|]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/flask/flask-3.1.0-py3-none-any.whl
        Installing python dependencies.  [|]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/python-dotenv/python_dotenv-1.0.1-py3-none-any.whl
        Installing python dependencies.  [/] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/inky/
        Installing python dependencies.  [-]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/inky/inky-2.0.0-py3-none-any.whl
        Installing python dependencies.  [-] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/requests/
        Installing python dependencies.  [\]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/requests/requests-2.32.3-py3-none-any.whl
        Installing python dependencies.  [|]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/werkzeug/werkzeug-3.1.3-py3-none-any.whl
        Installing python dependencies.  [|]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pytz/pytz-2024.2-py3-none-any.whl
        Installing python dependencies.  [/]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/numpy/numpy-2.2.1-cp311-cp311-linux_armv6l.whl
        Installing python dependencies.  [/]   WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLZeroReturnError(6, 'TLS/SSL connection has been closed (EOF) (_ssl.c:992)'))': /simple/numpy/numpy-2.2.1-cp311-cp311-linux_armv6l.whl
        Installing python dependencies.  [\] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/certifi/
        Installing python dependencies.  [-] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))': /simple/pydantic/
        Installing python dependencies.  [/] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/h11/
        Installing python dependencies.  [✔]
Adding executable to /usr/local/bin/inkypi
Copying config files to /home/inky/InkyPi/install/../src/config
        device.json already exists in /home/inky/InkyPi/install/../src/config [✔]
        plugins.json already exists in /home/inky/InkyPi/install/../src/config [✔]
Installing inkypi systemd service.
INKYPI Installation Complete! [✔]
[•] A reboot of your Raspberry Pi is required for the changes to take effect
[•] After your Pi is rebooted, you can access the web UI by going to 'inky.local' or '192.168.68.96' in your browser.
[•] If you encounter any issues or have suggestions, please submit them here: https://github.com/fatihak/InkyPi/issues
Would you like to restart your Raspberry Pi now? [Y/N] n
Please restart your Raspberry Pi later to apply changes by running 'sudo reboot now'.
inky@inky:~/InkyPi $ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=251 time=41.5 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=251 time=413 ms

fatihak, 2025-01-03:
Can you try upgrading pip and manually installing a package using the -v flag and provide the output:

`python3 -m venv myenv`
`source myenv/bin/activate`
`pip install --upgrade pip`
`pip install -v requests`

Also check the python and pip versions afterwords:
`python --version`
`pip --version`

Jason13L, 2025-01-03:
That seems to have worked manually: 

(myenv) inky@inky:~ $ python --version
Python 3.11.2
(myenv) inky@inky:~ $ pip --version
pip 24.3.1 from /home/inky/myenv/lib/python3.11/site-packages/pip (python 3.11)

What should I do next?

fatihak, 2025-01-03:
Hm can you try the installation again? Maybe it was the pip version

Jason13L, 2025-01-03:
Same issue as above. My hunch is that the slow processor is just timing out when trying to do the install. I don't have these issues on any of the higher-specced versions. Is there an easy method of downloading the dependencies manually versus script?

fatihak, 2025-01-03:
Yeah you can try manually installing the python packages in the virtual env that InkyPi uses:

`source "/usr/local/inkypi/venv_inkypi/bin/activate"`
`pip install -r install/requirements.txt`

And run `deactivate`

Since the rest of the installation completed, I think you should be able to reboot your pi to start the service?
`sudo reboot now` 
And check the status of the InkyPi service:
`sudo systemctl status inkypi.service`

In the meantime, I'm going to remove Pi Zero W from the Pi options in the README unless we can find a better solution going forward.

FrankX0, 2025-01-03:
Same problem here with a Pi Zero W.
Performing the manual install does not seem to do anything: all report "Requirement already satisfied".
Checking the status:
```
Jan 03 12:57:34 epaper inkypi[526]:     ^^^^^^^^^^^^^^^^^^
Jan 03 12:57:34 epaper inkypi[526]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/inky.py", li>
Jan 03 12:57:34 epaper inkypi[526]:     import numpy
Jan 03 12:57:34 epaper inkypi[526]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/numpy/__init__.py>
Jan 03 12:57:34 epaper inkypi[526]:     raise ImportError(msg) from e
Jan 03 12:57:34 epaper inkypi[526]: ImportError: Error importing numpy: you should not try to import numpy from
Jan 03 12:57:34 epaper inkypi[526]:         its source directory; please exit the numpy source tree, and relaunch
Jan 03 12:57:34 epaper inkypi[526]:         your python interpreter from there.
Jan 03 12:57:35 epaper systemd[1]: inkypi.service: Deactivated successfully.
Jan 03 12:57:35 epaper systemd[1]: inkypi.service: Consumed 5.200s CPU time.
```
I had a very similar issue with another project on the Pi Zero W recently: also was not able to get that to work.
Maybe it is related with the OS (Debian Bookworm).

I think I'm going to invest in a Pi Zero 2 W :-)

FrankX0, 2025-01-03:
I just tried on Debian Bullseye.
It feels faster and no network errors than with Bookworm.
But then I run into other issues:

- Install script cannot run: had to remove the reboot option from the script.
- Installation fails with: `No matching distribution found for numpy==2.2.1`
- Installation of Numpy (2.0.2) fails due to permission denied (in venv)
- Status of service: `ModuleNotFoundError: No module named 'PIL'`

fatihak, 2025-01-03:
Looks like there are similar issues reported in the inky github issues page:
https://github.com/pimoroni/inky/issues/190

And numpy documentation has these recommendations for installation on pi:
https://numpy.org/devdocs/user/troubleshooting-importerror.html#raspberry-pi

Based on this, can someone try these steps:

1. Uninstall InkyPi:
`sudo bash install/uninstall.sh`
2. Add a new line to install/debian-requirements.txt with `libatlas-base-dev`
3. Rerun installation:
`sudo bash install/install.sh`
4. Install pip packages manually if it fails:
`source "/usr/local/inkypi/venv_inkypi/bin/activate"`
`pip install -r install/requirements.txt`
`deactivate`
5. Restart the service check the status/logs:
`sudo systemctl restart inkypi.service`
`sudo systemctl status inkypi.service`

ejw71, 2025-01-03:
> pimoroni/inky#190 <https://github.com/pimoroni/inky/issues/190>

Followed all the steps and it looks like it still has the same errors:
Creating python virtual environment.
        Installing python dependencies.  [\] WARNING: Retrying
(Retry(total=4, connect=None, read=None, redirect=None, status=None)) after
connection broken by 'ProtocolError('Connection aborted.',
RemoteDisconnected('Remote end closed connection without response'))':
/simple/requests/
        Installing python dependencies.  [/]   WARNING: Retrying
(Retry(total=4, connect=None, read=None, redirect=None, status=None)) after
connection broken by 'ProtocolError('Connection aborted.',
RemoteDisconnected('Remote end closed connection without response'))':
/simple/requests/requests-2.32.3-py3-none-any.whl
        Installing python dependencies.  [\] WARNING: Retrying
(Retry(total=4, connect=None, read=None, redirect=None, status=None)) after
connection broken by 'ProtocolError('Connection aborted.',
RemoteDisconnected('Remote end closed connection without response'))':
/simple/httpcore/
        Installing python dependencies.  [✔]

And the status report indicates that the service still didn't start as far
as I can see.

○ inkypi.service - InkyPi App
     Loaded: loaded (/etc/systemd/system/inkypi.service; enabled; preset:
enabled)
     Active: inactive (dead) since Fri 2025-01-03 10:31:21 PST; 1s ago
   Duration: 5.705s
    Process: 1560 ExecStart=/usr/local/bin/inkypi -d (code=exited,
status=0/SUCCESS)
   Main PID: 1560 (code=exited, status=0/SUCCESS)
        CPU: 4.896s

Jan 03 10:31:21 inkypi inkypi[1562]:     ^^^^^^^^^^^^^^^^^^
Jan 03 10:31:21 inkypi inkypi[1562]:   File
"/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/inky.py", >
Jan 03 10:31:21 inkypi inkypi[1562]:     import numpy
Jan 03 10:31:21 inkypi inkypi[1562]:   File
"/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/numpy/__init__.>
Jan 03 10:31:21 inkypi inkypi[1562]:     raise ImportError(msg) from e
Jan 03 10:31:21 inkypi inkypi[1562]: ImportError: Error importing numpy:
you should not try to import numpy from
Jan 03 10:31:21 inkypi inkypi[1562]:         its source directory; please
exit the numpy source tree, and relaunch
Jan 03 10:31:21 inkypi inkypi[1562]:         your python interpreter from
there.
Jan 03 10:31:21 inkypi systemd[1]: inkypi.service: Deactivated successfully.
Jan 03 10:31:21 inkypi systemd[1]: inkypi.service: Consumed 4.896s CPU time.

Jason13L, 2025-01-03:
I get the same issue: 

Jan 03 11:41:27 inkySmall inkypi[630]:     ^^^^^^^^^^^^^^^^^^
Jan 03 11:41:27 inkySmall inkypi[630]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/inky.py">
Jan 03 11:41:27 inkySmall inkypi[630]:     import numpy
Jan 03 11:41:27 inkySmall inkypi[630]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/numpy/__init_>
Jan 03 11:41:27 inkySmall inkypi[630]:     raise ImportError(msg) from e
Jan 03 11:41:27 inkySmall inkypi[630]: ImportError: Error importing numpy: you should not try to import numpy from
Jan 03 11:41:27 inkySmall inkypi[630]:         its source directory; please exit the numpy source tree, and relaunch
Jan 03 11:41:27 inkySmall inkypi[630]:         your python interpreter from there.
Jan 03 11:41:27 inkySmall systemd[1]: inkypi.service: Deactivated successfully.
Jan 03 11:41:27 inkySmall systemd[1]: inkypi.service: Consumed 4.779s CPU time.

FrankX0, 2025-01-03:
**I have it running!**
The above error was solved by re-installing Pillow.
But I could not because of a permission issue.
So I did the following:

- `sudo su`
- `source "/usr/local/inkypi/venv_inkypi/bin/activate"`
- `pip uninstall Pillow`
- `pip install Pillow`
- `sudo systemctl restart inkypi.service`

I now have the e-paper showing inkypi and the webserver is running.
Hope this helps.

Jason13L, 2025-01-03:
I am still getting an error on mine but numpy seems to be behaving. Looks like an ImportError. 

Jan 03 13:16:57 inkySmall inkypi[769]: Traceback (most recent call last):
Jan 03 13:16:57 inkySmall inkypi[769]:   File "/home/inky/InkyPi/src/inkypi.py", line 18, in <module>
Jan 03 13:16:57 inkySmall inkypi[769]:     from utils.app_utils import generate_startup_image
Jan 03 13:16:57 inkySmall inkypi[769]:   File "/home/inky/InkyPi/src/utils/app_utils.py", line 6, in <module>
Jan 03 13:16:57 inkySmall inkypi[769]:     from PIL import Image, ImageDraw, ImageFont
Jan 03 13:16:57 inkySmall inkypi[769]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/Image.py">
Jan 03 13:16:57 inkySmall inkypi[769]:     from . import _imaging as core
Jan 03 13:16:57 inkySmall inkypi[769]: ImportError: libopenjp2.so.7: cannot open shared object file: No such file or di>
Jan 03 13:16:57 inkySmall systemd[1]: inkypi.service: Deactivated successfully.
Jan 03 13:16:57 inkySmall systemd[1]: inkypi.service: Consumed 1.512s CPU time.

FrankX0, 2025-01-03:
Please try: `sudo apt-get install libopenjp2-7`
Then restart the service.

Jason13L, 2025-01-03:
We have success!

fatihak, 2025-01-03:
So to summarize for anyone finding this later:
Install these debian packages:
`sudo apt-get install libopenblas-dev libopenjp2-7`

If pip install fails during installation, manually install the packages:
`source "/usr/local/inkypi/venv_inkypi/bin/activate"`
`pip install -r install/requirements.txt`
`deactivate`

For PIL ImportError: cannot import name '_imagingft' from 'PIL', uninstall and reinstall Pillow:
`sudo su`
`source "/usr/local/inkypi/venv_inkypi/bin/activate"`
`pip uninstall Pillow`
`pip install Pillow`
`deactivate`

Restart the service:
`sudo systemctl restart inkypi.service`

@FrankX0 @Jason13L does this capture all the issues? I can add the additional debian packages to the installation script and the rest to the troubleshooting page.

Jason13L, 2025-01-03:
I believe so. I can try from scratch to confirm. I know we also tried to add a new line to install/debian-requirements.txt with libatlas-base-dev but not sure if that impacted the results.

Jason13L, 2025-01-03:
Bah, just killed my pi with a stray nut while it was powered on (stupid magnets). Going to take me a bit more time to test, I don't have any more with headers and my other pi is in a case configured as Bjorn.

ejw71, 2025-01-04:
So this is now the full installation flow from a new OS install?  Did I get
the order correct?  I'll try it from scratch.

sudo apt-get install libopenblas-dev libopenjp2-7
git clone https://github.com/fatihak/InkyPi.git
cd InkyPi
sudo bash install/install.sh

If pip install fails during installation, manually install the packages:
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip install -r install/requirements.txt
deactivate

For PIL ImportError: cannot import name '_imagingft' from 'PIL', uninstall
and reinstall Pillow:
sudo su
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip uninstall Pillow
pip install Pillow
deactivate

Restart the service:
sudo systemctl restart inkypi.service

fatihak, 2025-01-04:
I believe thats correct, i don't think you should be seeing the PIL ImportError if you start with a new OS install since we do the apt-get install of the packages that were needed but added that just in case.

Let me know if that works, and I can push the changes. Also thank you all for helping troubleshoot the issue and testing!

ejw71, 2025-01-04:
Absolutely.  This is fun stuff.  Should I do an apt-update / apt-upgrade
before running through this installation flow?

ejw71, 2025-01-04:
Looks like it failed during the normal installation with the same error as
before but when i did this afterwards:

source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip install -r install/requirements.txt
deactivate
sudo systemctl restart inkypi.service
sudo systemctl status inkypi.service
it seems to have successfully started the inkypi.service but eventually
failed due to the PIL ImportError: cannot import name '_imagingft' from
'PIL',

as recommended, I uninstall and reinstall Pillow:
sudo su
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip uninstall Pillow
pip install Pillow
deactivate

and was successful at at last

Now trying to figure out how to get the plugins to work

jbolda, 2025-01-05:
> So to summarize for anyone finding this later: Install these debian packages: `sudo apt-get install libopenblas-dev libopenjp2-7`
>
> If pip install fails during installation, manually install the [...]

Just did a fresh install on a zero W on the latest (like 2024-09 or something), and I had to do all of these steps.

andrew-rosca, 2025-01-10:
I ran into these issues as well on a Pi W. The following steps resolved it for me (basically same what's described above)
```
sudo su
apt install libopenblas0
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip uninstall Pillow
pip install Pillow
exit
sudo systemctl restart inkypi.service
```

fatihak, 2025-01-25:
Thanks for all the feedback everyone, I updated the troubleshooting docs to include a section for the Raspberry Pi Zero W with this commit: https://github.com/fatihak/InkyPi/commit/7a4210e7ffb2cd2b91a1d99bfe4acc75a487ea8f

Closing this out :)

closed 2025-01-25 by fatihak

awesomeruss, 2026-01-01:
> i face below issue while running ppip install -r install/requirements.txt
> 
> Building wheel for pi-heif (pyproject.toml) ... error error: subprocess-exited-with-error
> 
> × Building wheel for pi-heif (pyproject.toml) did not run successfully. │ exit code: 1 ╰─> [87 lines of output] /tmp/pip-build-env-tugwzzxl/overlay/lib/python3.11/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated. !!
> 
> ```
>           ********************************************************************************
>           Please consider removing the following classifiers in favor of a SPDX license expression:
>   
>           License :: OSI Approved :: GNU Lesser General Public License v3 (LGPLv3)
>   
>           See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
>           ********************************************************************************
>   
>   !!
> ```
> 
> anyone can help?

I am also having this issue.
It's the first time I've tried to build this type of environment on a pi - is it normal for the build to take multiple hours?

awesomeruss, 2026-01-02:
More copypasta:

        Creating symlink from /home/admin/epaper/InkyPi/install/../src to /usr/local/inkypi/src [✔]
Creating python virtual environment.
        Installing python dependencies.  [\]   error: subprocess-exited-with-error

  × Building wheel for pi-heif (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [101 lines of output]
      /tmp/pip-build-env-281lppmq/overlay/lib/python3.13/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!

              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:

              License :: OSI Approved :: GNU Lesser General Public License v3 (LGPLv3)

              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************

      !!
        self._finalize_license_expression()
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/options.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/misc.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/heif.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/constants.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/as_plugin.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/_version.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/_lib_info.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/_deffered_error.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/__init__.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      copying pi_heif/HeifImagePlugin.py -> build/lib.linux-armv6l-cpython-313/pi_heif
      running build_ext
      Package libheif was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libheif.pc'
      to the PKG_CONFIG_PATH environment variable
      Package 'libheif', required by 'virtual:world', not found
      Package heif was not found in the pkg-config search path.
      Perhaps you should add the directory containing `heif.pc'
      to the PKG_CONFIG_PATH environment variable
      Package 'heif', required by 'virtual:world', not found

      The headers or library files could not be found for libheif,
      a required dependency when compiling Pillow-Heif from source.

      Please see the install instructions at:
         https://pillow-heif.readthedocs.io/en/latest/installation.html

      Looking for `libheif` using pkg-config.
      Looking for `heif` using pkg-config.
      Checking for include file 'heif.h' in '/usr/local/inkypi/venv_inkypi/include'
      Checking for include file 'heif.h' in '/usr/local/inkypi/venv_inkypi/include/libheif'
      Checking for include file 'heif.h' in '/usr/local/include'
      Checking for include file 'heif.h' in '/usr/local/include/libheif'
      Checking for include file 'heif.h' in '/usr/include'
      Checking for include file 'heif.h' in '/usr/include/libheif'
      Checking for include file 'heif.h' in '/usr/local/inkypi/venv_inkypi/include'
      Checking for include file 'heif.h' in '/usr/local/inkypi/venv_inkypi/include/libheif'
      Checking for include file 'heif.h' in '/usr/include/python3.13'
      Checking for include file 'heif.h' in '/usr/include/python3.13/libheif'
      Traceback (most recent call last):
        File "/usr/local/inkypi/venv_inkypi/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
          main()
          ~~~~^^
        File "/usr/local/inkypi/venv_inkypi/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
          json_out["return_val"] = hook(**hook_input["kwargs"])
[... 88 lines cut ...]
         Compiling once_cell v1.21.3
         Compiling tap v1.0.1
         Compiling cfg-if v1.0.4
         Compiling lexical-util v1.0.7
         Compiling lexical-parse-integer v1.0.6
         Compiling wyz v0.5.1
         Compiling num-bigint v0.4.6
         Compiling pyo3-macros v0.27.1
         Compiling jiter v0.12.0 (/tmp/pip-install-uyl_i8f_/jiter_02ec198a1e79408e99dacb9526bf920c/crates/jiter)
         Compiling funty v2.0.0
         Compiling unindent v0.2.4
         Compiling indoc v2.0.7
         Compiling bitvec v1.0.1
      error: failed to build archive at `/tmp/pip-install-uyl_i8f_/jiter_02ec198a1e79408e99dacb9526bf920c/target/release/deps/libbitvec-7b2f5d19660eaa2e.rlib`: No space left on device (os error 28)

      error: could not compile `bitvec` (lib) due to 1 previous error
      💥 maturin failed
        Caused by: Failed to build a native library through cargo
        Caused by: Cargo build finished with "exit status: 101": `env -u CARGO PYO3_BUILD_EXTENSION_MODULE="1" PYO3_ENVIRONMENT_SIGNATURE="cpython-3.13-64bit" PYO3_PYTHON="/usr/local/inkypi/venv_inkypi/bin/python" PYTHON_SYS_EXECUTABLE="/usr/local/inkypi/venv_inkypi/bin/python" "cargo" "rustc" "--profile" "release" "--features" "pyo3/generate-import-lib" "--message-format" "json-render-diagnostics" "--manifest-path" "/tmp/pip-install-uyl_i8f_/jiter_02ec198a1e79408e99dacb9526bf920c/crates/jiter-python/Cargo.toml" "--lib" "--crate-type" "cdylib"`
      Rust not found, installing into a temporary directory
      Error: command ['maturin', 'pep517', 'build-wheel', '-i', '/usr/local/inkypi/venv_inkypi/bin/python', '--compatibility', 'off'] returned non-zero exit status 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for jiter

phattmatt, 2026-01-02:
Hi @samkuang1120 and @awesomeruss,

Welcome to the project!

This issue has been closed and it's unlikely anybody will be watching for new posts to it.

Regarding the issue installing `pi-heif`, there are a few posts about the same issue:

https://github.com/fatihak/InkyPi/issues?q=is%3Aissue%20state%3Aclosed%20pi-heif

In particular:

https://github.com/fatihak/InkyPi/issues/424#issuecomment-3560494679

If you have any other problems not addressed by previous issues raised, please post a new issue for maximum visibility,