fatihak/InkyPi #9: Error! An error occurred: 'NoneType' object has no attribute 'save'

RAHoebe, 2025-01-04:
Hi, everything installed perfectly. But when I use my openAI project key I get this error:

Error! An error occurred: 'NoneType' object has no attribute 'save'

Best

fatihak, 2025-01-05:
Hi, just pushed a change to improve logging in the AI Image plugin, can you pull the latest changes and check the logs:
`git pull`
`sudo systemctl restart inkypi.service`
`journalctl -u inkypi -f`

Then try the ai image plugin and check for any errors related to the open ai request.

RAHoebe, 2025-01-05:
Hi, the error was on my site:

Jan 05 14:20:49 inkypi inkypi[51996]: 14:20:49 - ERROR -
plugins.ai_image.ai_image - Failed to make Open AI request: Error code: 400
- {'error': {'code': 'billing_hard_limit_reached', 'message': 'Billing hard
limit has been reached', 'param': None, 'type': 'invalid_request_error'}}

best, Ron

fatihak, 2025-01-05:
Yep that seems like it's related to the billing settings in your Open AI account.
Were you able to resolve the issue? Otherwise will be closing this out.

RAHoebe, 2025-01-05:
You close the issue, many thanks!

closed 2025-01-05 by fatihak