fatihak/InkyPi #11: Image plugin improvement suggestion

ejw71, 2025-01-07:
An improvement suggestion for the image plugin:

It would be great to be able to load either a pi local or online directory with a set of images and have it cycle through on a chosen interval similar to what I thought was well done on this project:
https://erikdevries.com/posts/diy-color-e-ink-picture-frame

Possible image sources:
- A directory of images stored locally
- A single image from a URL
- A directory of images from a URL (i.e. a shared google drive / dropbox folder)

Another minor improvement suggestion:  
Images appear to autoscale to the screen width only and not to height (i.e. it crops top and bottom if the image is too "tall"). Would be great to autoscale to either width or height depending on aspect ratio of image and whatever is larger

I'd try coding this myself, but this is very much not my forte so thought I'd throw it out there to the pros :)

Elite230, 2025-01-10:
FTP would also be nice
I can't help with that though

fatihak, 2025-01-11:
These are some good improvements ideas for the image plugin! Others had similar suggestions as well so I will try to get it to eventually but not sure when it'll be available, unless someone else wants to make the changes and create a PR.

anton-ovcharenko, 2025-06-07:
Hi, @fatihak, please take a look at [MR#153](https://github.com/fatihak/InkyPi/pull/153) that adds a new rendering strategy to image_upload plugin in addition to the existing one (cropping): fitting the image to the frame size. 
You are welcome to leave any comments for me to fix. Thank you!