fatihak/InkyPi #13: My Random Image Plugin does not save settings (but does work!)

RAHoebe, 2025-01-10:
Hi,

Thanks again for this nice code!

I made a plugin
[random_image.zip](https://github.com/user-attachments/files/18379144/random_image.zip)

I had to chmod-R 777 ~/Pictures  to get it working. The problem is: it will save the settings. How can I do this?

Best, Ron

P.S. I also noticed that if have f.e. the clock plugin scheduled, and than upload a static image with image upload the scheduler is not stopped. So the image shows for a while and then the clock show up again.

Best, Ron

fatihak, 2025-01-11:
Hi, can you elaborate on what you mean by settings not being saved? Which settings aren't being saved in the plugin code you provided?
In terms of the scheduled plugin, that's the intention behind the scheduling feature. You can do an immediate update to the display but it will return to the scheduled plugin after a while.

RAHoebe, 2025-01-11:
Hi, the folder and random/sequential are not saved when changed.

fatihak, 2025-01-21:
Hm the plugins are initialized when the service first starts so the plugin_config wouldn't have image_directory or display_mode in the __init__ function. You would need to move all of that logic in the generate_image function.

closed 2025-02-14 by fatihak