fatihak/InkyPi #14: Installation went good but cannot get access to intranet page and display does not show anything

nullfreizeit, 2025-01-11:
Hello,

Thank you for this project. I have installed everything on an raspberry pi zero with an 7,3 inch epaper color display.
However I cannot get access to bilderrahmen.local. Also the display does not show anything.
What can I do? Thank you.

[image]

fatihak, 2025-01-11:
Can you check the status of the service and the logs, you can find instructions here: [troubleshooting.md](https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md)

fatihak, 2025-01-11:
Just realized you mentioned Pi Zero, please check out the discussion in this issue: https://github.com/fatihak/InkyPi/issues/5

closed 2025-01-11 by fatihak

nullfreizeit, 2025-01-11:
Sorry, its a Raspberry Pi Zero 2 WH