fatihak/InkyPi #15: No EEPROM detected! You must manually initialise your Inky board

nullfreizeit, 2025-01-11:
I have installed it without any problems, but cannot get it working.
sudo systemctl status inkypi.service give me this message...

`○ inkypi.service - InkyPi App
     Loaded: loaded (/etc/systemd/system/inkypi.service; enabled; preset: enabled)
     Active: inactive (dead) since Sat 2025-01-11 01:37:52 CET; 8h ago
   Duration: 7.858s
    Process: 503 ExecStart=/usr/local/bin/inkypi -d (code=exited, status=0/SUCCESS)
   Main PID: 503 (code=exited, status=0/SUCCESS)
        CPU: 5.871s

Jan 11 01:37:52 bilderrahmen inkypi[527]:     display_manager = DisplayManager(device_config)
Jan 11 01:37:52 bilderrahmen inkypi[527]:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 11 01:37:52 bilderrahmen inkypi[527]:   File "/home/BilderAdm1n/InkyPi/src/display_manager.py", line 16, in __init__
Jan 11 01:37:52 bilderrahmen inkypi[527]:     self.inky_display = auto()
Jan 11 01:37:52 bilderrahmen inkypi[527]:                         ^^^^^^
Jan 11 01:37:52 bilderrahmen inkypi[527]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/auto.py", line 83, in auto
Jan 11 01:37:52 bilderrahmen inkypi[527]:     raise RuntimeError("No EEPROM detected! You must manually initialise your Inky board.")
Jan 11 01:37:52 bilderrahmen inkypi[527]: RuntimeError: No EEPROM detected! You must manually initialise your Inky board.
Jan 11 01:37:52 bilderrahmen systemd[1]: inkypi.service: Deactivated successfully.
Jan 11 01:37:52 bilderrahmen systemd[1]: inkypi.service: Consumed 5.871s CPU time.`

fatihak, 2025-01-11:
Unfortunately the waveshare displays aren't currently supported. I'm using the inky python library to connect to and update the display which doesn't work with the waveshare displays.

nullfreizeit, 2025-01-11:
Thank you anyway. I thought this is the problem.
I have managed to get the display working as per this instruction:
[https://www.waveshare.com/wiki/7.3inch_e-Paper_HAT_(F)_Manual#Working_With_Raspberry_Pi](url)

Would it be difficult to implement it?
I don't think you have planned to work on it, or?

fatihak, 2025-01-25:
I want to implement it in the future but not sure when I'll be able to get to it. I also need to get my hands on a waveshare display first.

regevt, 2025-01-27:
I managed to make it work with my 5.65 7 color display

In the display manager replace the 

```
        self.inky_display = auto()
        self.inky_display.set_border(self.inky_display.BLACK)
```

with

```
        self.epd = epd5in65f.EPD()
        self.epd.init()
```
(or you display driver from waveshare epd folder).

and all instances of

```
self.inky_display.show()
```

with

```
self.epd.display(self.epd.getbuffer(image))
```

fatihak, 2025-01-27:
@regevt Thank you for this! I figured it would be simple but I dont have a waveshare display so I haven't been able to test it out. Not sure how to handle all the different display drivers to make it easier during the install but will at least add documentation for this!

albl2008, 2025-03-17:
Im trying to do it for my wavashare display, but i can not, gpiozero error but i dont know why, ive have alreday installed that library in the enviroment, i done the replacements but stucks on starting web server, when try to upload the first image InkyPI @regevt any idea how to solve this?

regevt, 2025-03-17:
@albl2008 if you can share your code,I will try to run it on my display and see if I can find the issue.

albl2008, 2025-03-18:
@regevt thanks, yesterday i made it, just after post the comment, the problem was the spi was occupied because i have multiple rows duplicates in the /boot/firmware/config.txt, i think it was because the multiple times that i run the command for install inkypi

and the problem around the libraries that waveshare use the solution was activate the venv for the inkypi project and install with pip the libraries gpiozero, lgpio and RPi.GPIO

today i will test the weather api and ai api. I made a project very similar by my own but this is way more better. So thanks @fatihak

pflavio, 2025-03-31:
> Of course, you will need to copy the epd to the project directory.

Could you elaborate on that last part please? :)

regevt, 2025-04-01:
Also check that you use python env like in this tutorial https://python.land/virtual-environments/virtualenv 
The lates versions of python block installing globally by default and recommend using python environment.

I would also recommend before running the project in a service try running it manually and see if the display works.
It is very difficult debugging issues when running from a service.

pflavio, 2025-04-01:
> follow the guide in the Waveshare documentation, checking for SPI availability (https://www.waveshare.com/wiki/7.3inch_e-Paper_HAT_(E)_Manual#Overview in the Working With Raspberry Pi section). I also recommend reviewing the code to make sure the pins mentioned in the documentation are the same ones the code actually uses. 

I did with ```nano /boot/firmware/config.txt``` and it returns ```dtparam=spi=on```. Also ```ls /dev/spi*``` returns ```/dev/spidev0.0```. I went through both examples but only the C one actually draws something on the screen, the Python one does not (but also does not throw any errors). I'm currently checking what could be wrong.

> Once you run the example Waveshare provides, try restarting your Raspberry Pi and see if it boots Inky Pi. I also suggest, if you've installed Inky Pi multiple times, that you review the file mentioned in /boot/firmware/config.txt and make sure there are no duplicate lines.

No issues here as far as I can tell.

> Also, keep in mind that in the display.py file, you need epdconfig.py and the file corresponding to your display, which should be: https://github.com/waveshareteam/e-Paper/blob/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd7in3e.py

I just imported epd7in3e via ```from waveshare_epd import epd7in3e```. Was that not sufficient (if it had been found)?

pflavio, 2025-04-01:
> Also check that you use python env like in this tutorial https://python.land/virtual-environments/virtualenv The lates versions of python block installing globally by default and recommend using [...]

So I tried reinstalling everything system-wide (without pip) and running just inkypi.py, but still I couldn't get it to work. I'll keep trying.

albl2008, 2025-04-01:
@pflavio the command 
ls /dev/spi* should return 2 things, not only one

pflavio, 2025-04-01:
> [@pflavio](https://github.com/pflavio) the command ls /dev/spi* should return 2 things, not only one

Oh good catch! It did return two things when I initially set-up the device and ran the examples, but now it's only one. Maybe I'll just start from scratch and check in between where it goes wrong!

acscpt, 2025-04-14:
> > [@pflavio](https://github.com/pflavio) the command ls /dev/spi* should return 2 things, not only one
> 
> Oh good catch! It did return two things when I initially set-up the device and ran the examples, but now it's only one. Maybe I'll just start from scratch and check in between where it goes wrong!

The install script for inky-pi adds `dtoverlay=spi0-0cs` straight after the `dtparam=spi=on` and `dtparam=i2c_arm=on` entries.  This is to enable chip select for the inkypi device (see [Pimori Inky Install Manually](https://github.com/pimoroni/inky#install-stable-library-from-pypi-and-configure-manually).  This has the effect of disabling the other _spi_ line that the Waveshare device needs.   

[image]

You can fix it be removing that line from `/boot/firmware/config.txt`.

All should work then.

acscpt, 2025-04-15:
@Ch3my that's a great summary.

sjnims, 2025-04-16:
Note, the 10.3" screens don't have a universal driver in the GitHub repo, there's a separate repo available [here](https://github.com/waveshareteam/IT8951-ePaper).

acscpt, 2025-04-20:
I have added support for Waveshare EPD devices into my fork of InkyPi.  You can try it out by 
`
git clone ttps://github.com/acscpt/InkyPi-WS.git
git checkout develop
`
and follow the instructions as normal.

@fatihak  - happy for lodge a pull request to contribute back!  Thanks for this and the youtube videos!

saulob, 2025-04-26:
> I have added support for Waveshare EPD devices into my fork of InkyPi. You can try it out by
> 
> * git clone https://github.com/acscpt/InkyPi-WS.git
> * git checkout develop
> 
> and follow the instructions as normal.
> 
> [@fatihak](https://github.com/fatihak) - happy for lodge a pull request to contribute back! Thanks for this and the youtube videos!

Hi @acscpt 

I’m using a Waveshare 7 .3-inch e-Paper HAT (E)  on a Raspberry Pi Zero 2 W. The Waveshare Python demo epd7in3e_test.py works fine on its own

Steps I followed:

1) fresh Raspberry Pi OS Lite
2) git clone https://github.com/acscpt/InkyPi-WS.git 
3) git checkout develop

Then on reboot I got the EEPROM error:
_RuntimeError: No EEPROM detected! You must manually initialise your Inky board._

Is there an extra step needed to point the display manager to epd7in3e or should the develop branch auto-detect this Waveshare panel?

Thanks

[image]

acscpt, 2025-04-26:
Hi @saulob 

Did you follow the install instructions for my fork?  

I can see from the trace that you're instantiating the Inky driver.  

After checking out `develop` you need to run `sudo bash install\install -W epd7in3e` from the project root directory.  The epd7in3e driver is shipped with my fork.

saulob, 2025-04-26:
Hi @acscpt - thanks a lot for the quick reply!

Unfortunately the README in your fork still shows the original installation steps (screenshot attached), so I hadn’t seen the -W option.

To be sure I removed everything and started again:

git clone https://github.com/acscpt/InkyPi-WS.git
cd InkyPi-WS
git checkout develop
sudo bash install/install.sh -W epd7in3e

After reboot the service still stops with
_RuntimeError: No EEPROM detected! You must manually initialise your Inky board_

(full log in the screenshot)

The Waveshare Python demo (epd7in3e_test.py) still works fine, so the hardware wiring should be OK.

Is there another place where I need to disable the Inky auto-detect call (or explicitly select the epd7in3e display) that the install script might have missed?

Thanks again for your help!

[image]
[image]

saulob, 2025-04-28:
I just wanted to say thank you @acscpt for the fork and the help!

After analyzing the new log image I posted, I noticed that it was still using the old code, which didn’t make sense. So I decided it would be better to format my SD card and start from scratch.

I reinstalled everything carefully following your InkyPi-WS instructions, and now everything is working perfectly!

Thanks again for your work and support! Really appreciate it!

Attached is a picture showing my screen working :)

[image]

acscpt, 2025-05-02:
Glad you got it working and thank you!  And bigger thanks to @fatihak for this excellent software!  

_If we see far, it is because we stand on the shoulders of giants!_

acscpt, 2025-06-13:
> Hey, [@acscpt](https://github.com/acscpt)
> 
> Thanks so much for doing this. However, I cannot find these actual instructions from your fork. All I can see are the original instructions. Are your instructions, included the [...-W epd...] part at the end anywhere else other than here?
> 
> Thanks again

https://github.com/acscpt/InkyPi-WS/tree/master is the latest for the fork with my changes in, `main` is tracking this.

adjei7, 2025-06-13:
Yes, everything is working now. Thanks.

closed 2025-06-13 by fatihak

acscpt, 2026-01-11:
Hi there,

Did you install this on a clean image first, with no other attempts to install.

I got myself in a twist originally because the original install for the Pimoroni only enabled CS in /boot/config.txt and then subsequent re-installs just created a mess.

Check /boot/config.txt

[image]

The install script attempts to ensure everything is clean and be idempotent.  I've checked the documentation for the 5.83in and the 7.3in-f (that I have) and they look similar, as do the drivers.  

Can you also confirm that the correct driver is downloaded and availble int he waveshare_epd directory in the src tree?

turtlehopper81, 2026-02-21:
Is the waveshare ACeP suppose to flash for 20 seconds everytime it displays something new