fatihak/InkyPi #16: Clicking Update on plugins causes and error (but it does work)

rgelb, 2025-01-11:
Steps.  

1. Go into the Clock plugin (or any other)
2. Make a change and click Update.
3. Error comes up after a bit.

[image]

This is the error in the DevTools:

[image]

rgelb, 2025-01-17:
I should point out that this only happens in Firefox.  It doesn't happen in Chrome for some reason.

fatihak, 2025-01-21:
Ah I haven't tested the web ui on firefox, will take a look at this when I have some time

fatihak, 2025-02-14:
@rgelb Are you still seeing this error on Firefox? I've been testing the web UI on firefox and updating different plugins (even checked out the commit from the time you posted this) but couldn't recreate it.

If you're still seeing this, can you check what version of Firefox you have installed and also click on the links to see what line this is occurring?

rgelb, 2025-02-14:
@fatihak To update, is it enough to simply `git pull` or is there something else involved?

fatihak, 2025-02-17:
@rgelb You might need to run the update script which re-installs any dependencies added, you can find instructions here:
https://github.com/fatihak/InkyPi?tab=readme-ov-file#update

rgelb, 2025-02-17:
@fatihak I am still getting an error.  I updated per instructions... then clicked Update Now and got this:

[image]

rgelb, 2025-02-17:
In addition, I see `Schedule` has changed to `Add to Playlist`.  The dialog is a bit off - looks like a css issue.  It's an issue on both Chrome and Firefox.  

[image]

throwing max-width on the input fixes, but there is probably a better way to achieve that.  

`<input type="number" id="interval" name="interval" class="form-input" required="" min="1" placeholder="Enter a number" style="max-width: 150px;">`

Is `Add to Playlist` basically same thing as `Schedule`?  Anyway, it would be good to popup error messages (if any) before the dialog gets dismissed - otherwise I am having to retype everything I just typed a second ago.

closed 2025-06-23 by rgelb