fatihak/InkyPi #17: Dimensions of the wood inserts

rgelb, 2025-01-12:
Looking at the linked YouTube tutorial, I can see that you added wood inserts to your Ikea frame at [2:06](https://youtu.be/L5PvQj1vfC4?t=126).  

Would you happen to have the dimensions of those inserts?  I want to 3D print them, but I'd rather get it right the first time.

fatihak, 2025-01-15:
Those dowels are 0.25 inches thick, but I'm unsure about their lengths since I cut them through trial and error.

rgelb, 2025-01-15:
@fatihak I ended up just measuring it and 3D printing the dowels.  The top is bigger than bottom by 50%.  It was ever so slightly too big (maybe like by a fraction of 1mm).  I filed it down with a pair of scissors and the board fit like a glove.  

Attached, in case anyone finds it useful.  
[Glorious Juttuli-Allis.zip](https://github.com/user-attachments/files/18428161/Glorious.Juttuli-Allis.zip)

closed 2025-01-15 by rgelb