fatihak/InkyPi #18: Thank you & little screen issue

compoone, 2025-01-12:
Thank you for this wonderful project. I have placed everything into a slightly larger frame to allow room for a battery pack (power bank), enabling wireless operation by hanging the frame on the wall.
[image]

After updating the images several times, I encountered a small issue with the screen: every 4th row now remains yellow and no longer displays the image correctly (initially, it worked perfectly). Could this be a software issue, or has the screen unfortunately been damaged?
[image]

fatihak, 2025-01-12:
First of all I think your frame looks really good! Did you custom cut the picture mat to fit the dimensions?

Regarding the lines, i haven't encountered that myself and it seems like it's an issue with the screen. Are they appearing on every image? And is the display you're using new? If it is I would recommend reaching out to Pimoroni for support and possible replacement.

compoone, 2025-01-12:
Yes, I had a custom-made passepartout prepared for the frame.
Unfortunately, the issue occurs with every image, whether I set the screen to vertical or horizontal orientation, so it’s likely a problem with the panel.  I will contact the manufacturer to investigate the possible causes. Thank you for the advice.

fatihak, 2025-01-15:
No worries, let me know if you have any questions. Closing this issue in the meantime.

closed 2025-01-15 by fatihak