fatihak/InkyPi #29: How to constrain the width of the AI Text plugin a bit?

rgelb, 2025-02-03:
My device has a bit bigger borders than yours and, as a result, the margins look too close to the border.
I am looking through ai_text.py.  What number exactly should I change to give the text some breathing room on the sides?

[image]

fatihak, 2025-02-04:
You would need to play around with the two variables below, the first one is the text padding and the second is the border padding. You will need to restart the inkypi service when you make any changes. Let me know if that works!

```diff --git a/src/plugins/ai_text/ai_text.py b/src/plugins/ai_text/ai_text.py
index fbd1620..2958cbd 100644
--- a/src/plugins/ai_text/ai_text.py
+++ b/src/plugins/ai_text/ai_text.py
@@ -141,7 +141,7 @@ class AIText(BasePlugin):
         title_height = AIText.get_text_height(title_font, title)

         # Maximum text width in pixels
-        text_padding = max(dim*0.08, 1)
+        text_padding = max(dim*0.10, 1)
         max_text_width = w - (text_padding*2)

         wrapped_lines = AIText.wrap_lines(body, image_draw, font, max_text_width)
@@ -167,7 +167,7 @@ class AIText(BasePlugin):
         dim = min(w,h)
         image_draw = ImageDraw.Draw(image)
         width = max(int(dim*0.015), 1)
-        padding = max(dim*0.05, 1)
+        padding = max(dim*0.08, 1)

         if frame == "Corner":
             corner_length = max(dim*0.16, 1)```

fatihak, 2025-02-14:
Closing this issue as there are no updates.

closed 2025-02-14 by fatihak