fatihak/InkyPi #30: Schedule enhancements

z2k, 2025-02-04:
Congrats and thank you for a great project! 👍 

I have two suggestions to make it even greater:

**Current schedule visible somewhere in the web UI.**
It would be nice to see the current settings as well as what time the next update will run.

**Start schedule at a set time.**
Currently I use my InkyPi to display a newspaper frontpage, scheuled to update once per day. Updating more frequently would just refresh to the same image. Ideally I would have it update at 4am, but in order to achieve that, I believe I would need to set the schedule at 4am with the current flow...? Being able to set a "start time" would make the setup smoother.

Let me know what you think and many thanks!

fatihak, 2025-02-04:
Thank you!
This is already in the works with the "Playlists" feature which will allow you to view the schedule directly in the UI and configure multiple plugins to display at specific time periods. It's still not functional yet but the screenshot below is what I have so far.

Each plugin instance will still have its own refresh configuration to determine how often it should generate a new image, but I get your point about having a configurable start time rather than just a refresh frequency—especially for plugins like displaying a newspaper front page. I will try to include that as part of this update, let me know what you think of the new feature or if you have any suggestions.

[image]

z2k, 2025-02-04:
This sounds pretty much like it would solve both of the suggestions I listed, while at the same time making it way more dynamic. Amazing! :)

Looking forward to it already!

z2k, 2025-02-17:
After two days of testing I think it works great! Well done! 👍

fatihak, 2025-02-17:
@z2k Thanks for trying it out! The PR is now merged so you should be able to checkout the main branch and pull the latest changes, closing this issue

closed 2025-02-17 by fatihak