fatihak/InkyPi #31: New plugin

stuartw0, 2025-02-05:
I created a new plugin more as a proof of concept (a countdown tracker to a specific event/day). I intend on doing more with this, but happy to share in case anyone else can make use of it or improve.

I've never used GItHub, so have no idea about creating branches/Pull requests etc. Can you give me some pointers, or advise how best to get you the code?

Thanks
Stuart

fatihak, 2025-02-08:
Hi Stuart, here is some documentation on crating forks & opening pull requests:
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

You can also video youtube videos on how to do this. Alternatively if you want to zip the plugin files and provide them here I can take a look at adding your plugin since I was planning on adding countdown plugin.

stuartw0, 2025-02-08:
> Hi Stuart, here is some documentation on crating forks & opening pull requests: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo [...]

Thanks for the pointers. I think I've got the idea nd created a pull request for you.

Let me know if any issues.

closed 2025-02-14 by fatihak