fatihak/InkyPi #34: Plugin idea: Image URL Watcher

nemeth-it, 2025-02-08:
_With my limited Python knowledge, it would take very long to achieve this, but maybe an interesting plugin-suggestion/idea for @fatihak:_
___
**Name:** Image URL Watcher

**Description**
The Image URL Watcher plugin could enhance the functionality of InkyPi by enabling it to periodically check an external image URL and automatically update the E-Ink display when changes are detected. This allows for dynamic content display without the need for manual intervention.

**Features**
- **Image URL Monitoring**: Users can specify an external URL from which an image is regularly fetched.
- **Configurable Update Interval**: Define a time interval for the plugin to check the URL for changes.
- **Automatic Image Update**: If the plugin detects a change in the specified image source, it downloads the new image and updates the InkyPi display automatically.
- **Could also be used in the upcoming Playlist Feature**: By monitoring multiple image sources, different images from various sources can be displayed at specific times.
___
I think this would open up a lot of possibilities when people can just focus on updating their images on servers or other external sources and InkyPi just receives the result to update these.

nov1n, 2025-02-09:
I'm currently looking to do something similar: Update the frame daily with a random cartoon. It's possible to upload an image using the `update_now` endpoint. Below is a curl command to set the image over the network.

```bash
IMG="~/Downloads/my-img.png"
curl "http://inkypi.local/update_now" -X POST -F "imageFile=@${IMG}" -F "plugin_id=image_upload"
```

You could periodically run this in order to achieve something similar to what you describe in your ticket, without the need for external plugins.

A plugin might still be nice, as this api is private and undocumented, but I thought I'd share my current working solution for reference.

nemeth-it, 2025-02-09:
Thank you very much, this was the next thing I planned to investigate. You saved me a some time, thank you!

fatihak, 2025-02-09:
Pulling images from a custom url should be really straightforward to implement with a new plugin. I created a public trello board for the project and added this suggestion, you can view & vote on items here: https://trello.com/b/SWJYWqe4/inkypi

@nov1n Calling the update endpoint is actually pretty clever, I didn't think it would be used like that outside of the UI haha. I think having a plugin that polls a custom url is still a good suggestion tho, will try to tackle that sometime.

doofmars, 2025-04-19:
@nemeth-it I had similar issues and created a simple plugin.
Currently it can grab an image from a fixed URL
There is a PR open, you can get the plugin by checking out my fork or adding the required files manually.
https://github.com/fatihak/InkyPi/pull/100

With regards to the polling this should be done by the scheduling components of InkyPy (playlist) and not by the plugin itself.

In theory you should be able to create a playlist that updates the display with images from different URLs.

I will do some testing but any feedback is welcome :)

fatihak, 2025-06-23:
Closing due to lack of response, as mentioned above the Image URL plugin should address the request.

closed 2025-06-23 by fatihak