fatihak/InkyPi #37: error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8) error:

JKeithDaniels, 2025-02-13:
When I execute under ssh $ git clone https://github.com/fatihak/InkyPi.git
I get the following
Cloning into 'InkyPi'...
remote: Enumerating objects: 494, done.
remote: Counting objects: 100% (138/138), done.
remote: Compressing objects: 100% (75/75), done.
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 9805 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

I am using a Rapberry Pi Zero 2W and it is still attached to wi-fi and the internet.
This seems to be different to the previous and closed Pip installation error on Pi Zero W in troubleshooting.

fatihak, 2025-02-14:
This issue seems to be related to Git rather than the installation for this project. You can try some of the suggestions in this stack overflow:
https://stackoverflow.com/questions/59282476/error-rpc-failed-curl-92-http-2-stream-0-was-not-closed-cleanly-protocol-erro

JKeithDaniels, 2025-02-14:
The suggestion to change to http1.1, git clone and change back has worked -thank you

closed 2025-02-14 by JKeithDaniels