fatihak/InkyPi #48: Images - respect EXIF rotation flags

iamwoz, 2025-02-22:
Recently updated and loving having a playlist of images that can be displayed.

Walked past my InkyPi today and saw one of the images upside down, queue some googling and discovered that images have EXIF flags for rotation (seems like Apple Photos and Google Photos when you rotate an image doesn't actually edit the image but instead just updates the EXIF rotation info).

Possible to add support for this?

More info: https://jdhao.github.io/2019/07/31/image_rotation_exif_info/

fatihak, 2025-03-29:
I just pushed a change to address this issue so image exif rotations are respected. You'll have to re-upload any existing images for it to work. Let me know if its working as expected now.
```
cd InkyPi
git pull
sudo systemctl restart inkypi.service
```

iamwoz, 2025-03-29:
perfect thanks very much!

closed 2025-03-29 by iamwoz