fatihak/InkyPi #53: Plugin Cycle Interval not working

scott-friedman, 2025-03-12:
Install went great and have no issues manually pushing updates via plugins - however, it does not seem to cycle automatically in any way. I have tried various intervals using both minutes / hours, and there are no automatic updates being pushed to the display.

fatihak, 2025-03-12:
Can you check the logs for any errors [troublshooting.md](https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md#debugging)

scott-friedman, 2025-03-12:
[troubleshooting.md](https://github.com/user-attachments/files/19218873/troubleshooting.md)
yes here is the file with the errors, any advice?

fatihak, 2025-03-13:
Looks like thats just the troubleshooting documentation, can you reattach the file

scott-friedman, 2025-03-13:
Wait haha I fixed it. I didn't realize I had to have an active playlist to have it auto refresh - I was thinking if I only wanted one screen I could do that outside of a playlist. Problem solved :D

closed 2025-03-13 by scott-friedman