fatihak/InkyPi #55: Can this work with displaying a home assistant dashboard ?

mfmseth, 2025-03-14:
Wounding in there can be a plugin for home assistant dashboard

fatihak, 2025-03-18:
I'm not too familiar with Home Assistant dashboards, but would a screenshot plugin that loads a custom URL work for this use case?

bradsour, 2025-03-19:
> I'm not too familiar with Home Assistant dashboards, but would a screenshot plugin that loads a custom URL work for this use case?

Yes I do believe that would work. I was here looking for something similar myself.

mfmseth, 2025-03-19:
Yea the dashboard is just a website at the end of the day

jdent421, 2025-05-20:
The issue I ran into using the screenshot plugin is that you need to log into HA with user/pw to view dashboards. It looks like there is a potential workaround, but it requires modifying configuration.yaml in HA to bypass login for IP specific requests. Not sure if there is a more elegant solution for this while still using the screenshot plugin. Has anyone had success?

jcchalte, 2025-07-04:
Hi. Is there a way to setup custom http headers in your url screenshot plugin ? This could unlock many scénarios, including the Home Assistant one !

If there was a way to add a header to the http request, we might be able to use the Authentication header, with 'Bearer <token>' where token is a long lived token that can be created in your Home Assistant profile. I have not tested this but it should work in theory. Still waiting for my parts to arrive :-)

phattmatt, 2025-07-29:
I suggest closing this issue since there appears to be a working solution using a HA addon and the Image URL plugin: https://github.com/fatihak/InkyPi/issues/55#issuecomment-3036670330

Or bypassing the authentication in HA for the InkyPi IP address: https://github.com/fatihak/InkyPi/issues/55#issuecomment-2892670195

closed 2025-07-31 by phattmatt