fatihak/InkyPi #60: Weather error

cellfreak, 2025-03-17:
Getting this error. Error! An error occurred: 'NoneType' object has no attribute 'save'

I have created an API and saved it to the .env

stemow, 2025-03-19:
Same issue :(

fatihak, 2025-03-19:
@cellfreak @stemow This is likely an issue when when taking a screenshot to generate an image. Can you provide the logs? You can find instructions here: [troubleshooting.md](https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md)

Also, are you using a Pi Zero W?

cellfreak, 2025-03-19:
It works correctly with Opera browser. Fails with Edge and Chrome. [image]

stemow, 2025-03-21:
Looks like hardware limitation of Pi Zero:

Mar 21 07:28:51 stemow-inky inkypi[404]: 07:28:51 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:51] "GET /plugin/weather HTTP/1.1" 200 -
Mar 21 07:28:52 stemow-inky inkypi[404]: 07:28:52 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:52] "GET /static/styles/main.css HTTP/1.1" 304 -
Mar 21 07:28:52 stemow-inky inkypi[404]: 07:28:52 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:52] "GET /images/weather/icon.png HTTP/1.1" 304 -
Mar 21 07:28:52 stemow-inky inkypi[404]: 07:28:52 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:52] "GET /static/scripts/response_modal.js HTTP/1.1" 304 -
Mar 21 07:28:52 stemow-inky inkypi[404]: 07:28:52 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:52] "GET /images/base_plugin/frames/blank.png HTTP/1.1" 304 -
Mar 21 07:28:52 stemow-inky inkypi[404]: 07:28:52 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:52] "GET /images/base_plugin/frames/corner.png HTTP/1.1" 304 -
Mar 21 07:28:52 stemow-inky inkypi[404]: 07:28:52 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:52] "GET /images/base_plugin/frames/top_and_bottom.png HTTP/1.1" 304 -
Mar 21 07:28:52 stemow-inky inkypi[404]: 07:28:52 - INFO - werkzeug - 192.168.1.39 - - [21/Mar/2025 07:28:52] "GET /images/base_plugin/frames/rectangle.png HTTP/1.1" 304 -
Mar 21 07:28:57 stemow-inky inkypi[404]: 07:28:57 - INFO - refresh_task - Manual update requested
Mar 21 07:28:58 stemow-inky inkypi[404]: 07:28:58 - ERROR - utils.image_utils - Failed to take screenshot:
Mar 21 07:28:58 stemow-inky inkypi[404]: 07:28:58 - ERROR - utils.image_utils - The hardware on this system lacks support for NEON SIMD extensions.
Mar 21 07:28:58 stemow-inky inkypi[404]: We now require NEON or equivalent architecture extensions on ARM-based
Mar 21 07:28:58 stemow-inky inkypi[404]: machines. See https://lists.debian.org/debian-devel/2023/09/msg00175.html
Mar 21 07:28:58 stemow-inky inkypi[404]: for more information.
Mar 21 07:28:58 stemow-inky inkypi[404]: 07:28:58 - ERROR - root - Exception during refresh
Mar 21 07:28:58 stemow-inky inkypi[404]: Traceback (most recent call last):
Mar 21 07:28:58 stemow-inky inkypi[404]:   File "/home/stemow/InkyPi/src/refresh_task.py", line 107, in _run
Mar 21 07:28:58 stemow-inky inkypi[404]:     image_hash = compute_image_hash(image)
Mar 21 07:28:58 stemow-inky inkypi[404]:                  ^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 21 07:28:58 stemow-inky inkypi[404]:   File "/home/stemow/InkyPi/src/utils/image_utils.py", line 61, in compute_image_hash
Mar 21 07:28:58 stemow-inky inkypi[404]:     image = image.convert("RGB")
Mar 21 07:28:58 stemow-inky inkypi[404]:             ^^^^^^^^^^^^^
Mar 21 07:28:58 stemow-inky inkypi[404]: AttributeError: 'NoneType' object has no attribute 'convert'
Mar 21 07:28:58 stemow-inky inkypi[404]: 07:28:58 - INFO - werkzeug - 192.168.86.39 - - [21/Mar/2025 07:28:58] "POST /update_now HTTP/1.1" 500 -

dahorowitz, 2025-03-26:
FWIW, I'm getting the same error message and log error, and I'm using a Pi Zero 2W...

stemow, 2025-03-26:
How interesting. I pulled a Pi Zero 2W and set it up for this project and it works perfectly. I flashed PiOS bookworm with desktop image.

fatihak, 2025-03-29:
@cellfreak based on your latest comment, it seems like its working for you now. Is this issue good to close?

HRB-YuPai, 2025-04-08:
@fatihak 

Hello fatihak:

I got the same error - weather screenshot is dead

OS：Raspberry Pi OS 64bit lite Bookworm
hardware: RPI Zero 2w
Python Version: 3.11 along with pip3
network: both ethernet (usb-c to ethernet adapter)  and wifi are not work

Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:   File "/home/loneone/InkyPi/src/refresh_task.py", line 106, in _run
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:   File "/home/loneone/InkyPi/src/refresh_task.py", line 206, in execute
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:     return plugin.generate_image(self.plugin_settings, device_config)
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:   File "/home/loneone/InkyPi/src/plugins/weather/weather.py", line 75, in generate_image
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]:     raise RuntimeError("Failed to take screenshot, please check logs.")
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]: RuntimeError: Failed to take screenshot, please check logs.
Apr 09 06:41:24 LoneOne-Weather-Board inkypi[582]: 06:41:24 - INFO - werkzeug - 10.0.0.194 - - [09/Apr/2025 06:41:24] "POST /update_now HTTP/1.1" 500 -

fatihak, 2025-05-21:
@HRB-YuPai Could you try flashing the full Raspberry Pi OS Bookworm (Desktop version) instead of the Lite one and see if that resolves the issue?

closed 2025-05-21 by fatihak