fatihak/InkyPi #61: 3d printable mount

MatthewAlner, 2025-03-17:
Firstly great project thanks 😄!

I just wanted to say I made a 3D printable shim/surround/mount for the Inky Impression 7 & Ikea Rodlam for my build. Files are here [https://makerworld.com/en/models/1221196-ikea-rodlam-inky-impression-7-mount](https://makerworld.com/en/models/1221196-ikea-rodlam-inky-impression-7-mount#profileId-1238116)

[image]

[image]

I would have PRed it into the README but I didn't want to be so presumptuous 😅  links there if you want to include it, if not no worries 🤙

fatihak, 2025-03-18:
Thank you, I actually need this myself :) 
I'll create a separate doc page with different mounts/frames from the community and link to it from the README

dsbilling, 2025-03-19:
This one is also super nice: https://www.printables.com/model/1227797-pimoroni-inky-impression-picture-frame/files

fatihak, 2025-03-29:
Added to https://github.com/fatihak/InkyPi/blob/main/docs/community.md! Thanks again, closing this issue.

closed 2025-03-29 by fatihak