fatihak/InkyPi #63: Error updating Weather Details

petehallett, 2025-03-19:
, I have updated to the latest code base, I have added the Open Weather API key to the .env file in the root of the project folder, permissions "-rw-r--r--" .When I attempt to run Update Now in the Weather component of the Web UI, I get this error. "Error! An error occurred: Failed to retrieve weather data." Are there logs that I can check to see what is generating the error? 
Image sample below. 

[image]

fatihak, 2025-03-19:
Yes, can you provide the logs? You can find instructions in: [troubleshooting.md](https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md)

petehallett, 2025-03-20:
Thanks, here is the section that was logged. Please let me know if you need further testing or logs. 

`Mar 20 12:53:54 inky inkypi[653]: 12:53:54 - INFO - refresh_task - Not time to update display. | latest_update: 2025-03-20 11:58:57 | plugin_cycle_interval>
Mar 20 12:54:21 inky inkypi[653]: 12:54:21 - INFO - refresh_task - Manual update requested
Mar 20 12:54:21 inky inkypi[653]: 12:54:21 - ERROR - root - Failed to retrieve weather data: b'{"cod":401, "message": "Please note that using One Call 3.0 >
Mar 20 12:54:21 inky inkypi[653]: 12:54:21 - ERROR - root - Exception during refresh
Mar 20 12:54:21 inky inkypi[653]: Traceback (most recent call last):
Mar 20 12:54:21 inky inkypi[653]:   File "/home/admin/InkyPi/src/refresh_task.py", line 106, in _run
Mar 20 12:54:21 inky inkypi[653]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Mar 20 12:54:21 inky inkypi[653]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 20 12:54:21 inky inkypi[653]:   File "/home/admin/InkyPi/src/refresh_task.py", line 206, in execute
Mar 20 12:54:21 inky inkypi[653]:     return plugin.generate_image(self.plugin_settings, device_config)
Mar 20 12:54:21 inky inkypi[653]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 20 12:54:21 inky inkypi[653]:   File "/home/admin/InkyPi/src/plugins/weather/weather.py", line 58, in generate_image
Mar 20 12:54:21 inky inkypi[653]:     weather_data = self.get_weather_data(api_key, units, lat, long)
Mar 20 12:54:21 inky inkypi[653]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 20 12:54:21 inky inkypi[653]:   File "/home/admin/InkyPi/src/plugins/weather/weather.py", line 194, in get_weather_data
Mar 20 12:54:21 inky inkypi[653]:     raise RuntimeError("Failed to retrieve weather data.")
Mar 20 12:54:21 inky inkypi[653]: RuntimeError: Failed to retrieve weather data.
Mar 20 12:54:21 inky inkypi[653]: 12:54:21 - INFO - werkzeug - 10.1.1.35 - - [20/Mar/2025 12:54:21] "POST /update_now HTTP/1.1" 500 -
lines 55-100/100 (END)`

fatihak, 2025-03-20:
Hm, did you subscribe to the One Call Api 3.0? You can find instructions here: https://github.com/fatihak/InkyPi/blob/main/docs/api_keys.md#open-weather-map

petehallett, 2025-03-20:
Hi, worked out the issue, I have used OpenWeatherMap in the past and used on of my existing APK keys. Once I singed up for the V3.0 subscription, I created a new key and all was good. Obviously uses a different subscription based call than what I had setup before, Thanks again, love the new Weather Plugin.

closed 2025-03-20 by petehallett