fatihak/InkyPi #65: AI functions have stopped working.

cellfreak, 2025-03-19:
Mar 19 17:25:07 InkyPi inkypi[2275]: 17:25:07 - INFO - werkzeug - 192.168.0.5 - - [19/Mar/2025 17:25:07] ">
Mar 19 17:25:07 InkyPi inkypi[2275]: 17:25:07 - INFO - werkzeug - 192.168.0.5 - - [19/Mar/2025 17:25:07] ">
Mar 19 17:25:07 InkyPi inkypi[2275]: 17:25:07 - INFO - werkzeug - 192.168.0.5 - - [19/Mar/2025 17:25:07] ">
Mar 19 17:25:07 InkyPi inkypi[2275]: 17:25:07 - INFO - werkzeug - 192.168.0.5 - - [19/Mar/2025 17:25:07] ">
Mar 19 17:25:10 InkyPi inkypi[2275]: 17:25:10 - INFO - refresh_task - Manual update requested
Mar 19 17:25:10 InkyPi inkypi[2275]: 17:25:10 - INFO - refresh_task - Refreshing plugin instance. | plugin>
Mar 19 17:25:12 InkyPi inkypi[2275]: 17:25:12 - INFO - plugins.ai_text.ai_text - Getting random text promp>
Mar 19 17:25:14 InkyPi inkypi[2275]: 17:25:14 - INFO - httpx - HTTP Request: POST https://api.openai.com/v>
Mar 19 17:25:14 InkyPi inkypi[2275]: 17:25:14 - INFO - openai._base_client - Retrying request to /chat/com>
Mar 19 17:25:14 InkyPi inkypi[2275]: 17:25:14 - INFO - httpx - HTTP Request: POST https://api.openai.com/v>
Mar 19 17:25:14 InkyPi inkypi[2275]: 17:25:14 - INFO - openai._base_client - Retrying request to /chat/com>
Mar 19 17:25:15 InkyPi inkypi[2275]: 17:25:15 - INFO - httpx - HTTP Request: POST https://api.openai.com/v>
Mar 19 17:25:16 InkyPi inkypi[2275]: 17:25:16 - ERROR - plugins.ai_text.ai_text - Failed to make Open AI r>
Mar 19 17:25:16 InkyPi inkypi[2275]: 17:25:16 - ERROR - root - Exception during refresh
Mar 19 17:25:16 InkyPi inkypi[2275]: Traceback (most recent call last):
Mar 19 17:25:16 InkyPi inkypi[2275]:   File "/home/pi/InkyPi/src/plugins/ai_text/ai_text.py", line 43, in >
Mar 19 17:25:16 InkyPi inkypi[2275]:     prompt_response = AIText.fetch_text_prompt(ai_client, text_model,>
Mar 19 17:25:16 InkyPi inkypi[2275]:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>
Mar 19 17:25:16 InkyPi inkypi[2275]:   File "/home/pi/InkyPi/src/plugins/ai_text/ai_text.py", line 80, in >
Mar 19 17:25:16 InkyPi inkypi[2275]:     response = ai_client.chat.completions.create(
Mar 19 17:25:16 InkyPi inkypi[2275]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 19 17:25:16 InkyPi inkypi[2275]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/op>
Mar 19 17:25:16 InkyPi inkypi[2275]:     return func(*args, **kwargs)
Mar 19 17:25:16 InkyPi inkypi[2275]:            ^^^^^^^^^^^^^^^^^^^^^
Mar 19 17:25:16 InkyPi inkypi[2275]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/op>
Mar 19 17:25:16 InkyPi inkypi[2275]:     return self._post(
Mar 19 17:25:16 InkyPi inkypi[2275]:            ^^^^^^^^^^^
Mar 19 17:25:16 InkyPi inkypi[2275]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/op>
Mar 19 17:25:16 InkyPi inkypi[2275]:     return cast(ResponseT, self.request(cast_to, opts, stream=stream,>
Mar 19 17:25:16 InkyPi inkypi[2275]:                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>
Mar 19 17:25:16 InkyPi inkypi[2275]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/op>
Mar 19 17:25:16 InkyPi inkypi[2275]:     return self._request(
Mar 19 17:25:16 InkyPi inkypi[2275]:            ^^^^^^^^^^^^^^
Mar 19 17:25:16 InkyPi inkypi[2275]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/op>
set mark: ...skipping...

fatihak, 2025-03-20:
Hi, the important line is cut off in the logs you posted:
Mar 19 17:25:16 InkyPi inkypi[2275]: 17:25:16 - ERROR - plugins.ai_text.ai_text - Failed to make Open AI r>

Can you also check your OpenAI account, another user had reported a similar issue and realized they had reached a billing limit/ ran out of credit balance.

cellfreak, 2025-03-20:
I am below my credit limit. Will get another log later today.

cellfreak, 2025-03-20:
Mar 20 17:04:05 InkyPi inkypi[503]: 17:04:05 - INFO - refresh_task - Manual update requested
Mar 20 17:04:06 InkyPi inkypi[503]: 17:04:06 - INFO - plugins.ai_image.ai_image - Generating image for prompt: watercolor 2019 nissan leaf, model: dall-e-3, quality: standard
Mar 20 17:04:07 InkyPi inkypi[503]: 17:04:07 - INFO - httpx - HTTP Request: POST https://api.openai.com/v1/images/generations "HTTP/1.1 400 Bad Request"

fatihak, 2025-03-29:
I tested with the image prompts and settings from your screenshot but haven't been able to reproduce this issue

gertlind, 2025-04-10:
I also get errors with the openAi image.

Apr 10 16:01:06 inky-5 inkypi[1374]:   File "/home/pi/InkyPi/src/refresh_task.py", line 106, in _run
Apr 10 16:01:06 inky-5 inkypi[1374]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Apr 10 16:01:06 inky-5 inkypi[1374]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 10 16:01:06 inky-5 inkypi[1374]:   File "/home/pi/InkyPi/src/refresh_task.py", line 206, in execute
Apr 10 16:01:06 inky-5 inkypi[1374]:     return plugin.generate_image(self.plugin_settings, device_config)
Apr 10 16:01:06 inky-5 inkypi[1374]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 10 16:01:06 inky-5 inkypi[1374]:   File "/home/pi/InkyPi/src/plugins/ai_text/ai_text.py", line 46, in generate_image
Apr 10 16:01:06 inky-5 inkypi[1374]:     raise RuntimeError("Open AI request failure, please check logs.")
Apr 10 16:01:06 inky-5 inkypi[1374]: RuntimeError: Open AI request failure, please check logs.

fatihak, 2025-04-10:
@gertlind Is it failing on all ai image prompts? Can you also provide the logs before this?

phattmatt, 2025-06-22:
@cellfreak @gertlind 

Are you still seeing an issue, or can this be closed?