fatihak/InkyPi #67: Service EEPROM error

skynetsysadmin, 2025-03-22:
Trying to get this to work with a Pi 3 A+ (same issue with a Pi Zero 2W), the latest OS (Bookworm), and an Inky Impression 7.3. The installer script ran successfully...

> bret@Skynet-EInk:~ $ git clone https://github.com/fatihak/InkyPi.git
> Cloning into 'InkyPi'...
> remote: Enumerating objects: 517, done.
> remote: Counting objects: 100% (194/194), done.
> remote: Compressing objects: 100% (115/115), done.
> remote: Total 517 (delta 120), reused 79 (delta 79), pack-reused 323 (from 1)
> Receiving objects: 100% (517/517), 43.74 MiB | 1.54 MiB/s, done.
> Resolving deltas: 100% (187/187), done.
> bret@Skynet-EInk:~ $ cd InkyPi
> bret@Skynet-EInk:~/InkyPi $ sudo bash install/install.sh
> Checking if inkypi.service is running
>         inkypi.service not running [✔]
> Enabling interfaces required for inkypi
>         SPI Interface has been enabled. [✔]
>         I2C Interface has been enabled. [✔]
> Installing system dependencies.  [✔]
> Installing inkypi to /usr/local/inkypi
>         Removing existing installation found at /usr/local/inkypi [✔]
>         Creating symlink from /home/bret/InkyPi/install/../src to /usr/local/inkypi/src [✔]
> Creating python virtual environment.
>         Installing python dependencies.  [\] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pydantic/
>         Installing python dependencies.  [✔]
> Adding executable to /usr/local/bin/inkypi
> Copying config files to /home/bret/InkyPi/install/../src/config
>         Copying device.config to /home/bret/InkyPi/install/../src/config [✔]
> Installing inkypi systemd service.
> INKYPI Installation Complete! [✔]
> [•] A reboot of your Raspberry Pi is required for the changes to take effect
> [•] After your Pi is rebooted, you can access the web UI by going to 'Skynet-EInk.local' or '192.168.1.228' in your browser.
> [•] If you encounter any issues or have suggestions, please submit them here: https://github.com/fatihak/InkyPi/issues
> Would you like to restart your Raspberry Pi now? [Y/N] Y

After reboot, the same default image displays. It is not showing a logo, as described. I started troubleshooting and the service status reports.

> bret@Skynet-EInk:~ $ sudo systemctl status inkypi.service
> ○ inkypi.service - InkyPi App
>      Loaded: loaded (/etc/systemd/system/inkypi.service; enabled; preset: enabled)
>      Active: inactive (dead) since Sat 2025-03-22 11:49:47 MDT; 34s ago
>    Duration: 54.431s
>     Process: 794 ExecStart=/usr/local/bin/inkypi -d (code=exited, status=0/SUCCESS)
>    Main PID: 794 (code=exited, status=0/SUCCESS)
>         CPU: 4.216s
> 
> Mar 22 11:49:46 Skynet-EInk inkypi[818]:     display_manager = DisplayManager(device_config)
> Mar 22 11:49:46 Skynet-EInk inkypi[818]:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> Mar 22 11:49:46 Skynet-EInk inkypi[818]:   File "/home/bret/InkyPi/src/display_manager.py", line 11, in __init__
> Mar 22 11:49:46 Skynet-EInk inkypi[818]:     self.inky_display = auto()
> Mar 22 11:49:46 Skynet-EInk inkypi[818]:                         ^^^^^^
> Mar 22 11:49:46 Skynet-EInk inkypi[818]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/auto.p>
> Mar 22 11:49:46 Skynet-EInk inkypi[818]:     raise RuntimeError("No EEPROM detected! You must manually initialise your >
> Mar 22 11:49:46 Skynet-EInk inkypi[818]: RuntimeError: No EEPROM detected! You must manually initialise your Inky board.
> Mar 22 11:49:47 Skynet-EInk systemd[1]: inkypi.service: Deactivated successfully.
> Mar 22 11:49:47 Skynet-EInk systemd[1]: inkypi.service: Consumed 4.216s CPU time.

Searching for that 'No EEPROM' error indicated that it is usually caused by the SPI not being enabled. It has been enabled via raspi-config,and the installer script seems to agree that it is enabled.

I have seen other users having issue with the 'inky' library not auto-detecting some boards.

I tried the instructions from Pimironi and their default install script. "curl https://get.pimoroni.com/inky | bash"

But, that failed...it looks like a venv issue?

> bret@Skynet-EInk:~ $ curl https://get.pimoroni.com/inky | bash
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100 34932  100 34932    0     0  28003      0  0:00:01  0:00:01 --:--:-- 28012
> 
> This script will install everything needed to use
> Inky pHAT & wHAT
> 
> Always be careful when running scripts and commands copied
> from the internet. Ensure they are from a trusted source.
> 
> This script should -- only be run on a Raspberry Pi with RPi OS --
> other systems and SBCs are not supported and may explode!
> 
> If you want to see what this script does before running it,
> you should run: 'curl https://get.pimoroni.com/inky'
> 
> Note: Inky pHAT & wHAT requires SPI communication
> Note: Inky pHAT & wHAT requires I2C communication
> 
> Do you wish to continue? [y/N] y
> 
> Checking environment...
> Updating apt indexes...
> ............
> 
> Checking hardware requirements...
> 
> Checking for packages required for GPIO control...
> raspi-gpio is already installed
> .....................RPi.GPIO installed and up-to-date
> 
> SPI already enabled
> 
> Checking packages required by SPI interface...
> spidev installed and up-to-date
> 
> I2C already enabled
> 
> Checking packages required by I2C interface...
> Python 2 smbus installed and up-to-date
> Python 3 smbus installed and up-to-date
> 
> Inky pHAT & wHAT comes with examples and documentation that you may wish to install.
> Performing a full install will ensure those resources are installed,
> along with all required dependencies. It may however take a while!
> 
> Do you wish to perform a full install? [y/N] Y
> 
> Checking for dependencies...
> python3-numpy is already installed
> python3-pil is already installed
> python3-lxml is already installed
> python3-smbus is already installed
> 
> Installing Inky pHAT & wHAT library for Python 3...
> 
> error: externally-managed-environment
> 
> × This environment is externally managed
> ╰─> To install Python packages system-wide, try apt install
>     python3-xyz, where xyz is the package you are trying to
>     install.
> 
>     If you wish to install a non-Debian-packaged Python package,
>     create a virtual environment using python3 -m venv path/to/venv.
>     Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
>     sure you have python3-full installed.
> 
>     For more information visit http://rptl.io/venv
> 
> note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
> hint: See PEP 668 for the detailed specification.
> Python 3 library install failed!
> If problems persist, visit forums.pimoroni.com for support\

I am not sure what else to try to troubleshoot the issue. Any suggestions?

Thanks

fatihak, 2025-03-23:
From InkyPi installation logs you provided, it seems like it failed to install the python dependencies:
`Installing python dependencies. [] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pydantic/`

Can you try connecting to the InkyPi python venv and manually installing the python libraries:
```
sudo su
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip install -r install/requirements.txt
deactivate
```

skynetsysadmin, 2025-03-23:
It failed on the first try, but a retry installed the dependency from the initial install script. I also ran the initial install script more than one to make sure the dependencies were there. 

Running the commands manually as you recommended seems to confirm that they are installed.

> root@Skynet-EInk:/home/bret/InkyPi# source "/usr/local/inkypi/venv_inkypi/bin/activate"
> (venv_inkypi) root@Skynet-EInk:/home/bret/InkyPi# pip install -r install/requirements.txt
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Requirement already satisfied: flask==3.1.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 1)) (3.1.0)
> Requirement already satisfied: python-dotenv==1.0.1 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 2)) (1.0.1)
> Requirement already satisfied: inky==2.0.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 3)) (2.0.0)
> Requirement already satisfied: requests==2.32.3 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 4)) (2.32.3)
> Requirement already satisfied: urllib3==2.3.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 5)) (2.3.0)
> Requirement already satisfied: werkzeug==3.1.3 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 6)) (3.1.3)
> Requirement already satisfied: pillow==11.0.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 7)) (11.0.0)
> Requirement already satisfied: pytz==2024.2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 8)) (2024.2)
> Requirement already satisfied: openai==1.58.1 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 9)) (1.58.1)
> Requirement already satisfied: numpy==2.2.1 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from -r install/requirements.txt (line 10)) (2.2.1)
> Requirement already satisfied: Jinja2>=3.1.2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from flask==3.1.0->-r install/requirements.txt (line 1)) (3.1.6)
> Requirement already satisfied: itsdangerous>=2.2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from flask==3.1.0->-r install/requirements.txt (line 1)) (2.2.0)
> Requirement already satisfied: click>=8.1.3 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from flask==3.1.0->-r install/requirements.txt (line 1)) (8.1.8)
> Requirement already satisfied: blinker>=1.9 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from flask==3.1.0->-r install/requirements.txt (line 1)) (1.9.0)
> Requirement already satisfied: gpiodevice>=0.0.3 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from inky==2.0.0->-r install/requirements.txt (line 3)) (0.0.5)
> Requirement already satisfied: smbus2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from inky==2.0.0->-r install/requirements.txt (line 3)) (0.5.0)
> Requirement already satisfied: spidev in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from inky==2.0.0->-r install/requirements.txt (line 3)) (3.6)
> Requirement already satisfied: charset-normalizer<4,>=2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from requests==2.32.3->-r install/requirements.txt (line 4)) (3.4.1)
> Requirement already satisfied: idna<4,>=2.5 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from requests==2.32.3->-r install/requirements.txt (line 4)) (3.10)
> Requirement already satisfied: certifi>=2017.4.17 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from requests==2.32.3->-r install/requirements.txt (line 4)) (2025.1.31)
> Requirement already satisfied: MarkupSafe>=2.1.1 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from werkzeug==3.1.3->-r install/requirements.txt (line 6)) (3.0.2)
> Requirement already satisfied: anyio<5,>=3.5.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r install/requirements.txt (line 9)) (4.9.0)
> Requirement already satisfied: distro<2,>=1.7.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r install/requirements.txt (line 9)) (1.9.0)
> Requirement already satisfied: httpx<1,>=0.23.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r install/requirements.txt (line 9)) (0.28.1)
> Requirement already satisfied: jiter<1,>=0.4.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r install/requirements.txt (line 9)) (0.9.0)
> Requirement already satisfied: pydantic<3,>=1.9.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r install/requirements.txt (line 9)) (2.10.6)
> Requirement already satisfied: sniffio in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r install/requirements.txt (line 9)) (1.3.1)
> Requirement already satisfied: tqdm>4 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r install/requirements.txt (line 9)) (4.67.1)
[... cut ...]
> Mar 23 06:41:16 Skynet-EInk inkypi[819]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/auto.p>
> Mar 23 06:41:16 Skynet-EInk inkypi[819]:     raise RuntimeError("No EEPROM detected! You must manually initialise your >
> Mar 23 06:41:16 Skynet-EInk inkypi[819]: RuntimeError: No EEPROM detected! You must manually initialise your Inky board.
> Mar 23 06:41:18 Skynet-EInk systemd[1]: inkypi.service: Deactivated successfully.
> Mar 23 06:41:18 Skynet-EInk systemd[1]: inkypi.service: Consumed 4.143s CPU time.

skynetsysadmin, 2025-03-23:
Even the Pimironi install linked from the product page where I bought the display is failing. The examples from them say the same general thing: that it cannot detect the EEPROM.

> (pimoroni) bret@Skynet-EInk:~/Pimoroni/inky $ cd ~/Pimoroni/inky/examples
> (pimoroni) bret@Skynet-EInk:~/Pimoroni/inky/examples $ python name-badge.py --type "auto" --colour "red" --name "Inigo Montoya"
> Traceback (most recent call last):
>   File "/home/bret/Pimoroni/inky/examples/name-badge.py", line 5, in <module>
>     from font_hanken_grotesk import HankenGroteskBold, HankenGroteskMedium
> ModuleNotFoundError: No module named 'font_hanken_grotesk'
> (pimoroni) bret@Skynet-EInk:~/Pimoroni/inky/examples $ ls
> 7color  clean.py  identify.py  logo.py  name-badge.py  phat  tests  what
> (pimoroni) bret@Skynet-EInk:~/Pimoroni/inky/examples $ python3 clean.py
> Inky pHAT: Clean
> 
> Displays solid blocks of red, black, and white to clean the Inky pHAT
> display of any ghosting.
> 
> 
> Failed to detect an Inky board. Trying --type/--colour arguments instead...
> 
> usage: clean.py [-h] [--simulate] --type {what,phat,phatssd1608,impressions,7colour,whatssd1683,impressions73}
>                 [--colour {red,black,yellow}]
> clean.py: error: the following arguments are required: --type/-t
> (pimoroni) bret@Skynet-EInk:~/Pimoroni/inky/examples $ python3 logo.py
> Inky pHAT/wHAT: Logo
> 
> Displays the Inky pHAT/wHAT logo.
> 
> 
> Failed to detect an Inky board. Trying --type/--colour arguments instead...
> 
> usage: logo.py [-h] [--simulate] --type {what,phat,phatssd1608,impressions,7colour,whatssd1683,impressions73}
>                [--colour {red,black,yellow}]
> logo.py: error: the following arguments are required: --type/-t
> (pimoroni) bret@Skynet-EInk:~/Pimoroni/inky/examples $ python3 identify.py
> 
> No display EEPROM detected,
> you might have an old Inky board that doesn't have an EEPROM - eg: early Inky pHAT boards.
> 
> Try running examples with --colour <black/red/yellow>
> 
> Or writing your code using:
> 
> from inky.phat import InkyPHAT
> 
> display = InkyPHAT("<black/red/yellow>")

The board is this one:

https://www.pishop.us/product/inky-impression-7-3-7-colour-epaper-e-ink-hat/

scotthsieh0503, 2025-03-24:
@skynetsysadmin the issue you described happened to me as well. From what I can gather from the Pimironi and examples that it might actually be a faulty screen or a bug I their firmware where it is unable to auto-detect the screen properly. If you tried to run the example code provided https://github.com/pimoroni/inky/tree/main/examples that same error persist. 

What I end up doing is instead of using: 
`from inky import Auto`

I directly set the screen I have. with
`from inky.inky_ac073tc1a import Inky`

fatihak, 2025-03-24:
As @scotthsieh0503 suggested this seems like an issue related to the Inky library or a faulty screen.

From the Pimoroni Inky [README](https://github.com/pimoroni/inky?tab=readme-ov-file#manual-setup):
> If you have an older Inky without an EEPROM, you can specify the type manually. The Inky library contains modules for both the pHAT and wHAT, load the Inky pHAT one as follows:

Can you try manually importing the Inky library and instantiating it in src/display_manager.py:
```
 import os
-from inky.auto import auto
+from inky.inky_ac073tc1a import Inky
 from utils.image_utils import resize_image, change_orientation
 from plugins.plugin_registry import get_plugin_instance

@@ -8,7 +8,7 @@ class DisplayManager:
     def __init__(self, device_config):
         """Manages the display and rendering of images."""
         self.device_config = device_config
-        self.inky_display = auto()
+        self.inky_display = Inky()
```

Then restart the inkypi service:
```
sudo systemctl restart inkypi.service
```

fatihak, 2025-03-28:
@skynetsysadmin I saw your comment in https://github.com/fatihak/InkyPi/issues/70, is this Issue good to close? I'll add a section in the troubleshooting docs for the workaround.

skynetsysadmin, 2025-03-28:
Yes, sorry I just saw that I added that to 70 and not 67 as intended. 67
can be closed. Thank you.

fatihak, 2025-03-29:
Added a section in the troubleshooting docs for this error: https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md#no-eeprom-detected
Closing this out.

closed 2025-03-29 by fatihak