fatihak/InkyPi #68: Fresh install on RPi Zero2W

myitguy78, 2025-03-24:
**Describe the bug**
After installing a fresh install of Raspian 64bit, cloning the repo and then installing via script, the Python install appears to hang 2-3 times and then on the 4th, it appears to install. After the reboot, I cannot access the web interface

**Expected behavior**
I would expect the system to reboot, and then I can access via the web interface to configure

**To Reproduce**
Steps to reproduce the behavior:
1. Install Raspberry Pi OS (64 bit, GUI)
2. Clone repo as per instructions
3. Run install as per instructions
4. Reboot after python install

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
RPi Zero 2W

[image]

Treyarch, 2025-03-24:
I had the same problem with my install, it seems broken at the moment. 
I ran the script and it finished the install with no errors but upon reboot I get an error when looking though the logs. I had no web interface and no display on the screen.
I figured out that all the python dependencies did not install in the virtual environment created. I had to manually run the requirements.txt to install the depencencies to get it working and then it ran fine.
There are steps in the documentation / troubleshouting section to activate the virtual environment and then just pip install -r requirements.txt 
It worked for me so I'm just giving my feedback from 2 days ago, it might help.

fatihak, 2025-03-24:
As @Treyarch suggested, can you try manually installing the python dependencies following the instructions here: [Pip Installation Error](https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md#pip-installation-error)

n0-ind3x, 2025-03-26:
i installed pi lite. no error during install. after rebooting display dud not change. tried installing requirements as suggested.. just to see.. all already satisfied. service error. 

[image]

fatihak, 2025-05-13:
@n0-ind3x are you still having this issue?

If so can you try uninstalling and reinstalling pillow:
`sudo su
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip uninstall Pillow
pip install Pillow
deactivate`

myitguy78, 2025-05-13:
Yes, the issue still persists. It looks like there might be a new release
however, and I plan to install that soon. I will also try your
recommendation.

fatihak, 2025-05-21:
@n0-ind3x Can you flash the full Pi OS Bookworm Desktop, not the Lite version.

closed 2025-06-22 by fatihak