fatihak/InkyPi #70: Install error setuptools.errors.InvalidConfigError

csabibela, 2025-03-24:
I tried running the install.sh script on a clean Raspberry Pi OS. I received the following error message:

rpi@pic:~/InkyPi $ sudo bash install/install.sh
Checking if inkypi.service is running
        inkypi.service not running [✔]
Enabling interfaces required for inkypi
        SPI Interface has been enabled. [✔]
        I2C Interface has been enabled. [✔]
Installing system dependencies.  [✔]
Installing inkypi to /usr/local/inkypi
        Removing existing installation found at /usr/local/inkypi [✔]
        Creating symlink from /home/rpi/InkyPi/install/../src to /usr/local/inkypi/src [✔]
Creating python virtual environment.
        Installing python dependencies.  [|]   error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      Traceback (most recent call last):
        File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 334, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 304, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 320, in run_setup
          exec(code, locals())
        File "<string>", line 26, in <module>
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 116, in setup
          _install_setup_requires(attrs)
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 87, in _install_setup_requires
          dist.parse_config_files(ignore_option_errors=True)
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 730, in parse_config_files
          self._parse_config_files(filenames=inifiles)
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 599, in _parse_config_files
          opt = self._enforce_underscore(opt, section)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-cs8uhugv/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 629, in _enforce_underscore
          raise InvalidConfigError(
      setuptools.errors.InvalidConfigError: Invalid dash-separated key 'description-file' in 'metadata' (setup.cfg), please use the underscore name 'description_file' instead.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
        Installing python dependencies.  [✔]
Adding executable to /usr/local/bin/inkypi
Copying config files to /home/rpi/InkyPi/install/../src/config
        device.json already exists in /home/rpi/InkyPi/install/../src/config [✔]
Installing inkypi systemd service.
INKYPI Installation Complete! [✔]
[•] A reboot of your Raspberry Pi is required for the changes to take effect
[•] After your Pi is rebooted, you can access the web UI by going to 'pic.local' or '192.168.xxx.xxx' in your browser.
[•] If you encounter any issues or have suggestions, please submit them here: https://github.com/fatihak/InkyPi/issues
Would you like to restart your Raspberry Pi now? [Y/N] ^C

gth758y, 2025-03-24:
I ran into the same error on a clean install this morning.

csabibela, 2025-03-24:
Workaround:

mkdir /tmp/spidev
curl -o /tmp/spidev/spidev-3.7.tar.gz https://github.com/doceme/py-spidev/archive/refs/tags/v3.7.tar.gz
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip install spidev -f file:///tmp/spidev/
pip install -r install/requirements.txt
systemctl start inkypi.service

fatihak, 2025-03-24:
It looks like this issue stems from a breaking change in setuptools introduced today, which disrupted the Python library ecosystem but has since been reverted: https://github.com/pypa/setuptools/issues/4910
Are you still seeing this issue on a fresh install? If so, can you try activating the venv and upgrading setuptools:
```bash
sudo su
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip install --upgrade setuptools
pip install -r install/requirements.txt
deactivate
```

gth758y, 2025-03-25:
I tried another fresh install a bit ago and did not encounter this error. Thanks!

skynetsysadmin, 2025-03-25:
OK, I was able to fix this thanks to a couple of suggestions from @fatihak and @scotthsieh0503. Basically, the auto detect doesn't work with this particular board, so the display_manager.py needed a couple of modifications.

At the top of the script, instead of 'auto', it needed a specific board name, 'inky_ac073tc1a'. Also, changed the import variable from auto to Inky. Techniocally, you could probably keep the import auto and it would still work. But, the change is clearer if you use a new import name.

```
 import os
 from inky.inky_ac073tc1a import Inky <--- changed this line
 from utils.image_utils import resize_image, change_orientation
 from plugins.plugin_registry import get_plugin_instance
```

In a section of code a little lower in the display_manager.py, I modified one line.

From:
```
        self.inky_display = auto()
```

To

```
        self.inky_display = Inky()
```

Then I just restarted the inky.pi service. That woke up the board, and I can also access the web interface. Everything seems to be working fine now. Thank you to both @fatihak and @scotthsieh0503 for your help.

fatihak, 2025-03-28:
@csabibela are you still encountering this error? If not going to close this issue.

fatihak, 2025-04-10:
Closing this since I didnt hear back & since this was related to the setuptools issue which was resolved. Feel free to comment or reopen the issue if you're still seeing the same error

closed 2025-04-10 by fatihak