fatihak/InkyPi #77: temperature line in weather plugin

stevenidavis, 2025-03-28:
**Describe the bug**
The bottom trend line for the temperature does not seem to be scaled correctly. The range seems about right for the expected highs and lows but the line never goes above the bottom 10%

**Expected behavior**
The trend line would go from the top to the bottom of the range.

**To Reproduce**
run with weather graph on 3 day forecast

**Logs**
Provide the InkyPi service logs, see [troubleshooting.md](./docs/troubleshooting.md) for instructions on how to view logs.

**Screenshots**

[image]

**Additional context**
Raspberry Pi 4 with InkyImpression 5.7 running Raspberry Pi OS 64-bit

fatihak, 2025-03-29:
This was intentional. When the difference between the max and min temperatures was small, I added an offset to improve the appearance of the line graph. However, I've just updated the code to reduce this offset, so it _should_ scale better. Feel free to pull the latest changes and give it a try!
```
cd InkyPi
git pull
sudo systemctl restart inkypi.service
```

cooperep1, 2025-04-08:
I built this project a couple days ago, but I used a Raspberry Pi 5. The weather line is doing the same thing as the OP.  Am I doing something wrong?  Everything else is working beautifully, and I love the project. So cool!

fatihak, 2025-05-21:
@stevenidavis @cooperep1 Just pushed a fix that should resolve this. Can you pull the latest changes and see if it looks good on your end?

cooperep1, 2025-05-22:
> > [@stevenidavis](https://github.com/stevenidavis) [@cooperep1](https://github.com/cooperep1) Just pushed a fix that should resolve this. Can you pull the latest changes and see if it looks good on your end?
> 
> I will try it tonight after work. Thank you very much.

Thank you so much. It is working perfectly for me now!

fatihak, 2025-05-22:
Awesome, closing this issue!

closed 2025-05-22 by fatihak