fatihak/InkyPi #86: Bug: Weather plugin screenshoot not working

EnzoReverdi, 2025-04-01:
Failed to take screenshoot on a fresh install because of hardware limitation apparently, Pi zero 2W

`pi@pi:~/project/InkyPi $ journalctl -n 20 -u inkypi.service
Apr 01 09:09:08 pi inkypi[1758]: 09:09:08 - INFO - refresh_task - Active playlist 'Default' has no plugins.
Apr 01 09:09:24 pi inkypi[1758]: 09:09:24 - INFO - refresh_task - Manual update requested
Apr 01 09:09:26 pi inkypi[1758]: 09:09:26 - ERROR - utils.image_utils - Failed to take screenshot:
Apr 01 09:09:26 pi inkypi[1758]: 09:09:26 - ERROR - utils.image_utils - The hardware on this system lacks support for NEON SIMD extensions.
Apr 01 09:09:26 pi inkypi[1758]: We now require NEON or equivalent architecture extensions on ARM-based
Apr 01 09:09:26 pi inkypi[1758]: machines. See https://lists.debian.org/debian-devel/2023/09/msg00175.html
Apr 01 09:09:26 pi inkypi[1758]: for more information.
Apr 01 09:09:26 pi inkypi[1758]: 09:09:26 - ERROR - root - Exception during refresh
Apr 01 09:09:26 pi inkypi[1758]: Traceback (most recent call last):
Apr 01 09:09:26 pi inkypi[1758]:   File "/home/pi/project/InkyPi/src/refresh_task.py", line 106, in _run
Apr 01 09:09:26 pi inkypi[1758]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Apr 01 09:09:26 pi inkypi[1758]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 01 09:09:26 pi inkypi[1758]:   File "/home/pi/project/InkyPi/src/refresh_task.py", line 206, in execute
Apr 01 09:09:26 pi inkypi[1758]:     return plugin.generate_image(self.plugin_settings, device_config)
Apr 01 09:09:26 pi inkypi[1758]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 01 09:09:26 pi inkypi[1758]:   File "/home/pi/project/InkyPi/src/plugins/weather/weather.py", line 76, in generate_image
Apr 01 09:09:26 pi inkypi[1758]:     raise RuntimeError("Failed to take screenshot, please check logs.")
Apr 01 09:09:26 pi inkypi[1758]: RuntimeError: Failed to take screenshot, please check logs.`

fatihak, 2025-04-04:
Which OS do you have installed on your Pi?

danielboos1, 2025-04-04:
Same issue - Raspberrypi lite 64bit
journalctl -u inkypi --no-pager --lines=50
Apr 05 02:30:52 raspberrypi inkypi[580]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Apr 05 02:30:52 raspberrypi inkypi[580]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 05 02:30:52 raspberrypi inkypi[580]:   File "/home/danie/InkyPi/src/refresh_task.py", line 206, in execute
Apr 05 02:30:52 raspberrypi inkypi[580]:     return plugin.generate_image(self.plugin_settings, device_config)
Apr 05 02:30:52 raspberrypi inkypi[580]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 05 02:30:52 raspberrypi inkypi[580]:   File "/home/danie/InkyPi/src/plugins/weather/weather.py", line 58, in generate_image
Apr 05 02:30:52 raspberrypi inkypi[580]:     weather_data = self.get_weather_data(api_key, units, lat, long)
Apr 05 02:30:52 raspberrypi inkypi[580]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 05 02:30:52 raspberrypi inkypi[580]:   File "/home/danie/InkyPi/src/plugins/weather/weather.py", line 197, in get_weather_data
Apr 05 02:30:52 raspberrypi inkypi[580]:     raise RuntimeError("Failed to retrieve weather data.")
Apr 05 02:30:52 raspberrypi inkypi[580]: RuntimeError: Failed to retrieve weather data.
Apr 05 02:30:52 raspberrypi inkypi[580]: 02:30:52 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:30:52] "POST /update_now HTTP/1.1" 500 -
Apr 05 02:30:56 raspberrypi inkypi[580]: 02:30:56 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:30:56] "GET /static/styles/main.css HTTP/1.1" 304 -
Apr 05 02:31:06 raspberrypi inkypi[580]: 02:31:06 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:06] "GET /plugin/weather HTTP/1.1" 200 -
Apr 05 02:31:06 raspberrypi inkypi[580]: 02:31:06 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:06] "GET /static/styles/main.css HTTP/1.1" 304 -
Apr 05 02:31:06 raspberrypi inkypi[580]: 02:31:06 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:06] "GET /static/scripts/response_modal.js HTTP/1.1" 304 -
Apr 05 02:31:06 raspberrypi inkypi[580]: 02:31:06 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:06] "GET /images/weather/icon.png HTTP/1.1" 304 -
Apr 05 02:31:06 raspberrypi inkypi[580]: 02:31:06 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:06] "GET /images/base_plugin/frames/blank.png HTTP/1.1" 304 -
Apr 05 02:31:06 raspberrypi inkypi[580]: 02:31:06 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:06] "GET /images/base_plugin/frames/corner.png HTTP/1.1" 304 -
Apr 05 02:31:07 raspberrypi inkypi[580]: 02:31:07 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:07] "GET /images/base_plugin/frames/top_and_bottom.png HTTP/1.1" 304 -
Apr 05 02:31:07 raspberrypi inkypi[580]: 02:31:07 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:31:07] "GET /images/base_plugin/frames/rectangle.png HTTP/1.1" 304 -
Apr 05 02:31:52 raspberrypi inkypi[580]: 02:31:52 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-04-05 02:31:52
Apr 05 02:31:52 raspberrypi inkypi[580]: 02:31:52 - INFO - refresh_task - Active playlist 'Default' has no plugins.
Apr 05 02:32:41 raspberrypi inkypi[580]: 02:32:41 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:41] "GET /plugin/weather HTTP/1.1" 200 -
Apr 05 02:32:41 raspberrypi inkypi[580]: 02:32:41 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:41] "GET /static/styles/main.css HTTP/1.1" 304 -
Apr 05 02:32:42 raspberrypi inkypi[580]: 02:32:42 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:42] "GET /static/scripts/response_modal.js HTTP/1.1" 304 -
Apr 05 02:32:42 raspberrypi inkypi[580]: 02:32:42 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:42] "GET /images/weather/icon.png HTTP/1.1" 304 -
Apr 05 02:32:42 raspberrypi inkypi[580]: 02:32:42 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:42] "GET /images/base_plugin/frames/blank.png HTTP/1.1" 304 -
Apr 05 02:32:42 raspberrypi inkypi[580]: 02:32:42 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:42] "GET /images/base_plugin/frames/corner.png HTTP/1.1" 304 -
Apr 05 02:32:42 raspberrypi inkypi[580]: 02:32:42 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:42] "GET /images/base_plugin/frames/top_and_bottom.png HTTP/1.1" 304 -
Apr 05 02:32:42 raspberrypi inkypi[580]: 02:32:42 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:42] "GET /images/base_plugin/frames/rectangle.png HTTP/1.1" 304 -
Apr 05 02:32:44 raspberrypi inkypi[580]: 02:32:44 - INFO - refresh_task - Manual update requested
Apr 05 02:32:45 raspberrypi inkypi[580]: 02:32:45 - ERROR - root - Failed to retrieve weather data: b'{"cod":401, "message": "Please note that using One Call 3.0 requires a separate subscription to the One Call by Call plan. Learn more here https://openweathermap.org/price. If you have a valid subscription to the One Call by Call plan, but still receive this error, then please see https://openweathermap.org/faq#error401 for more info."}'
Apr 05 02:32:45 raspberrypi inkypi[580]: 02:32:45 - ERROR - root - Exception during refresh
Apr 05 02:32:45 raspberrypi inkypi[580]: Traceback (most recent call last):
Apr 05 02:32:45 raspberrypi inkypi[580]:   File "/home/danie/InkyPi/src/refresh_task.py", line 106, in _run
Apr 05 02:32:45 raspberrypi inkypi[580]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
[... cut ...]
Apr 05 02:32:45 raspberrypi inkypi[580]:     raise RuntimeError("Failed to retrieve weather data.")
Apr 05 02:32:45 raspberrypi inkypi[580]: RuntimeError: Failed to retrieve weather data.
Apr 05 02:32:45 raspberrypi inkypi[580]: 02:32:45 - INFO - werkzeug - 192.168.1.83 - - [05/Apr/2025 02:32:45] "POST /update_now HTTP/1.1" 500 -
Apr 05 02:33:45 raspberrypi inkypi[580]: 02:33:45 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-04-05 02:33:45
Apr 05 02:33:45 raspberrypi inkypi[580]: 02:33:45 - INFO - refresh_task - Active playlist 'Default' has no plugins.

victor-h-costa-m, 2025-04-07:
Hello, I also have the weather plugin being unable to screenshot on the Pi Zero 2W. When running "journalctl -u inkypi -n 100", I get this:

`22:04:59 - ERROR - utils.image_utils - Failed to take screenshot:
Apr 07 22:04:59 raspberrypi inkypi[2038]: 22:04:59 - ERROR - utils.image_utils - Old Headless mode has been removed from the Chrome binary. Please use the new Headless mode (https://developer.chrome.com/docs/chromium/new-headless) or the chrome-headless-shell which is a standalone implementation of the old Headless mode (https://developer.chrome.com/blog/chrome-headless-shell).
Apr 07 22:05:00 raspberrypi inkypi[2038]: 22:05:00 - ERROR - root - Exception during refresh
Apr 07 22:05:00 raspberrypi inkypi[2038]: Traceback (most recent call last):
Apr 07 22:05:00 raspberrypi inkypi[2038]:   File "/home/user/InkyPi/src/refresh_task.py", line 106, in _run
Apr 07 22:05:00 raspberrypi inkypi[2038]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Apr 07 22:05:00 raspberrypi inkypi[2038]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 07 22:05:00 raspberrypi inkypi[2038]:   File "/home/user/InkyPi/src/refresh_task.py", line 250, in execute
Apr 07 22:05:00 raspberrypi inkypi[2038]:     image = plugin.generate_image(self.plugin_instance.settings, device_config)
Apr 07 22:05:00 raspberrypi inkypi[2038]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 07 22:05:00 raspberrypi inkypi[2038]:   File "/home/user/InkyPi/src/plugins/weather/weather.py", line 75, in generate_image
Apr 07 22:05:00 raspberrypi inkypi[2038]:     raise RuntimeError("Failed to take screenshot, please check logs.")
Apr 07 22:05:00 raspberrypi inkypi[2038]: RuntimeError: Failed to take screenshot, please check logs.`

I am on Linux raspberrypi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux.

MikeBinvt, 2025-04-08:
I am getting the same Headless shell error on a brand new install using Raspberry pi Debian version 12.10

`Apr 07 20:58:47 weatherpi inkypi[2113]: 20:58:47 - INFO - refresh_task - Manual update requested
Apr 07 20:59:23 weatherpi inkypi[2113]: 20:59:23 - ERROR - utils.image_utils - Failed to take screenshot:
Apr 07 20:59:23 weatherpi inkypi[2113]: 20:59:23 - ERROR - utils.image_utils - Old Headless mode has been removed from the Chrome binary. Please use the new Headless mode (https://developer.chrome.com/docs/chromium/new-headless) or the chrome-headless-shell which is a standalone implementation of the old Headless mode (https://developer.chrome.com/blog/chrome-headless-shell).
Apr 07 20:59:24 weatherpi inkypi[2113]: 20:59:24 - ERROR - root - Exception during refresh
Apr 07 20:59:24 weatherpi inkypi[2113]: Traceback (most recent call last):
Apr 07 20:59:24 weatherpi inkypi[2113]:   File "/home/mike/InkyPi/src/refresh_task.py", line 106, in _run
Apr 07 20:59:24 weatherpi inkypi[2113]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Apr 07 20:59:24 weatherpi inkypi[2113]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 07 20:59:24 weatherpi inkypi[2113]:   File "/home/mike/InkyPi/src/refresh_task.py", line 206, in execute
Apr 07 20:59:24 weatherpi inkypi[2113]:     return plugin.generate_image(self.plugin_settings, device_config)
Apr 07 20:59:24 weatherpi inkypi[2113]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 07 20:59:24 weatherpi inkypi[2113]:   File "/home/mike/InkyPi/src/plugins/weather/weather.py", line 75, in generate_image
Apr 07 20:59:24 weatherpi inkypi[2113]:     raise RuntimeError("Failed to take screenshot, please check logs.")
Apr 07 20:59:24 weatherpi inkypi[2113]: RuntimeError: Failed to take screenshot, please check logs.
`

CGives, 2025-04-08:
after updating my pi with sudo apt update && sudo apt upgrade -y yesterday I am now getting the same headless shell error

victor-h-costa-m, 2025-04-08:
I'll try to revert to an older version of Chromium when I get home. I think it might be the latest version that has changed how the headless shell works ("Old Headless mode has been removed from the Chrome binary. Please use the new Headless"). The move to the new headless mode should be implemented though, as I am guessing this issue will become more prominent when people update chromium.

robertofalk, 2025-04-08:
replacing `--headless=old` with --headless=new` [here](https://github.com/fatihak/InkyPi/blob/main/src/utils/image_utils.py#L91C42-L91C56) solved the problem in my tests.

HRB-YuPai, 2025-04-08:
I got the same error:
OS

> replacing `--headless=old` with `--headless=new` [here](https://github.com/fatihak/InkyPi/blob/main/src/utils/image_utils.py#L91C42-L91C56) solved the problem in my tests.

I will have a try now haha

MikeBinvt, 2025-04-09:
> replacing `--headless=old` with `--headless=new` [here](https://github.com/fatihak/InkyPi/blob/main/src/utils/image_utils.py#L91C42-L91C56) solved the problem in my tests.

That fixed it for me as well!  Thanks

HRB-YuPai, 2025-04-09:
there is something wrong on my side. When I making the screenshotm it didn't detect the correct resolution of my screen. 1/5 of whe area is wasted. how could I fix it?

fatihak, 2025-04-09:
Hey everyone, I was able to reproduce this error after upgrading my chromium-browser version.

As part of a recent change, Chrome removed the "old" Headless mode and introduced a new, standalone chrome-headless-shell binary for use cases like automation and screenshotting, which is ideal for this project. Here is the [official announcement](https://developer.chrome.com/blog/chrome-headless-shell)

Unfortunately, chrome-headless-shell is not available for Linux ARM64 which makes it unusable for this project. You can track the related issues here:
- https://github.com/GoogleChromeLabs/chrome-for-testing/issues/1
- https://issues.chromium.org/issues/374811603
Feel free to upvote or react to these to help signal interest.

I tried using the new headless mode via the main chrome binary but there seems to be a bug where the bottom ~95 pixels of the screenshot come out blank (screenshot below). This was the reason I was sticking with the old headless mode in the first place.

At this point, I’m not sure of the best workaround. One option is to oversize the screenshot and crop it manually, but that feels pretty hacky. Open to any suggestions.

[image]

HRB-YuPai, 2025-04-09:
> Hey everyone, I was able to reproduce this error after upgrading my chromium-browser version.
> 
> As part of a recent change, Chrome removed the "old" Headless mode and introduced a new, standalone chrome-headless-shell binary for use cases like automation and screenshotting, which is ideal for this project. Here is the [official announcement](https://developer.chrome.com/blog/chrome-headless-shell)
> 
> Unfortunately, chrome-headless-shell is not available for Linux ARM64 which makes it unusable for this project. You can track the related issues here:
> 
> * [Linux ARM64 support? GoogleChromeLabs/chrome-for-testing#1](https://github.com/GoogleChromeLabs/chrome-for-testing/issues/1)
> * https://issues.chromium.org/issues/374811603
>   Feel free to upvote or react to these to help signal interest.
> 
> I tried using the new headless mode via the main chrome binary but there seems to be a bug where the bottom ~95 pixels of the screenshot come out blank (screenshot below). This was the reason I was sticking with the old headless mode in the first place.
> 
> At this point, I’m not sure of the best workaround. One option is to oversize the screenshot and crop it manually, but that feels pretty hacky. Open to any suggestions.
> 
> [image]

But I feel curious, does the Newspaper screenshot use the same method as the weather? Becayse the Newspaprt works fine , resolution is correct.

HRB-YuPai, 2025-04-09:
Then I have a not good solution to fix this issue temporarily wnem you finding the better solution ---  revert to an older version of chromium which supports the "--headless=old", but I dont know the version number

HRB-YuPai, 2025-04-09:
If anyone fix this issue by downgrading chromium, please let me know the build number, in appreciation

Jorkim, 2025-04-10:
Interestingly, i do not have this problem at all. But i have to say i am using the fork from IndexOutOfMJ. Everything is up to date.

HRB-YuPai, 2025-04-10:
> Interestingly, i do not have this problem at all. But i have to say i am using the fork from IndexOutOfMJ. Everything is up to date.

can you please give me a git link?

HRB-YuPai, 2025-04-10:
> Hey everyone, I was able to reproduce this error after upgrading my chromium-browser version.
> 
> As part of a recent change, Chrome removed the "old" Headless mode and introduced a new, standalone chrome-headless-shell binary for use cases like automation and screenshotting, which is ideal for this project. Here is the [official announcement](https://developer.chrome.com/blog/chrome-headless-shell)
> 
> Unfortunately, chrome-headless-shell is not available for Linux ARM64 which makes it unusable for this project. You can track the related issues here:
> 
> * [Linux ARM64 support? GoogleChromeLabs/chrome-for-testing#1](https://github.com/GoogleChromeLabs/chrome-for-testing/issues/1)
> * https://issues.chromium.org/issues/374811603
>   Feel free to upvote or react to these to help signal interest.
> 
> I tried using the new headless mode via the main chrome binary but there seems to be a bug where the bottom ~95 pixels of the screenshot come out blank (screenshot below). This was the reason I was sticking with the old headless mode in the first place.
> 
> At this point, I’m not sure of the best workaround. One option is to oversize the screenshot and crop it manually, but that feels pretty hacky. Open to any suggestions.
> 
> Edit: Found an open issue for this bug, please vote so it gets prioritized: https://issues.chromium.org/issues/405165895
> 
> [image]

Is chromium the only option for this? how about use the firefox?

fatihak, 2025-04-10:
> Interestingly, i do not have this problem at all. But i have to say i am using the fork from IndexOutOfMJ. Everything is up to date.

You will likely run this into this error if you run "sudo apt upgrade chromium-browser" which will update chromium-browser. IndexOutOfMJ's fork is using the same logic

HRB-YuPai, 2025-04-10:
> You will likely run this into this error if you run "sudo apt upgrade chromium-browser" which will update chromium-browser. IndexOutOfMJ's fork is using the same logic

Where is the LOG  of this project?

HRB-YuPai, 2025-04-11:
[image]

Temporary fix for this bug:

The image_utils.py file in /InkyPi/src/utils/

at line 89, add a new variable: 

new_dimension_1 = dimensions[1] + (dimensions[1] * 0.2)

Then fix the command:

command = [
            "chromium-browser", target, "--headless=new",
            f"--screenshot={img_file_path}", f'--window-size={dimensions[0]},{new_dimension_1}',
            "--no-sandbox", "--disable-gpu", "--disable-software-rasterizer",
            "--disable-dev-shm-usage", "--hide-scrollbars"
        ]

in --window-size   replace the original  dimensions[1]  to  new_dimension_1 

After all fix, save and update the file to the Pi

Restart the service or reboot

skynetsysadmin, 2025-04-11:
> [image]
> 
> Temporary fix for this bug:
> 
> The image_utils.py file in /InkyPi/src/utils/
> 
> at line 89 (in the take_screenshot function), add a new variable:
> 
> new_dimension_1 = dimensions[1] + (dimensions[1] * 0.2)
> 
> if 0.2 is too stick to the bottom edge, you may change it to 0.15, may be better
> 
> Then fix the command:
> 
> command = [ "chromium-browser", target, "--headless=new", f"--screenshot={img_file_path}", f'--window-size={dimensions[0]},{new_dimension_1}', "--no-sandbox", "--disable-gpu", "--disable-software-rasterizer", "--disable-dev-shm-usage", "--hide-scrollbars" ]
> 
> in --window-size replace the original dimensions[1] to new_dimension_1
> 
> After all fix, save and update the file to the Pi
> 
> Restart the service or reboot

I had a similar thought yesterday. But, in my case, it made the issue worse, not better. I have a 7.3" 7 color 800x480 display. 

Setting it to 800x576 (.2 multiplier) looks like this:

[image]

800x552 (.15 multiplier) looks like this:

[image]

I imagine that 800x525 might be slightly better than the native 800x480, but it will still have a large gap at the bottom.

HRB-YuPai, 2025-04-11:
this is just a temporary fix, better than before id enough for now

HRB-YuPai, 2025-04-11:
My screen is waveshare 800x448, this is fine. you can play with the that two variables according to your hardware to find a better solution for yourself. the operation is the same as dimension1

HRB-YuPai, 2025-04-11:
interesting thing is i tried firefox and htmltopdf both of them are failed

cyberdudePete, 2025-04-12:
Hi all, not much of a programmer, and for some reason, having trouble with syntax with line "new_dimension_1 = dimensions[1] + (dimensions[1] * 0.2)"

I keep getting journal error about incorrect use of tabs and indents.

I have moved it in and out but still get same error.

Can you please screenshot the "Take_screenshot" function so I can see proper placement?

Would be very appreciated.

 
[image]

stemow, 2025-04-12:
> Hi all, not much of a programmer, and for some reason, having trouble with syntax with line "new_dimension_1 = dimensions[1] + (dimensions[1] * 0.2)"
> 
> I keep getting journal error about incorrect use of tabs and indents.
> 
> I have moved it in and out but still get same error.
> 
> Can you please screenshot the "Take_screenshot" function so I can see proper placement?
> 
> Would be very appreciated.
> 
> <img alt="Image" width="711" src="https://private-user-images.githubusercontent.com/2941454/433059210-8cff27a3-b2ef-4ce3-a472-2803cc151213.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQ0ODY0MjYsIm5iZiI6MTc0NDQ4NjEyNiwicGF0aCI6Ii8yOTQxNDU0LzQzMzA1OTIxMC04Y2ZmMjdhMy1iMmVmLTRjZTMtYTQ3Mi0yODAzY2MxNTEyMTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDQxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA0MTJUMTkyODQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGNiOTAxNDRkMTdlMmU1MzRhMDUyYzQxZjYxYWJkN2EyYWU4ODA3YTUzYTJiYjAyYWQxYzY2ZDYyN2Y0MzA3YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.1UtwYE2_6vIgav53O8zllMdqsFFOcKOE2dLnin276fA">

I am also not a programmer, very much a n00b. I got the same error when I used a tab indent before “new_dimension_1 = dimensions[1] + (dimensions[1] * 0.2)”

I fixed by using a bunch of spaces.

HRB-YuPai, 2025-04-12:
> I am also not a programmer, very much a n00b. I got the same error when I used a tab indent before “new_dimension_1 = dimensions[1] + (dimensions[1] * 0.2)”
>
> I fixed by using a bunch of spaces.

https://file-download.loneonegio.com:52471/image_utils.py

SInce your image is broken, I cannot see the error. Simply replace the file I am currenty using, dont forget to change the permissions.

HRB-YuPai, 2025-04-12:
> Hi all, not much of a programmer, and for some reason, having trouble with syntax with line "new_dimension_1 = dimensions[1] + (dimensions[1] * 0.2)"
>
> I keep getting journal error about incorrect [...]

you have an extra ] there at the second new_dimension_1 

if you don't know how to fix, simply replace the file i provided.

https://file-download.loneonegio.com:52471/image_utils.py

fatihak, 2025-04-13:
Hey everyone,

After some investigation, I found that chromium-headless-shell (the legacy headless mode as a standalone package) is available as a Debian package. I’ve updated the code to use this instead and was able to see it working.

Please pull the latest changes and let me know if the issue is resolved. You’ll need to run the update script (see [instructions here](https://github.com/fatihak/InkyPi/tree/main?tab=readme-ov-file#update), which will install the new package and reboot your Pi afterwards.

Thanks for your patience!

skynetsysadmin, 2025-04-14:
Yea! Full screen again. Thanks!

hornett-69, 2025-05-07:
Fresh install today on raspi 64bit on Pi4 with 7.3 screen... getting "Error! An error occurred: Failed to take screenshot, please check logs."

[image]

EABonney73, 2025-05-07:
I am getting the same error of failed to take screenshot, please check logs. This is on a fresh install as of today using an impression73, the newest version of the eink display.

[image]

EABonney73, 2025-05-08:
So I was able to fix this error today by manually installing the chromium-headless-shell with apt on my Pi.

sudo apt install chromium-headless-shell

then I just restarted InkyPi and I was able to set the location and add it to my Playlist and updated the display just fine.

Thought I would drop a note here as it doesn't look like doing a fresh install of the OS and InkyPi is enough, you appear to have to install the shell manually.

fatihak, 2025-06-05:
@hornett-69 are you still seeing the error?

fatihak, 2025-06-05:
Closing this out based on the responses after the fix, feel free to open a new issue if you're still having issues!