fatihak/InkyPi #98: Error running update script - RPi Zero 2W on Bookworm

spspspspspsp, 2025-04-15:
**Describe the bug**
Update script throws errors after an initially successful install. Tried to update 14 April 2015.

**Expected behavior**
Update normally.

**To Reproduce**
Steps to reproduce the behavior:
> cd InkyPi
> git pull
> sudo bash install/update.sh

**Logs**
> sudo bash install/update.sh
Get:1 http://archive.raspberrypi.com/debian bookworm InRelease [39.1 kB]
Get:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [535 kB]
Get:4 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages [14.5 MB]
Get:5 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [534 kB]
Fetched 15.6 MB in 46s (337 kB/s)
Reading package lists... Done
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Installing system dependencies.
Updating Python dependencies...
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pytz/
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/cli/base_command.py", line 160, in exc_logging_wrapper
    status = run_func(*args)
             ^^^^^^^^^^^^^^^
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/cli/req_command.py", line 247, in wrapper
    return func(self, options, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/commands/install.py", line 507, in run
    installed = install_given_reqs(
                ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/req/__init__.py", line 73, in install_given_reqs
    requirement.install(
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/req/req_install.py", line 796, in install
    install_wheel(
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/operations/install/wheel.py", line 729, in install_wheel
    _install_wheel(
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/operations/install/wheel.py", line 589, in _install_wheel
    file.save()
  File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/pip/_internal/operations/install/wheel.py", line 388, in save
    shutil.copyfileobj(f, dest)
  File "/usr/lib/python3.11/shutil.py", line 197, in copyfileobj
    buf = fsrc_read(length)
          ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/zipfile.py", line 954, in read
    data = self._read1(n)
           ^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/zipfile.py", line 1030, in _read1
    data = self._decompressor.decompress(data, n)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
zlib.error: Error -3 while decompressing data: invalid distance too far back
Restarting inkypi service.
Update completed. [✔]

**Additional context**
Running on RPi Zero 2W, using Bookworm.

fatihak, 2025-05-21:
I pushed a change to improve the update script. Can you pull the latest changes and see if you're still running into this? Also, do you have the Pi OS Bookwork Desktop or Lite Version?

spspspspspsp, 2025-05-21:
> I pushed a change to improve the update script. Can you pull the latest changes and see if you're still running into this? Also, do you have the Pi OS Bookworm Desktop or Lite Version?

I'll try checking soon. Thanks. I believe I'm running the "lite"
version of Bookworm.

fatihak, 2025-05-21:
That might be the issue. Can you flash the desktop version instead? Others have indicated that resolved their issues

alex-tro, 2025-06-11:
Good hint with the Lite / Desktop version.
I was on the Lite version and “sudo bash install/update.sh” got stuck with the message “**Installing system dependencies...**”

I have installed components from the desktop version with these commands (without guarantee):
`sudo apt update && sudo apt upgrade -y`
`sudo apt install --no-install-recommends raspberrypi-ui-mods lxsession lightdm -y`

The update then went through.

Reinstalling the desktop version is certainly the safest option.

phattmatt, 2025-06-22:
I suggest closing this issue as resolved.

The original problem seems related to PIP issues rather than InkyPi, and OP has not updated with any further issues.

closed 2025-06-22 by fatihak

fatihak, 2025-06-22:
@phattmatt thanks, feel free to close issues if you feel like they're resolved. I believe you should have the permissions for it