fatihak/InkyPi #104: Playlists never refresh – refresh_task always outputs “Not time to update display”

saulob, 2025-04-28:
**Describe the bug**
After adding one or more items to a playlist, the screen never updates.
The scheduler keeps logging INFO – refresh_task – Not time to update display even though the Refresh interval (e.g. “Every 2 minutes”) has already elapsed.

**Expected behavior**
Each plugin in the playlist should display on the e-paper at its configured interval.

**To Reproduce**
1 Open the InkyPi web UI.
2 Add a Clock plugin (“Every 2 minutes”) → Add to Playlist.
3 (Optional) add additional plugins (“teste”, “teste2”).
3 Wait ≥ 2 minutes
3 Observe the screen & logs – nothing changes; log shows “Not time to update display”.

**Logs**
Apr 28 14:14:13 inkypi[681]: INFO – refresh_task – Running interval refresh check. | current_time: 2025-04-28 14:14:13
Apr 28 14:14:13 inkypi[681]: DEBUG – config – Getting key plugin_cycle_interval_seconds from config.
Apr 28 14:14:13 inkypi[681]: INFO – refresh_task – Not time to update display. | latest_update: 2025-04-28 14:06:21 | plugin_cycle_interval: 3600

**Screenshots**
[image]
[image]
[image]

1 Service log with repeated Not time to update display message.
2 Clock plugin set to refresh every 2 minutes.
3 Playlist containing three clock items.

**Additional context**
Pi model: Raspberry Pi Zero 2 W
Display: Waveshare 7.3-inch e-Paper (E) epd7in3e
InkyPi version: acscpt/InkyPi-WS fork – branch develop (clean install)
Install cmd: sudo bash install/install.sh -W epd7in3e
Timezone: America/São_Paulo
plugin_cycle_interval - 120 s (2 min) in UI

I have reproduced with 1, 2 and 3 playlist items; behaviour is the same.
Let me know if I can provide any other info – thanks!

tuftsanti, 2025-05-08:
I thought I was the only one, or had set it up incorrectly. Thanks for creating this.

fatihak, 2025-05-13:
Whats the "Plugin Cycle Interval" set to in your device settings? That defines the each plugin is displayed before cycling regardless of the refresh intervals of individual plugins. You can find more details [here](https://github.com/fatihak/InkyPi/discussions/41).

fatihak, 2025-05-18:
Apologies, I didn't mean to say checkout the Playlists branch as the instructions on the announcement state, just that it details the playlists feature and the different settings. Can you try updating using the instructions here:
https://github.com/fatihak/InkyPi/tree/main?tab=readme-ov-file#update

tuftsanti, 2025-05-18:
> Apologies, I didn't mean to say checkout the Playlists branch as the instructions on the announcement state, just that it details the playlists feature and the different settings. Can you try updating using the instructions here: https://github.com/fatihak/InkyPi/tree/main?tab=readme-ov-file#update

I've been using a Pi Zero W2 until I just threw an old Pi3 in instead. This is much smoother and actually got through the update process. Not sure yet if the playlists will refresh, but will find out sooner than later

tuftsanti, 2025-05-19:
The problem is not reconnecting to a network I've found. I successfully rebuilt to a Pi3 and everything was working in my office, which is 10 ft from my wireless router. When I move it to another room, it never seems to connect to another router which is ~15ft away in that other room. I can't connect to the .local web access until I move it back to the office. What am I missing here?

tuftsanti, 2025-05-23:
I think I've solved my issue at least. In the test lab I have a 5.1V 2.5A PSU. In it's home base I discovered I only have a 5V 0.5A PSU. Upgrading the power supply got me back online with the Pi Zero W 2.

phattmatt, 2025-06-23:
Are there any updates on this issue? Or can it be closed?

phattmatt, 2025-06-23:
Ah, that sounds more like the issue being tracked here: https://github.com/fatihak/InkyPi/issues/159

I'm going to close this issue as:

- The original reporter has not updated for a while 
- Your issue doesn't appear to have the same symptoms as originally reported

Take a look at the other issue, as I believe your symptoms are more closely related to those described there.

tuftsanti, 2025-06-24:
> Ah, that sounds more like the issue being tracked here: [#159](https://github.com/fatihak/InkyPi/issues/159)
>
> I'm going to close this issue as:
>
>     * The original reporter has not updated for a [...]

Why close it, who's it hurting to keep it active? It's not resolved just because OP never responded. I literally have the same problem in this post. I only have similar issues to the issue you linked...

phattmatt, 2025-06-24:
Maybe I've misunderstood.

The original issue raised seems to report the InkyPi service is running, but never decides to update the display, as demonstrated by the log file entry "Not time to update display".

Are you seeing the same symptoms? I.e. The InkyPi service is running, but the log files report it never updates?

Robooto, 2025-07-04:
I had this same issue until I lowered my "Plugin Cycle Interval" to be less than the refresh time on the plugin.

tuftsanti, 2025-07-06:
> I had this same issue until I lowered my "Plugin Cycle Interval" to be less than the refresh time on the plugin.

How did you do that?

tuftsanti, 2025-07-24:
> [@tuftsanti](https://github.com/tuftsanti) You can change the Plugin Cycle Interval from the settings page in the UI. This determines how often the display should refresh, regardless of the refresh settings for a plugin. I also recently merged in some optimizations to help with the Pi crashing, can you pull the latest changes and see if it's still occurring?

Not only did it not fix it, it broke it 😉. I had converted to a RPi3 replacing the Zero 2W, which had run and updated successfully for a few weeks. I did the git pull and rebooted, seemed fine. I switched back to the Zero and no updates. I went back to the Pi3 as it was working before, now that's not working either. I'll have to fiddle with it, but I shouldn't have messed with success I think.

fatihak, 2025-07-29:
The two time settings exist for a reason:

- Plugin refresh time controls when the plugin pulls new data or regenerates an image. Some plugins update once per day (like NASA APOD), others can refresh more frequently. Some even cost money to refresh (like AI-related plugins), so it doesn’t make sense to refresh them constantly.
- Playlist cycle interval determines how long each plugin stays on screen before switching to the next. Without this, there’d be no way to manage how often the screen changes between plugins that refresh at different rates.

If your plugin hasn’t refreshed in 2–3 days when it's supposed to refresh daily, then its likely that there was an error during the refresh. You can try to force an immediate refresh in the playlist page by clicking the "Display Now" button and checking the logs for any errors .

phattmatt, 2025-08-26:
> I’ve been using the “Display Now” button almost every day, because the scheduled refresh times don’t match São Paulo time. Do you know why that happens?
> 
> Is there a way to see what time the system is actually using internally? Even though I selected America/Sao_Paulo (as shown in the screenshots), the playlist still refreshes at the wrong times.
> 
> It would really help if the system could show the current time/timezone after we select it. Right now I have no idea if it’s applying the timezone correctly :(

It might be worth checking the Pi's system timezone:

```
pi@rpi5nvme:~ $ timedatectl status
               Local time: Tue 2025-08-26 13:56:56 BST
           Universal time: Tue 2025-08-26 12:56:56 UTC
                 RTC time: Tue 2025-08-26 12:56:56
                Time zone: Europe/London (BST, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

closed 2025-09-14 by phattmatt