fatihak/InkyPi #106: Selecting Playlist returns Internal Server Error

MatrixRetoastet, 2025-05-01:
Hi,

I'm able to click on every plugin button in the dashboard but when I click on Playlist I'm getting an Internal Server Error (Error 500).
The logs mention the following: `ValueError: day is out of range for month`

Here's the complete log:

> May 01 13:23:35 inky-frame inkypi[1742]: 13:23:35 - ERROR - inkypi - Exception on /playlist [GET]
> May 01 13:23:35 inky-frame inkypi[1742]: Traceback (most recent call last):
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/flask/app.py", line 1511, in wsgi_app
> May 01 13:23:35 inky-frame inkypi[1742]:     response = self.full_dispatch_request()
> May 01 13:23:35 inky-frame inkypi[1742]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/flask/app.py", line 919, in full_dispatch_request
> May 01 13:23:35 inky-frame inkypi[1742]:     rv = self.handle_user_exception(e)
> May 01 13:23:35 inky-frame inkypi[1742]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/flask/app.py", line 917, in full_dispatch_request
> May 01 13:23:35 inky-frame inkypi[1742]:     rv = self.dispatch_request()
> May 01 13:23:35 inky-frame inkypi[1742]:          ^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/flask/app.py", line 902, in dispatch_request
> May 01 13:23:35 inky-frame inkypi[1742]:     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
> May 01 13:23:35 inky-frame inkypi[1742]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/home/inky-frame/InkyPi/src/blueprints/playlist.py", line 76, in playlists
> May 01 13:23:35 inky-frame inkypi[1742]:     return render_template(
> May 01 13:23:35 inky-frame inkypi[1742]:            ^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/flask/templating.py", line 150, in render_template
> May 01 13:23:35 inky-frame inkypi[1742]:     return _render(app, template, context)
> May 01 13:23:35 inky-frame inkypi[1742]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/flask/templating.py", line 131, in _render
> May 01 13:23:35 inky-frame inkypi[1742]:     rv = template.render(context)
> May 01 13:23:35 inky-frame inkypi[1742]:          ^^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/jinja2/environment.py", line 1295, in render
> May 01 13:23:35 inky-frame inkypi[1742]:     self.environment.handle_exception()
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/jinja2/environment.py", line 942, in handle_exception
> May 01 13:23:35 inky-frame inkypi[1742]:     raise rewrite_traceback_stack(source=source)
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/home/inky-frame/InkyPi/src/templates/playlist.html", line 256, in top-level template code
> May 01 13:23:35 inky-frame inkypi[1742]:     {% set refresh_time = plugin_instance.latest_refresh_time | format_relative_time %}
> May 01 13:23:35 inky-frame inkypi[1742]:     ^^^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]:   File "/home/inky-frame/InkyPi/src/blueprints/playlist.py", line 189, in format_relative_time
> May 01 13:23:35 inky-frame inkypi[1742]:     elif dt.date() == (now.date().replace(day=now.day - 1)):
> May 01 13:23:35 inky-frame inkypi[1742]:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> May 01 13:23:35 inky-frame inkypi[1742]: ValueError: day is out of range for month
> May 01 13:23:35 inky-frame inkypi[1742]: 13:23:35 - INFO - werkzeug - 192.168.178.59 - - [01/May/2025 13:23:35] "GET /playlist HTTP/1.1" 500 -

Dogers, 2025-05-01:
I've just hit this and come looking - you need to install `chromium-headless-shell`, run a `git pull` to get the latest code and then restart the service/pi to pick it up

MatrixRetoastet, 2025-05-01:
> I've just hit something similar and come looking - fix for me was to install `chromium-headless-shell`, run a `git pull` to get the latest code and then restart the service/pi to pick it up..

That is already installed in the latest version. There's also new version for the repository.

MatrixRetoastet, 2025-05-05:
Update: I tried it again and it's working now. I think it was a problem with the time calculation on the first of every month. I didn't update or install anything and it's working now.

closed 2025-05-05 by MatrixRetoastet