fatihak/InkyPi #107: Selection of weather icons

ckran, 2025-05-01:
It would be great to be able to select from an alternate set (or two) of weather icons. A set with solid colors could be more readable from a distance.

sjnims, 2025-05-01:
@ckran - [Weather Icons](https://erikflowers.github.io/weather-icons/) ships with some alternates (not all). Take a look and see if this gets you what you want.

sjnims, 2025-05-04:
The actual SVG's are in there. [SVGs](https://github.com/erikflowers/weather-icons/tree/master/svg) - I'm assuming there's some post processing that goes on to combine all of the SVGs into the fonts. Much easier to use a single font and a string of characters that map to a glyph in the font file, and do all the other fun things you can do with fonts rather than having to deal with tons of individual image files. I'm doing something similar in my project where I'm using the weather icons SVGs and combining them into a single sprite.

fatihak, 2025-05-13:
Marking this as enhancement but not sure if it will be implemented unless there is enough interest. As mentioned, you can replace the pngs in ~/InkyPi/src/plugins/weather/icons/ based on your preferences

ckran, 2025-05-20:
Thanks for considering this. When it comes to weather icons you can't please everyone as evidenced of the thousands avaialable. 

However I do question the openweathermap.com choice of icons for **shower rain** (09d.png) which shows clouds with rain and **rain** (10d.png) which shows a sun peeking out from the clouds. I googled "rain or showers" to learn that the NWS and other sources say that rain is continuous and uniform in intensity while showers, not so much. So it seems to me that OWM reversed the icons. BTW I'm not the first to say that. Also if you look at the OWM [Weather Icons](https://openweathermap.org/weather-conditions) page the conditions 3XX Drizzle code get the 09d and the 4XX Rain codes mostly get the 10d. So what's with the sun?