fatihak/InkyPi #119: ImportError: cannot import name '_imagingft' from 'PIL'

x3m, 2025-05-11:
fresh install of raspberry pi os + Inky display connected to it, followed instructions, but unfortunately getting this errors. 

\May 11 13:30:22 raspberrypi systemd[1]: Started inkypi.service - InkyPi App.
May 11 13:30:23 raspberrypi inkypi[1422]: 13:30:23 - INFO - __main__ - Starting web server
May 11 13:30:23 raspberrypi inkypi[1422]: 13:30:23 - INFO - refresh_task - Starting refresh task
May 11 13:30:23 raspberrypi inkypi[1422]: 13:30:23 - INFO - __main__ - Startup flag is set, displaying startup image
May 11 13:30:23 raspberrypi inkypi[1422]: Traceback (most recent call last):
May 11 13:30:23 raspberrypi inkypi[1422]:   File "/home/marlowbg/InkyPi/src/inkypi.py", line 69, in <module>
May 11 13:30:23 raspberrypi inkypi[1422]:     img = generate_startup_image(device_config.get_resolution())
May 11 13:30:23 raspberrypi inkypi[1422]:           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
May 11 13:30:23 raspberrypi inkypi[1422]:   File "/home/marlowbg/InkyPi/src/utils/app_utils.py", line 114, in generate_startup_image
May 11 13:30:23 raspberrypi inkypi[1422]:     image_draw.text((width/2, height/2), "inkypi", anchor="mm", fill=text_color, font=get_font("Jost", title_font_size))
May 11 13:30:23 raspberrypi inkypi[1422]:                                                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
May 11 13:30:23 raspberrypi inkypi[1422]:   File "/home/marlowbg/InkyPi/src/utils/app_utils.py", line 79, in get_font
May 11 13:30:23 raspberrypi inkypi[1422]:     return ImageFont.truetype(font_path, font_size)
May 11 13:30:23 raspberrypi inkypi[1422]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
May 11 13:30:23 raspberrypi inkypi[1422]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/ImageFont.py", line 880, in truetype
May 11 13:30:23 raspberrypi inkypi[1422]:     return freetype(font)
May 11 13:30:23 raspberrypi inkypi[1422]:            ^^^^^^^^^^^^^^
May 11 13:30:23 raspberrypi inkypi[1422]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/ImageFont.py", line 877, in freetype
May 11 13:30:23 raspberrypi inkypi[1422]:     return FreeTypeFont(font, size, index, encoding, layout_engine)
May 11 13:30:23 raspberrypi inkypi[1422]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
May 11 13:30:23 raspberrypi inkypi[1422]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/ImageFont.py", line 228, in __init__
May 11 13:30:23 raspberrypi inkypi[1422]:     raise core.ex
May 11 13:30:23 raspberrypi inkypi[1422]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/ImageFont.py", line 67, in <module>
May 11 13:30:23 raspberrypi inkypi[1422]:     from . import _imagingft as core
**May 11 13:30:23 raspberrypi inkypi[1422]: ImportError: cannot import name '_imagingft' from 'PIL' (/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/__init__.py)**
May 11 13:30:23 raspberrypi systemd[1]: inkypi.service: Deactivated successfully.
May 11 13:30:23 raspberrypi systemd[1]: inkypi.service: Consumed 1.869s CPU time.

fatihak, 2025-05-13:
Did you encounter any errors during installation? This is likely due to the Python Pillow dependency not being installed. Can you try following the instructions [here](https://github.com/fatihak/InkyPi/discussions/41) to manually install the python dependencies.

m-minning, 2025-05-16:
Did you use Raspberry Pi OS Lite? I ran into the same issue, installed the full OS and it worked as expected.

fatihak, 2025-05-18:
@x3m Can you install the full OS as @m-minning suggested?

x3m, 2025-05-19:
> Did you encounter any errors during installation? Can you try following the instructions [here](https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md#recommended-solution-1) to manually install the re-install Pillow.

yes, i tried with this even before writing here - didn't help, same issue.

x3m, 2025-05-19:
> Did you use Raspberry Pi OS Lite? I ran into the same issue, installed the full OS and it worked as expected.

as far as I know i'm with the full version, unless I'm missing something?

How to verify that I'm running the full and not the lite?

```
marlowbg@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-10-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 94f7acf599bd0fd119f523b5afefaccf7a7a1024, stage5
```

fatihak, 2025-05-21:
@x3m Can you run "tasksel --list-tasks"?

I believe it should output "u desktop	Debian desktop environment" if you have the desktop version

michaelsage, 2025-05-22:
I am having this issue to, can confirm I am running full desktop and no errors during the install

fatihak, 2025-05-23:
@michaelsage can you try uninstalling Pillow/PIL and re-installing:
```bash
sudo su
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip uninstall PIL Pillow
pip install Pillow
deactivate
```

michaelsage, 2025-05-23:
Issue persists I'm afraid

May 23 16:25:06 slowmovie inkypi[1077]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/ImageFont.py", line 877, in freetype
May 23 16:25:06 slowmovie inkypi[1077]:     return FreeTypeFont(font, size, index, encoding, layout_engine)
May 23 16:25:06 slowmovie inkypi[1077]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
May 23 16:25:06 slowmovie inkypi[1077]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/ImageFont.py", line 228, in __init__
May 23 16:25:06 slowmovie inkypi[1077]:     raise core.ex
May 23 16:25:06 slowmovie inkypi[1077]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/ImageFont.py", line 67, in <module>
May 23 16:25:06 slowmovie inkypi[1077]:     from . import _imagingft as core
May 23 16:25:06 slowmovie inkypi[1077]: ImportError: cannot import name '_imagingft' from 'PIL' (/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/__init__.py)
May 23 16:25:06 slowmovie systemd[1]: inkypi.service: Deactivated successfully.
May 23 16:25:06 slowmovie systemd[1]: inkypi.service: Consumed 4.476s CPU time.

michaelsage, 2025-05-25:
Yup that fixed it for me too