fatihak/InkyPi #126: Weather API exceeding 1000

pinpoint90, 2025-05-18:
I am using the weather plugin with screen update once every hour. I got a notice today what I have used 1000 api calls. Is this a bug and what may be the cause?

[image]

pinpoint90, 2025-05-19:
Trying to manually update weather plugin, "Error! An error occurred: 'NoneType' object cannot be interpreted as an integer"

pinpoint90, 2025-05-20:
Strangely, this happens when choosing latitude and longitude on my current location. When changing to an other city there is no error.

fatihak, 2025-05-20:
@pinpoint90 do you mind sharing the city you're trying so I can try to recreate the issue? If you're not comfortable sharing it here you can add me on discord @ fmetinak

pinpoint90, 2025-05-20:
Tromsø, Norway

fatihak, 2025-05-21:
Looked into this and apparently OpenWeatherMap doesn't return sunrise/sunset values "for polar areas in midnight sun and polar night periods" [docs](https://openweathermap.org/api/one-call-3#current)

Just pushed a change to skip those metrics if it's not found in the response. Can you pull the latest changes and see if it's resolved?

closed 2025-05-21 by fatihak