fatihak/InkyPi #130: Display doesnt refresh

chunk100, 2025-05-25:
Time display does not update. 
After successful install, selecting a cloclk is successful and shows the current time.  However, the time doesnt update as would be expected periodically.  It just remains on the time when I select the clock.

Install all seems successful, no error messages and can access menu on pi.local site no problems.  Can also SSH in no problems.  Have experimented a lot with different SD cards (plenty of space) and power supplies (plenty of power now!).  

Using a pi zero 2 W with an Inky Impression 7.3 Spectra

Only possible difference I can see is this is the "2025" edition from pimoroni:

https://shop.pimoroni.com/products/inky-impression-7-3?variant=55186435244411

Looked at troubleshooting and following logs below.  Once a minute there is this:

May 25 10:14:09 piwx inkypi[658]: 10:14:09 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-05-25 10:14:09
May 25 10:14:09 piwx inkypi[658]: 10:14:09 - INFO - refresh_task - Active playlist 'Default' has no plugins.

And the last set of log entries with anything different are here:

May 25 10:12:09 piwx inkypi[658]: 10:12:09 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-05-25 10:12:09
May 25 10:12:09 piwx inkypi[658]: 10:12:09 - INFO - refresh_task - Active playlist 'Default' has no plugins.
May 25 10:12:42 piwx inkypi[658]: 10:12:42 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:42] "GET / HTTP/1.1" 200 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /static/styles/main.css HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /images/clock/icon.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /images/newspaper/icon.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /images/weather/icon.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /images/image_upload/icon.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /images/screenshot/icon.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /static/icons/settings.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /static/images/current_image.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /static/icons/playlist.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /images/ai_image/icon.png HTTP/1.1" 304 -
May 25 10:12:43 piwx inkypi[658]: 10:12:43 - INFO - werkzeug - 192.168.1.38 - - [25/May/2025 10:12:43] "GET /images/ai_text/icon.png HTTP/1.1" 304 -
May 25 10:13:09 piwx inkypi[658]: 10:13:09 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-05-25 10:13:09
May 25 10:13:09 piwx inkypi[658]: 10:13:09 - INFO - refresh_task - Active playlist 'Default' has no plugins.

anton-ovcharenko, 2025-05-26:
Also, it does not work for me (same PIM773).
`May 26 17:04:11 inkypi inkypi[545]: 17:04:11 - INFO - refresh_task - Manual update requested
May 26 17:04:11 inkypi inkypi[545]: 17:04:11 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'clock1'
May 26 17:04:11 inkypi inkypi[545]: 17:04:11 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Playlist', 'playlist': 'Default', 'plugin_id': 'clock', 'plugin_instance': 'clock1', 'refresh_time': '2025-05-26T17:04:11.282602+02:00', 'image_hash': 'b2f375605ab29fb4848b532e8229c992e69e296f91d19ac6aea96cfaf98a1c32'}
May 26 17:04:39 inkypi inkypi[545]: 17:04:39 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:39] "POST /display_plugin_instance HTTP/1.1" 200 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /playlist HTTP/1.1" 200 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /static/styles/main.css HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /static/scripts/response_modal.js HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /images/weather/icon.png HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /static/icons/playlist.png HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /static/icons/settings.png HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /static/icons/display.png HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /static/icons/edit.png HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /static/icons/remove.png HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /images/image_upload/icon.png HTTP/1.1" 304 -
May 26 17:04:40 inkypi inkypi[545]: 17:04:40 - INFO - werkzeug - 192.168.0.101 - - [26/May/2025 17:04:40] "GET /images/clock/icon.png HTTP/1.1" 304 -
May 26 17:05:39 inkypi inkypi[545]: 17:05:39 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-05-26 17:05:39
May 26 17:05:39 inkypi inkypi[545]: 17:05:39 - INFO - refresh_task - Not time to update display. | latest_update: 2025-05-26 17:04:11 | plugin_cycle_interval: 3600
May 26 17:06:39 inkypi inkypi[545]: 17:06:39 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-05-26 17:06:39
May 26 17:06:39 inkypi inkypi[545]: 17:06:39 - INFO - refresh_task - Not time to update display. | latest_update: 2025-05-26 17:04:11 | plugin_cycle_interval: 3600
`
and no changes on the screen

fatihak, 2025-05-26:
@chunk100 You need to add the plugin to a playlist with the "Add to Playlist" button and set an interval for it to refresh. "Update Now" only does an immediate update.

chunk100, 2025-05-30:
@fatihak Thank you for the advice.
When selecting Clock, I picked the clock face of choice then went to 'Add to Playlist' and added a refresh once a minute.
I selected 'Updated Now' and the clock correctly updated but didnt refresh.
I then went to the Settings menu and updated the Plugin Cycle menu to once per minute.  Only once this was done did the clock update once per minute.
I found it a bit confusing as there are Playlists and Plugins which I think I now understand but it looks like they only update if there are one of each.

For the clock it might just be better to select a 'Refresh time' instead of 'Add to Playlist' to make it simpler.
Also, once Ive set up a Playlist, is there a way to view / edit them?

closed 2025-05-30 by chunk100