fatihak/InkyPi #140: Internal Server Error message when trying to access playlists

menthol1979, 2025-06-01:
Hello, 

On numerous tries to find out the best settings for optimal display on a 2025 Inky Impression 7,3, now I cannot access the Playlists. 

When clicking on the playlists icon on top right of the config page, I get:

**Internal Server Error**

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

Being a linux noob, can anyone assist on what kind of log is needed for assistance and where to find it? 

Thanks!

phattmatt, 2025-06-01:
Also getting the same error. Appears related to the handling of dates. From the log file:

> Jun 01 15:23:46 inkypi inkypi[544]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/jinja2/environment.py", line 942, in handle_exception
> Jun 01 15:23:46 inkypi inkypi[544]:     raise rewrite_traceback_stack(source=source)
> Jun 01 15:23:46 inkypi inkypi[544]:   File "/home/pi/InkyPi/src/templates/playlist.html", line 256, in top-level template code
> Jun 01 15:23:46 inkypi inkypi[544]:     {% set refresh_time = plugin_instance.latest_refresh_time | format_relative_time %}
> Jun 01 15:23:46 inkypi inkypi[544]:     ^^^^^^^^^^^^^^^^^^^^^^^^^
> Jun 01 15:23:46 inkypi inkypi[544]:   File "/home/pi/InkyPi/src/blueprints/playlist.py", line 189, in format_relative_time
> Jun 01 15:23:46 inkypi inkypi[544]:     elif dt.date() == (now.date().replace(day=now.day - 1)):
> Jun 01 15:23:46 inkypi inkypi[544]:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> Jun 01 15:23:46 inkypi inkypi[544]: ValueError: day is out of range for month
> Jun 01 15:23:46 inkypi inkypi[544]: 15:23:46 - INFO - werkzeug - 192.168.1.122 - - [01/Jun/2025 15:23:46] "GET /playlist HTTP/1.1" 500 -

phattmatt, 2025-06-01:
The "day=now.day - 1" bit will result in a '0' day on the first of any month. Looks like Python has some functions for dealing with dates that may be 'safer' to use (timedelta).

Looks like the this may be a quick fix (caveat, I am no Dev):

Edit the following file "/home/pi/InkyPi/src/blueprints/playlist.py"

Replace (line 4):

`from datetime import datetime`

With:

`from datetime import datetime, timedelta`

And replace (line 189):

`    elif dt.date() == (now.date().replace(day=now.day - 1)):`

With:

'    elif dt.date() == (now.date() - timedelta(days=1)):'

phattmatt, 2025-06-01:
Created a small PR with the above changes:

https://github.com/fatihak/InkyPi/pull/141

fatihak, 2025-06-01:
Thanks for the PR, merged! Let me know if the issue is resolved