fatihak/InkyPi #152: Use open source open-meteo.com for weather

jaredreich, 2025-06-05:
Just a consideration, perhaps more future-proof since it's also open source:
https://open-meteo.com
https://github.com/open-meteo/open-meteo

Thanks so much for creating this lib, amazing work!

Leberkasbepi, 2025-06-12:
I made my own weather plugin with the open-meteo api for me in Europe. Its more accurate then the us system.

pwmackin, 2025-06-14:
I'm a noob--how would I install this plugin?  
Where can I see a list of new plugins besides the 9 that come with the InkyPi install?
OpenWeatherMap doesn't work for me.

> I made my own weather plugin with the open-meteo api for me in Europe. Its more accurate then the us system.

Leberkasbepi, 2025-06-14:
Im also a noob and don't work with GitHub. But I can share you tomorrow the plugin as a zip file if you want. Installing is very easy. The inkypi install file do this for you.

Leberkasbepi, 2025-06-15:
@pwmackin Here is my plugin: https://www.swisstransfer.com/d/b38e5087-3ecf-4669-b22e-6913aa1fa895
If you have worries that there are viruses in there, you can check it with virustotal.com.

You have to uninstall InkyPi before with `cd InkyPi && sudo bash install/uninstall.sh` - then store the file in the plugin folder, create the folder "openmeteo" also in the plugins folder and use in the InkyPi directory this command `tar -xzvf /home/admin/InkyPi/src/plugins/openmeteo.tar.gz -C /home/admin/InkyPi/src/plugins` 
Now all files should be extracted in the openmeteo folder. It's very important that only the files are in this folder, not a separate openmeteo folder!

Now install the InkyPi again. If you are in the InkyPi directory, enter `sudo bash install/install.sh` 
That's it!

Leberkasbepi, 2025-06-28:
You can easily search the German words in visual studio and change it to English, so it should display the right words.

When you tell me how I can share the code here, maybe I can do it. 

I'm not a developer, so I don't use the git functionality and it was easier for me to change the code direct.

I don't get the moon phases running, cuz openmeteo don't serve this information in the same API.
So I was very happy it running with the normal wheater data.

If the other issue with the freezing is solved, I can try to ad more api for this.

pwmackin, 2025-07-25:
Hey, thanks.. I'm still figuring out how to share the code.  I think it would be easy for me to edit.
But in the meantime, the bigger issue I have, is my  InkyPi wont stay up for more than 24 hours  or so without crashing.
I have to unplug / replug almost daily.  It's the Pi Zero W model (or is it W2 ? Cheap one)
Anyway when I get around to fixing these issues I'll try to get with you about the source code but thanks again!
P

Leberkasbepi, 2025-07-25:
This issue is known, see here. https://github.com/fatihak/InkyPi/issues/159

closed 2025-07-26 by fatihak

TonyTTTT, 2025-07-29:
Is an API Key needed for open-metro?  (I don't see it mentioned in ~/InkyPi/docs/api_keys.md