fatihak/InkyPi #154: Screenshot functionality not working on Raspberry Pi 0 (chromium-headless throws a seg fault)

VRvinny, 2025-06-07:
Similar to the stdout of this issue https://github.com/fatihak/InkyPi/issues/86, the screenshot functionality throws an error on a Raspberry Pi 0. Digging around to find the issue, I found out that chromium and all its variants aren't supported on it, it throws a segmentation fault anytime you try and run the command from a shell.

For what it's worth to any future reader, I found a workaround using wkhtmltoimage.
Find the latest release here https://wkhtmltopdf.org/downloads.html and then build it with
```
sudo apt update
sudo apt install libssl1.1 xfonts-75dpi xfonts-base
sudo dpkg -i wkhtmltox_0.12.6.1-2.raspberrypi.bullseye_armhf.deb 
```
Then modify the images_util.py file with the following:
```
        command = [
            "wkhtmltoimage", 
            "--width", f"{dimensions[0]}", 
            "--height", f"{dimensions[1]}", 
            "--images",  
            "--enable-javascript",  
            "--javascript-delay", "1000",  
            "--enable-local-file-access", 
            f"{target}",  
            f"{img_file_path}"
        ]
```

fatihak, 2025-06-29:
Thanks for this find. The Pi Zero is not recommended for this project but it might be good to include a disclaimer somewhere in the README for this, and include your fix somewhere in the docs for those that might be interested.

VRvinny, 2025-07-19:
Unfortuately not, my only tip to debug is to try and run the code locally and use a browser to see what you are screenshotting.

I found that there's a big difference in rendering between Chromium and wkhtmltoimage in that chromium will 'fill in the gaps' if you haven't written your HTML + CSS + JS correctly whereas wkhtmltoimage will interpret your page literally. I needed to tweak my CSS + JS to get the alignments right with wkhtmltoimage when I originally had it working with Chromium

fatihak, 2025-07-23:
Yeah I had looked into wkhtmltoimage when initially building the screenshotting feature and ran into this issue. From what i understand wkhtmltoimage uses an outdated Qt WebKit rendering engine and does not support a lot of newer HTML & Javascript features that the weather or calendar plugins use.
It would be nice to replace chromium with wkhtmltoimage alltogether since it's more lightweight, but that would require refactoring the existing plugins.

phattmatt, 2025-07-30:
wkhtmltoimage has been archived as a project and hasn't been updated in 2 years:

https://wkhtmltopdf.org/status.html

https://github.com/wkhtmltopdf/wkhtmltopdf

I would recommend not using it for rendering external HTML.

phattmatt, 2025-08-21:
Going to close this issue since wkhtmltoimage will not be supported in InkyPi.