fatihak/InkyPi #157: New installation python import error, so service is inactive (dead)

Kelniimir, 2025-06-10:
**Describe the bug**
- Bought Inky Impressions 7.3 (older 7 color screen). 
- Using Raspberry Pi 2W.
- Installed Raspbian Bookworm. 
- Extended partition to occupy whole MicroSD (32Gb)
- Updated and upgraded.
- Passed installation step with errors first during venv creation. 
- Retried after reboot, without using tmux, succeeded. 
- Cannot access WebGUI, service is inactive with reason "ImportError: cannot import name '_imagingft' from 'PIL' (/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL/__init__.py)" 
- Considered possible resolution as https://github.com/fatihak/InkyPi/issues/82
- Created script to execute /usr/local/inkypi/venv_inkypi/bin/python3 -m pip install --upgrade Pillow. Executed with sudo. Without sudo, there were permission issues to overwrite some files so upgrade always failed. Successfully upgraded it to 11.2.1. It didn't resolved the issue. 

**Expected behavior**
It is expected that after executing installation script, service would be running without interruption. Otherwise, restoration steps to be added to Troubleshooting page on Git. 

**To Reproduce**
Steps to reproduce the behavior:
1. Use Raspberry Pi 2W
2. Burn latest Raspbian Bookworm on MicroSD using official Imager. 
3. Connect to device over SSH via external Ethernet adapter, as due to changes in Bookworm, reconfigured WiFi spot isn't active. Configure it, switch to WiFi for following operations.
4. Extend partition using "sudo raspi-config" - Advanced Options - Expand Filesystem. 
5. Reboot. 
6. Update and upgrade using:
sudo apt -y update &&
sudo apt -y upgrade &&
sudo apt -y dist-upgrade &&
sudo apt autoremove &&
date &&
sudo apt install tmux -y &&
sudo apt install mosh -y &&
date
7. Reboot. 
8. Create custom "git" folder in Home directory. Clone git in there.
9. Launch install script. Face errors during creating venv. 
10. Reboot. 
11. Launch install script again. No errors this time, reboot. 
12. Unable to open WebGUI. Service is inactive (dead). 
13. Following https://github.com/fatihak/InkyPi/issues/82, upgraded to Pillow-11.2.1 using file with row "/usr/local/inkypi/venv_inkypi/bin/python3 -m pip install --upgrade Pillow", which was then executed with sudo sh under venv. Couldn't launch sudo with command in venv, and without it, was getting permission error:
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: 'GribStubImagePlugin.py'
Check the permissions. 
14. Service still is getting inactive after reboot, error hasn't changed (error text provided in attached file "defect.txt".

**Logs**
<details>
<summary>Inky debug</summary>
```

(venv_inkypi) pi@PiZero2Soldered:~/git/InkyPi/docs $ journalctl -u inkypi -n 100
Jun 10 18:47:55 PiZero2Soldered systemd[1]: Started inkypi.service - InkyPi App.
Jun 10 18:48:02 PiZero2Soldered inkypi[854]: /usr/local/inkypi/venv_inkypi/lib/python3.11/site>Jun 10 18:48:02 PiZero2Soldered inkypi[854]:   from pkg_resources import declare_namespace
Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - __main__ - Starting web server
Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:04 PiZero2Soldered inkypi[854]: 18:48:04 - INFO - config - Reading plugin info fr>Jun 10 18:48:17 PiZero2Soldered inkypi[854]: 18:48:17 - INFO - refresh_task - Starting refresh>Jun 10 18:48:17 PiZero2Soldered inkypi[854]: 18:48:17 - INFO - __main__ - Startup flag is set,>Jun 10 18:48:17 PiZero2Soldered inkypi[854]: Traceback (most recent call last):
Jun 10 18:48:17 PiZero2Soldered inkypi[854]:   File "/home/pi/git/InkyPi/src/inkypi.py", line >Jun 10 18:48:17 PiZero2Soldered inkypi[854]:     img = generate_startup_image(device_config.ge>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:   File "/home/pi/git/InkyPi/src/utils/app_utils.p>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:     image_draw.text((width/2, height/2), "inkypi">Jun 10 18:48:17 PiZero2Soldered inkypi[854]:                                                  >Jun 10 18:48:17 PiZero2Soldered inkypi[854]:   File "/home/pi/git/InkyPi/src/utils/app_utils.p>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:     return ImageFont.truetype(font_path, font_siz>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:   File "/usr/local/inkypi/venv_inkypi/lib/python3>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:     return freetype(font)
Jun 10 18:48:17 PiZero2Soldered inkypi[854]:            ^^^^^^^^^^^^^^
Jun 10 18:48:17 PiZero2Soldered inkypi[854]:   File "/usr/local/inkypi/venv_inkypi/lib/python3>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:     return FreeTypeFont(font, size, index, encodi>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:   File "/usr/local/inkypi/venv_inkypi/lib/python3>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:     raise core.ex
Jun 10 18:48:17 PiZero2Soldered inkypi[854]:   File "/usr/local/inkypi/venv_inkypi/lib/python3>Jun 10 18:48:17 PiZero2Soldered inkypi[854]:     from . import _imagingft as core
Jun 10 18:48:17 PiZero2Soldered inkypi[854]: ImportError: cannot import name '_imagingft' from>Jun 10 18:48:33 PiZero2Soldered systemd[1]: inkypi.service: Deactivated successfully.
Jun 10 18:48:33 PiZero2Soldered systemd[1]: inkypi.service: Consumed 8.345s CPU time.
lines 1-38/38 (END)
```
</details>

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
- Inky Impression 7.3 screen (older, 7 color)
- Newly ordered Raspberry Pi 2W
- 32Gb MicroSD
- cat /etc/os-release
[... 21 lines cut ...]
Requirement already satisfied: itsdangerous>=2.2.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from flask==3.1.1->-r requirements.txt (line 1)) (2.2.0)
Requirement already satisfied: jinja2>=3.1.2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from flask==3.1.1->-r requirements.txt (line 1)) (3.1.6)
Requirement already satisfied: markupsafe>=2.1.1 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from flask==3.1.1->-r requirements.txt (line 1)) (3.0.2)
Requirement already satisfied: gpiodevice>=0.0.3 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from inky==2.1.0->-r requirements.txt (line 3)) (0.0.5)
Requirement already satisfied: smbus2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from inky==2.1.0->-r requirements.txt (line 3)) (0.5.0)
Requirement already satisfied: spidev in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from inky==2.1.0->-r requirements.txt (line 3)) (3.7)
Requirement already satisfied: charset-normalizer<4,>=2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from requests==2.32.3->-r requirements.txt (line 4)) (3.4.2)
Requirement already satisfied: idna<4,>=2.5 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from requests==2.32.3->-r requirements.txt (line 4)) (3.10)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from requests==2.32.3->-r requirements.txt (line 4)) (2025.4.26)
Requirement already satisfied: anyio<5,>=3.5.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r requirements.txt (line 9)) (4.9.0)
Requirement already satisfied: distro<2,>=1.7.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r requirements.txt (line 9)) (1.9.0)
Requirement already satisfied: httpx<1,>=0.23.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from openai==1.58.1->-r requirements.txt (line 9)) (0.28.1)
[... cut ...]
Requirement already satisfied: pydantic-core==2.33.2 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from pydantic<3,>=1.9.0->openai==1.58.1->-r requirements.txt (line 9)) (2.33.2)
Requirement already satisfied: typing-inspection>=0.4.0 in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from pydantic<3,>=1.9.0->openai==1.58.1->-r requirements.txt (line 9)) (0.4.1)
Requirement already satisfied: gpiod in /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages (from gpiodevice>=0.0.3->inky==2.1.0->-r requirements.txt (line 3)) (2.3.0)
```
</details>

phattmatt, 2025-06-10:
Seems like a similar issue reported here: https://github.com/fatihak/InkyPi/issues/119

A suggested fix can be found in this comment: https://github.com/fatihak/InkyPi/issues/119#issuecomment-2909404624

```
sudo su
source "/usr/local/inkypi/venv_inkypi/bin/activate"
pip uninstall PIL Pillow
sudo apt-get install libfreetype6-dev 
pip install Pillow
deactivate
```

You should be able to debug and see it works with:

`echo "from PIL import _imagingft" | python && echo "Ok."`

Kelniimir, 2025-06-10:
Hello @phattmatt

I actually was going through similar process, but fitting it to my situation.

- It seems it is indeed related to whenever full or Lite version of Raspbian is in use.
In my case with Pi 2W, I'm not even sure full is a) possible and b) viable due to limited RAM.
Therefore users running low power Raspberries, or intentionally using Lite distro for headless experience might be affected. 

- It indeed seems to be related to fact that libfreetype6-dev and it's dependencies are not installed, or installed after Pillow (as per short googling, it also could cause similar issues).

- It seems Pillon newer than 11.0 which is part of requirement is indeed not needed for correct functioning of this code. So updating it manually was not necessary.

- `sudo su source "/usr/local/inkypi/venv_inkypi/bin/activate"` unfortunately gave me `su: user source does not exist or the user entry does not contain all the required fields`. So I did it slightly different, while still keeping same idea.

1. Manually install `libfreetype6-dev` outside of venv. (It can be done in venv,right after step 6, prior 7, but I wanted to follow original idea - that FULL ISO is supplied with it, therefore it is installed before Pillow. And for this reason people who use Pi 3-4-5 and use FULL distro do not experience this).
2. touch <filename>
3. sudo nano <filename>
4. Fill following:
```
/usr/local/inkypi/venv_inkypi/bin/python3 -m pip uninstall PIL Pillow &&
/usr/local/inkypi/venv_inkypi/bin/python3 -m pip install --no-cache-dir Pillow
```
5. Save with Ctrl+O, exit with Ctrl+X
6. Execute `source "/usr/local/inkypi/venv_inkypi/bin/activate"`
7. Execute `sudo sh <filename>`
8. Execute `sudo reboot`

**This resolved issue for me. **
Service is active for 15 minutes, while before it was failing in 10 seconds after launch.

Reason for full path on step 4 - without sudo I couldn't neither upgrade not uninstall Pillow, and with sudo it ignored venv and tried executing it system wide.

phattmatt, 2025-06-10:
Glad you got it working. 

I'm running InkyPi on a Raspberry Pi Zero 2W using Raspberry Pi OS 64-bit Lite and haven't run into any of these issues, so it's some kind of transient issue.

I've just run a fresh install of InkyPi on a fresh RPiOS (64-bit - Lite), which has worked, and I see the following:

```
pi@inkypi-dev:~/InkyPi $ cat /etc/rpi-issue
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage2
```
Where stage2 means a `Lite` image was used to install RPiOS, and stage4/5 is a Desktop/Full image.
```
pi@inkypi-dev:~/InkyPi $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@inkypi-dev:~/InkyPi $ uname -a
Linux inkypi-dev 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```
```
pi@inkypi-dev:~/InkyPi $ apt show libfreetype6-dev
Package: libfreetype6-dev
Version: 2.12.1+dfsg-5+deb12u4
Priority: optional
Section: libdevel
Source: freetype
Maintainer: Hugh McMaster <hugh.mcmaster@outlook.com>
Installed-Size: 70.7 kB
Depends: libfreetype-dev (= 2.12.1+dfsg-5+deb12u4)
Homepage: https://freetype.org
Tag: devel::lang:c, devel::library, implemented-in::c, role::devel-lib,
 works-with::font
Download-Size: 46.7 kB
APT-Sources: http://deb.debian.org/debian bookworm/main arm64 Packages
Description: FreeType 2 font engine, development files (transitional package)
 The FreeType project is a team of volunteers who develop free,
 portable and high-quality software solutions for digital typography.
 They specifically target embedded systems and focus on providing small,
 efficient and ubiquitous products.
 .
 This package contains all of the supplementary files you need to develop your
 own programs using the FreeType 2 library.
 .
 This is a transitional package. It can safely be removed.
```
```
pi@inkypi-dev:~/InkyPi $ /usr/local/inkypi/venv_inkypi/bin/python -m pip list
Package            Version
------------------ ---------
annotated-types    0.7.0
anyio              4.9.0
blinker            1.9.0
certifi            2025.4.26
charset-normalizer 3.4.2
click              8.2.1
distro             1.9.0
Flask              3.1.1
gpiod              2.3.0
gpiodevice         0.0.5
h11                0.16.0
httpcore           1.0.9
httpx              0.28.1
idna               3.10
inky               2.1.0
itsdangerous       2.2.0
Jinja2             3.1.6
jiter              0.10.0
MarkupSafe         3.0.2
numpy              2.2.1
openai             1.58.1
pillow             11.0.0
pip                25.1.1
pydantic           2.11.5
pydantic_core      2.33.2
python-dotenv      1.0.1
pytz               2025.2
requests           2.32.3
setuptools         80.9.0
smbus2             0.5.0
sniffio            1.3.1
spidev             3.7
tqdm               4.67.1
typing_extensions  4.14.0
typing-inspection  0.4.1
urllib3            2.3.0
Werkzeug           3.1.3
wheel              0.45.1
```

So it's confusing why this is happening.

Kelniimir, 2025-06-10:
What I can add as per your comment
- I seems I'm running Raspbian 32-bit released 2025-05-13. 
I can see it in Imager on Win PC, as it shows "Cached". 
Justification is - I original wanted to use PiZero1W, but struggled to enable reconfigured WiFi untill PiZero2W arrived, together with MicroUSB Ethernet adapter.
And I for PiZero1W, 64bit OS is NOT OFFERED. Probably not supported.

For other info
<details>
  <summary>Output</summary>
```

pi@PiZero2Soldered:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f548f25325e28bea61904f5038a9b9fd0c92b07b, stage4
pi@PiZero2Soldered:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@PiZero2Soldered:~ $ /usr/local/inkypi/venv_inkypi/bin/python -m pip list
Package            Version
------------------ ---------
annotated-types    0.7.0
anyio              4.9.0
blinker            1.9.0
certifi            2025.4.26
charset-normalizer 3.4.2
click              8.2.1
distro             1.9.0
Flask              3.1.1
gpiod              2.3.0
gpiodevice         0.0.5
h11                0.16.0
httpcore           1.0.9
httpx              0.28.1
idna               3.10
inky               2.1.0
itsdangerous       2.2.0
Jinja2             3.1.6
jiter              0.10.0
MarkupSafe         3.0.2
numpy              2.2.1
openai             1.58.1
pillow             11.2.1
pip                25.1.1
pydantic           2.11.5
pydantic_core      2.33.2
python-dotenv      1.0.1
pytz               2025.2
requests           2.32.3
setuptools         80.9.0
smbus2             0.5.0
sniffio            1.3.1
spidev             3.7
tqdm               4.67.1
typing_extensions  4.14.0
typing-inspection  0.4.1
urllib3            2.3.0
Werkzeug           3.1.3
wheel              0.45.1
```
</details>

phattmatt, 2025-06-10:
I think I've worked out what's happening.

If installing Pillow on a 64-bit (aarch64) release of RPiOS then there is a pre-built `.whl` package available which includes 'freetype' support built-in. No need for `libfreetype-dev` to be installed.

<details><summary>64-bit - Built-in support for freetype (and others)</summary>
<p>

```
Python 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PIL import features
>>> features.pilinfo()
--------------------------------------------------------------------
Pillow 11.0.0
Python 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0]
--------------------------------------------------------------------
Python executable is /home/pi/tutorial-env2/bin/python
Environment Python files loaded from /home/pi/tutorial-env2
System Python files loaded from /usr
--------------------------------------------------------------------
Python Pillow modules loaded from /home/pi/tutorial-env2/lib/python3.11/site-packages/PIL
Binary Pillow modules loaded from /home/pi/tutorial-env2/lib/python3.11/site-packages/PIL
--------------------------------------------------------------------
--- PIL CORE support ok, compiled for 11.0.0
--- TKINTER support ok, loaded 8.6
--- FREETYPE2 support ok, loaded 2.13.2
--- LITTLECMS2 support ok, loaded 2.16
--- WEBP support ok, loaded 1.4.0
--- JPEG support ok, compiled for libjpeg-turbo 3.0.4
--- OPENJPEG (JPEG2000) support ok, loaded 2.5.2
--- ZLIB (PNG/ZIP) support ok, loaded 1.2.13
--- LIBTIFF support ok, loaded 4.6.0
--- RAQM (Bidirectional Text) support ok, loaded 0.10.1, fribidi 1.0.8, harfbuzz 10.0.1
*** LIBIMAGEQUANT (Quantization method) support not installed
--- XCB (X protocol) support ok
--------------------------------------------------------------------
``` 

</p>
</details> 

If installing Pillow on a 32-bit (armhf/arm7l) release of RPiOS then there is no pre-built `.whl` package available. 

In this case pip downloads the Pillow source code and builds the binarys; because the `libfreetype-dev` package is missing then no freetype support is compiled into the Pillow package.

<details><summary>32-bit - No freetype support</summary>
<p>

```
(venv_inkypi) pi@inkypi-dev:~/InkyPi $ python
Python 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PIL import features
>>> features.pilinfo()
--------------------------------------------------------------------
Pillow 11.0.0
Python 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0]
--------------------------------------------------------------------
Python executable is /usr/local/inkypi/venv_inkypi/bin/python
Environment Python files loaded from /usr/local/inkypi/venv_inkypi
System Python files loaded from /usr
--------------------------------------------------------------------
Python Pillow modules loaded from /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL
Binary Pillow modules loaded from /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL
--------------------------------------------------------------------
--- PIL CORE support ok, compiled for 11.0.0
*** TKINTER support not installed
*** FREETYPE2 support not installed
*** LITTLECMS2 support not installed
--- WEBP support ok, loaded 1.2.4
--- JPEG support ok, compiled for libjpeg-turbo 2.1.5
*** OPENJPEG (JPEG2000) support not installed
--- ZLIB (PNG/ZIP) support ok, loaded 1.2.13
--- LIBTIFF support ok, loaded 4.5.0
*** RAQM (Bidirectional Text) support not installed
*** LIBIMAGEQUANT (Quantization method) support not installed
*** XCB (X protocol) support not installed
```

</p>
</details> 

<details><summary>32-bit - After installing libfreetype-dev and then installing Pillow</summary>
<p>

```
Python 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>  from PIL import features
  File "<stdin>", line 1
    from PIL import features
IndentationError: unexpected indent
>>> from PIL import features
>>> features.pilinfo()
--------------------------------------------------------------------
Pillow 11.0.0
Python 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0]
--------------------------------------------------------------------
Python executable is /usr/local/inkypi/venv_inkypi/bin/python
Environment Python files loaded from /usr/local/inkypi/venv_inkypi
System Python files loaded from /usr
--------------------------------------------------------------------
Python Pillow modules loaded from /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL
Binary Pillow modules loaded from /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/PIL
--------------------------------------------------------------------
--- PIL CORE support ok, compiled for 11.0.0
*** TKINTER support not installed
--- FREETYPE2 support ok, loaded 2.12.1
*** LITTLECMS2 support not installed
--- WEBP support ok, loaded 1.2.4
--- JPEG support ok, compiled for libjpeg-turbo 2.1.5
*** OPENJPEG (JPEG2000) support not installed
--- ZLIB (PNG/ZIP) support ok, loaded 1.2.13
--- LIBTIFF support ok, loaded 4.5.0
*** RAQM (Bidirectional Text) support not installed
*** LIBIMAGEQUANT (Quantization method) support not installed
*** XCB (X protocol) support not installed
--------------------------------------------------------------------
``` 

</p>
</details> 

List of packages available: https://pypi.org/project/pillow/11.0.0/#files

I suggest:

- Adding a note about installing InkPi on 32-bit releases of Raspberry Pi OS to the installation instructions.
- Consider adding additional prerequisites (at least `libfreetype-dev`) to the `debian-requirements.txt`

fatihak, 2025-06-11:
I added libfreetype6-dev to debian-requirements.txt so it gets installed during the install script. That should resolve the issue based on the discussion here, let me know if this is good to close.

fatihak, 2025-06-11:
Yep thanks @phattmatt for looking into this!