fatihak/InkyPi #161: AI Image playlist doesn't save the Quality setting.

rgelb, 2025-06-13:
**Describe the bug**
I attempt to change the Quality setting from Standard to HD. 
Save it.  Go back out and back in.  The setting is still Standard.

Video describes it better than I could.  And yes, I am on the latest version, just did an update prior to this.

https://github.com/user-attachments/assets/b4b17a93-37b5-4ef0-b50f-75dfbe29f6c5

phattmatt, 2025-06-14:
I think I found the issue in the `./InkyPi/src/plugins/ai_image/settings.html` file:

Replacing
```
            document.getElementById('textModel').value = pluginSettings.textModel;
```
With
```
            document.getElementById('imageModel').value = pluginSettings.imageModel;
```

Seems to fix the issue for me.

Small PR to fix:

https://github.com/fatihak/InkyPi/pull/162

fatihak, 2025-06-18:
Closing this since there was no response, feel free to reopen if you're having issues

closed 2025-06-18 by fatihak

rgelb, 2025-06-18:
@phattmatt @fatihak Thank you gentlemen.  It worked!