fatihak/InkyPi #163: Add support for delayed rendering using --virtual-time-budget for proper screenshot timing

Tjarkbro, 2025-06-15:
Hi there,

first of all, thank you for your great work on this project!

I've been using the take_screenshot functionality to capture screenshots of web pages rendered by Chromium, specifically for integrations such as Home Assistant. However, I ran into an issue: the screenshots are often taken too early, before the page has fully rendered or the JavaScript has finished executing.

To fix this, I modified the command by adding a few Chromium flags that significantly improved the results:
**image_utils.py**
```
"--virtual-time-budget=10000",
"--run-all-compositor-stages-before-draw"
```
In particular, --virtual-time-budget=10000 (10 seconds) allows Chromium to wait for scripts to finish and rendering to settle before capturing the screenshot.

With this addition, I was finally able to take reliable screenshots of dynamic interfaces like Home Assistant dashboards.

Would you consider adding support for this flag or even making it configurable via function parameters?

Thanks again for your work!

Best regards!

cello8619, 2025-06-27:
@Tjarkbro  What file and where do you add this code so I can do this now before the enhancement is officially added.  I created a looker dashboard and I am having the same issue.

phattmatt, 2025-06-29:
> [@Tjarkbro](https://github.com/Tjarkbro) What file and where do you add this code so I can do this now before the enhancement is officially added. I created a looker dashboard and I am having the same issue.

The file is mentioned in the original post:

[src/utils/image_utils.py](https://github.com/fatihak/InkyPi/blob/main/src/utils/image_utils.py)

With the changes needing to be applied to this command:

https://github.com/fatihak/InkyPi/blob/a75377d178465539ba87d57aaf3a25f08822bd34/src/utils/image_utils.py#L110

Tjarkbro, 2025-07-10:
I have an additional piece of information for all **Home Assistant** users. 
I have created a separate page just for InkyPi that also fits on the screen. This should be possible using the F12 developer tools and a separate area of 800 x 480 pixels. I also recommend the HACS extension kiosk mode to hide the title bar.

[image]

In order for the InkyPi to be able to log in to Home Assistant in the network, an additional adjustment is required in configuration.yaml.

**This setting in configuration.yaml makes it possible to bypass the login screen without entering a password. This should be used with caution, preferably only in local networks.**

In addition, a separate user is required for InkyPi in Home Assistant (of which we only need the ID).

[image]

192.168.178.135 is the fixed IP of the InkyPi in my network.

```
homeassistant:
  auth_providers:
    - type: trusted_networks
      trusted_networks:
        - 192.168.178.135/32
      trusted_users:
        192.168.178.135:
          - d67511259e8e4671b0ee28b6bf25bfee ## specific user for InkyPi
      allow_bypass_login: true
    - type: homeassistant
  allowlist_external_dirs:
    - /config
```

You can also log in once with this user in Homeassistant to make another important setting - **set the user's default dashboar**d - this should always be the InkyPi page.

acesarewld, 2025-07-28:
> Hi there,
>
> first of all, thank you for your great work on this project!
>
> I've been using the take_screenshot functionality to capture screenshots of web pages rendered by Chromium, specifically for [...]

I've tried exactly this, expecting it to work — however whenever I use those 2 additional lines it just locks up when I try to use the WEB GUI to try and pass the URL and capture a screenshot. Is it because I'm running a Pi 0W that is limited?