fatihak/InkyPi #170: Playlists screen requires extra horizontal scrolling on the iPhone.

rgelb, 2025-06-23:
Picture is better than a thousand words.  Video is worth a million.

https://github.com/user-attachments/assets/2fbceda0-243d-4395-8bc6-d6f047631e82

phattmatt, 2025-06-23:
Looks like some minor div/css change would improve this slightly.

I suspect adding `flex-wrap: wrap;` to .plugin-item:

```CSS
.plugin-item {
    display: flex;
    flex-wrap: wrap;
    align-items: center;
    padding: 10px 25px 10px 40px;
    border-top: 2px solid #e0e0e0;
    gap: 10px;
}
```
Will help a little, but looks untidy without adding a DIV to separate the "Displayed Now" element and adding a DIV to group the Refresh Time and the other buttons together.

phattmatt, 2025-06-23:
I've made some improvements that I think improve the situation:

https://github.com/fatihak/InkyPi/pull/171

On narrow screens the "action" buttons should move together onto a new line to give more space for the plugin instance name.

Names with whitespace are wrapped neatly, but names without will still be slightly problematic:

[image]

fatihak, 2025-06-23:
Merged @phattmatt's PR, let me know if this is resolved