fatihak/InkyPi #176: Newspaper - Landscape Support

cello8619, 2025-06-25:
**Describe the bug**
Landscape doesn't show the newspaper correctly (when Invert Image is set).

**Expected behavior**
I would expect to see the top of the newspaper, like it does if the display inverted setting is not set

**To Reproduce**
Steps to reproduce the behavior:
1. Settings: Orientation Horizontal + Invert Image Enabled
2. Have a newspaper displayed
3. See error.
4. Switch settings back to Invert Image Disabled
5. Newspaper shows correctly

**Logs**
Provide the InkyPi service logs, see [troubleshooting.md](./docs/troubleshooting.md) for instructions on how to view logs.
n 24 20:30:53 InkyPi inkypi[10410]: 20:30:53 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'Sun Sentinel'
Jun 24 20:30:53 InkyPi inkypi[10410]: 20:30:53 - DEBUG - urllib3.connectionpool - Starting new HTTPS connection (1): cdn.freedomforum.org:443
Jun 24 20:30:53 InkyPi inkypi[10410]: 20:30:53 - DEBUG - urllib3.connectionpool - https://cdn.freedomforum.org:443 "GET /dfp/jpg25/lg/FL_SS.jpg HTTP/1.1" 404 None
Jun 24 20:30:53 InkyPi inkypi[10410]: 20:30:53 - ERROR - utils.image_utils - Received non-200 response from https://cdn.freedomforum.org/dfp/jpg25/lg/FL_SS.jpg: status_code: 404
Jun 24 20:30:53 InkyPi inkypi[10410]: 20:30:53 - DEBUG - urllib3.connectionpool - Starting new HTTPS connection (1): cdn.freedomforum.org:443
Jun 24 20:30:53 InkyPi inkypi[10410]: 20:30:53 - DEBUG - urllib3.connectionpool - https://cdn.freedomforum.org:443 "GET /dfp/jpg24/lg/FL_SS.jpg HTTP/1.1" 200 661021
Jun 24 20:30:54 InkyPi inkypi[10410]: 20:30:54 - INFO - root - Found FL_SS front cover for 2025-06-24
Jun 24 20:30:54 InkyPi inkypi[10410]: 20:30:54 - DEBUG - config - Getting key resolution from config.
Jun 24 20:30:55 InkyPi inkypi[10410]: 20:30:55 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Playlist', 'playlist': 'Default', 'plugin_id': 'newspaper', 'plugin_instance': 'Sun Sentinel', 'refresh_time': '2025-06-24T20:30:53.435234-04:00', 'image_hash': 'e40e23d3344670cc00d1152a43944b61ebc319b38844803e54a35a0118754a5d'}
Jun 24 20:30:55 InkyPi inkypi[10410]: 20:30:55 - INFO - display.display_manager - Saving image to /home/admin/InkyPi/src/static/images/current_image.png
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - DEBUG - config - Getting key orientation from config.
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - DEBUG - config - Getting key inverted_image from config.
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - DEBUG - config - Getting key resolution from config.
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - DEBUG - config - Getting key image_settings from config.
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - INFO - display.inky_display - Displaying image to Inky display.
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - DEBUG - config - Getting key None from config.
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:30:56] "GET /settings HTTP/1.1" 200 -
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:30:56] "GET /static/scripts/response_modal.js HTTP/1.1" 304 -
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:30:56] "GET /static/styles/main.css HTTP/1.1" 304 -
Jun 24 20:30:56 InkyPi inkypi[10410]: 20:30:56 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:30:56] "GET /static/icons/settings.png HTTP/1.1" 304 -
Jun 24 20:31:24 InkyPi inkypi[10410]: 20:31:24 - DEBUG - config - Writing device config to /home/admin/InkyPi/src/config/device.json
Jun 24 20:31:24 InkyPi inkypi[10410]: 20:31:24 - DEBUG - config - Getting key scheduler_sleep_time from config.
Jun 24 20:31:24 InkyPi inkypi[10410]: 20:31:24 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:31:24] "POST /display_plugin_instance HTTP/1.1" 200 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - DEBUG - config - Getting key None from config.
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET / HTTP/1.1" 200 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /static/styles/main.css HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /static/icons/settings.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /static/icons/playlist.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/ai_image/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /static/images/current_image.png HTTP/1.1" 200 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/ai_text/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/image_upload/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/apod/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/calendar/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/clock/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/image_url/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/newspaper/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/weather/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/screenshot/icon.png HTTP/1.1" 304 -
Jun 24 20:32:19 InkyPi inkypi[10410]: 20:32:19 - INFO - werkzeug - 192.168.1.101 - - [24/Jun/2025 20:32:19] "GET /images/wpotd/icon.png HTTP/1.1" 304 -
Jun 24 20:32:24 InkyPi inkypi[10410]: 20:32:24 - DEBUG - config - Getting key timezone from config.
Jun 24 20:32:24 InkyPi inkypi[10410]: 20:32:24 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-06-24 20:32:24
Jun 24 20:32:24 InkyPi inkypi[10410]: 20:32:24 - DEBUG - config - Getting key plugin_cycle_interval_seconds from config.
Jun 24 20:32:24 InkyPi inkypi[10410]: 20:32:24 - INFO - refresh_task - Not time to update display. | latest_update: 2025-06-24 20:30:53 | plugin_cycle_interval: 300
Jun 24 20:32:24 InkyPi inkypi[10410]: 20:32:24 - DEBUG - config - Getting key scheduler_sleep_time from config.
Jun 24 20:33:24 InkyPi inkypi[10410]: 20:33:24 - DEBUG - config - Getting key timezone from config.
Jun 24 20:33:24 InkyPi inkypi[10410]: 20:33:24 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-06-24 20:33:24
Jun 24 20:33:24 InkyPi inkypi[10410]: 20:33:24 - DEBUG - config - Getting key plugin_cycle_interval_seconds from config.
Jun 24 20:33:24 InkyPi inkypi[10410]: 20:33:24 - INFO - refresh_task - Not time to update display. | latest_update: 2025-06-24 20:30:53 | plugin_cycle_interval: 300
Jun 24 20:33:24 InkyPi inkypi[10410]: 20:33:24 - DEBUG - config - Getting key scheduler_sleep_time from config.
Jun 24 20:34:24 InkyPi inkypi[10410]: 20:34:24 - DEBUG - config - Getting key timezone from config.
Jun 24 20:34:24 InkyPi inkypi[10410]: 20:34:24 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-06-24 20:34:24
Jun 24 20:34:24 InkyPi inkypi[10410]: 20:34:24 - DEBUG - config - Getting key plugin_cycle_interval_seconds from config.
Jun 24 20:34:24 InkyPi inkypi[10410]: 20:34:24 - INFO - refresh_task - Not time to update display. | latest_update: 2025-06-24 20:30:53 | plugin_cycle_interval: 300
[... 11 lines cut ...]
Jun 24 20:36:24 InkyPi inkypi[10410]: 20:36:24 - DEBUG - PIL.PngImagePlugin - STREAM b'IHDR' 16 13
Jun 24 20:36:24 InkyPi inkypi[10410]: 20:36:24 - DEBUG - PIL.PngImagePlugin - STREAM b'IDAT' 41 65536
[... cut ...]
Jun 24 20:36:25 InkyPi inkypi[10410]: 20:36:25 - INFO - display.inky_display - Displaying image to Inky display.
Jun 24 20:36:53 InkyPi inkypi[10410]: 20:36:53 - DEBUG - config - Writing device config to /home/admin/InkyPi/src/config/device.json
Jun 24 20:36:53 InkyPi inkypi[10410]: 20:36:53 - DEBUG - config - Getting key scheduler_sleep_time from config.

**Screenshots**
Settings Screen:
[image]

Main Page Preview:
[image]

InkyPi Display: 
[image]

**Additional context**
Raspberry Pi Zero 2W
Display: Inky Impression 7.3in 2025 Edition - PIM773

phattmatt, 2025-06-25:
This is happening because the image passed from the Newspaper plugin is taller than the display.

When [display_manager.py](https://github.com/fatihak/InkyPi/blob/3503a5fa19909047fdb8b5d8d9bbbc5c71290d13/src/display/display_manager.py#L65) processes the image it is rotated before being resized to fit the display.

A simple fix may be to resize before rotating.

phattmatt, 2025-06-25:
Just tried changing the order of the resize and rotation and it's looking good so far for the inverted horizontal case.

Change from: 

```python
        # Resize and adjust orientation
        image = change_orientation(image, self.device_config.get_config("orientation"), self.device_config.get_config("inverted_image"))
        image = resize_image(image, self.device_config.get_resolution(), image_settings)
```

To:

```python
        # Resize and adjust orientation
        image = resize_image(image, self.device_config.get_resolution(), image_settings)
        image = change_orientation(image, self.device_config.get_config("orientation"), self.device_config.get_config("inverted_image"))
```

Seems work for multiple plugins when the invert image option is selected.

Just need to check for vertical case.

phattmatt, 2025-06-25:
Look like this change is working:

[src/display/display_manager.py](https://github.com/fatihak/InkyPi/blob/main/src/display/display_manager.py)

From:

```python 
        # Resize and adjust orientation
        image = change_orientation(image, self.device_config.get_config("orientation"), self.device_config.get_config("inverted_image"))
        image = resize_image(image, self.device_config.get_resolution(), image_settings)
```
To:
```python 
        # Resize and adjust orientation
        image = change_orientation(image, self.device_config.get_config("orientation"))
        image = resize_image(image, self.device_config.get_resolution(), image_settings)
        if self.device_config.get_config("inverted_image"): image = image.rotate(180)
```

I'll raise a small PR.

phattmatt, 2025-06-25:
https://github.com/fatihak/InkyPi/pull/177

cello8619, 2025-06-25:
I just tested and looks good to me in all orientations.

fatihak, 2025-06-25:
Merged the change!