fatihak/InkyPi #186: DALL·E 3 plugin fails with HTTP 429 Too Many Requests (but DALL·E 2 works fine)

saulob, 2025-06-30:
When I try to use the DALL·E 3 image generation plugin, it fails with a 429 Too Many Requests error from OpenAI.
However, generating images using DALL·E 2 works perfectly, even when triggered repeatedly.
Also, using ChatGPT 4 (text generation) works fine too.
This only happens with DALL·E 3.

Expected the image to be generated using DALL·E 3 like it does with DALL·E 2 or ChatGPT text prompts, without hitting a rate limit.

Install the latest InkyPi version (downloaded today)

Configure the AI plugin to use DALL·E 3
Trigger an image generation (manual refresh or scheduled)
Error appears in log with HTTP 429

Logs
Relevant portion:
`un 30 16:38:44 quadro inkypi[1841]:     raise HTTPStatusError(message, request=request, response=self)

Jun 30 16:38:44 quadro inkypi[1841]: httpx.HTTPStatusError: Client error '429 Too Many Requests' for url 'https://api.openai.com/v1/images/generations'

Jun 30 16:38:44 quadro inkypi[1841]: For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/429

Jun 30 16:38:44 quadro inkypi[1841]: During handling of the above exception, another exception occurred:

Jun 30 16:38:44 quadro inkypi[1841]: Traceback (most recent call last):

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 1040, in _request

Jun 30 16:38:44 quadro inkypi[1841]:     response.raise_for_status()

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/httpx/_models.py", line 829, in raise_for_status

Jun 30 16:38:44 quadro inkypi[1841]:     raise HTTPStatusError(message, request=request, response=self)

Jun 30 16:38:44 quadro inkypi[1841]: httpx.HTTPStatusError: Client error '429 Too Many Requests' for url 'https://api.openai.com/v1/images/generations'

Jun 30 16:38:44 quadro inkypi[1841]: For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/429

Jun 30 16:38:44 quadro inkypi[1841]: 16:38:44 - DEBUG - openai._base_client - Re-raising status error

Jun 30 16:38:44 quadro inkypi[1841]: 16:38:44 - ERROR - plugins.ai_image.ai_image - Failed to make Open AI request: Error code: 429 - {'error': {'message': None, 'type': 'image_generation_user_error', 'param': None, 'code': None}}

Jun 30 16:38:44 quadro inkypi[1841]: 16:38:44 - ERROR - refresh_task - Exception during refresh

Jun 30 16:38:44 quadro inkypi[1841]: Traceback (most recent call last):

Jun 30 16:38:44 quadro inkypi[1841]:   File "/home/saulob/InkyPi/src/plugins/ai_image/ai_image.py", line 47, in generate_image

Jun 30 16:38:44 quadro inkypi[1841]:     image = AIImage.fetch_image(

Jun 30 16:38:44 quadro inkypi[1841]:             ^^^^^^^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/home/saulob/InkyPi/src/plugins/ai_image/ai_image.py", line 80, in fetch_image

Jun 30 16:38:44 quadro inkypi[1841]:     response = ai_client.images.generate(**args)

Jun 30 16:38:44 quadro inkypi[1841]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/resources/images.py", line 264, in generate

Jun 30 16:38:44 quadro inkypi[1841]:     return self._post(

Jun 30 16:38:44 quadro inkypi[1841]:            ^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 1280, in post

Jun 30 16:38:44 quadro inkypi[1841]:     return cast(ResponseT, self.request(cast_to, opts, stream=stream, stream_cls=stream_cls))

Jun 30 16:38:44 quadro inkypi[1841]:                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 957, in request

Jun 30 16:38:44 quadro inkypi[1841]:     return self._request(

Jun 30 16:38:44 quadro inkypi[1841]:            ^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 1046, in _request

Jun 30 16:38:44 quadro inkypi[1841]:     return self._retry_request(

Jun 30 16:38:44 quadro inkypi[1841]:            ^^^^^^^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 1095, in _retry_request

Jun 30 16:38:44 quadro inkypi[1841]:     return self._request(

Jun 30 16:38:44 quadro inkypi[1841]:            ^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 1046, in _request

Jun 30 16:38:44 quadro inkypi[1841]:     return self._retry_request(

Jun 30 16:38:44 quadro inkypi[1841]:            ^^^^^^^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 1095, in _retry_request

Jun 30 16:38:44 quadro inkypi[1841]:     return self._request(

Jun 30 16:38:44 quadro inkypi[1841]:            ^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/openai/_base_client.py", line 1061, in _request

Jun 30 16:38:44 quadro inkypi[1841]:     raise self._make_status_error_from_response(err.response) from None

Jun 30 16:38:44 quadro inkypi[1841]: openai.RateLimitError: Error code: 429 - {'error': {'message': None, 'type': 'image_generation_user_error', 'param': None, 'code': None}}

Jun 30 16:38:44 quadro inkypi[1841]: During handling of the above exception, another exception occurred:

Jun 30 16:38:44 quadro inkypi[1841]: Traceback (most recent call last):

Jun 30 16:38:44 quadro inkypi[1841]:   File "/home/saulob/InkyPi/src/refresh_task.py", line 114, in _run

Jun 30 16:38:44 quadro inkypi[1841]:     image = refresh_action.execute(plugin, self.device_config, current_dt)

Jun 30 16:38:44 quadro inkypi[1841]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/home/saulob/InkyPi/src/refresh_task.py", line 234, in execute

Jun 30 16:38:44 quadro inkypi[1841]:     return plugin.generate_image(self.plugin_settings, device_config)

Jun 30 16:38:44 quadro inkypi[1841]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

Jun 30 16:38:44 quadro inkypi[1841]:   File "/home/saulob/InkyPi/src/plugins/ai_image/ai_image.py", line 56, in generate_image`

Display: Waveshare epd7in3e
Using latest InkyPi version downloaded on June 30

phattmatt, 2025-07-02:
I have a test script you can use to try and generate an image with the same API key.

1. Log into your InkPi Raspberry Pi over SSH
2. Save the script to your InkyPi directory (maybe '/home/pi/InkyPi')
3. Make sure the '.env' file has a API Key specified
4. Run: `source /usr/local/inkypi/venv_inkypi/bin/activate`

Then run the script:

```bash
(venv_inkypi) pi@inkypi-dev:~/InkyPi $ python check_dalle3_access.py
✅ Image generated.
🖼️ URL: https://oaidalleapiprodscus.blob.core.windows.net/private/org-I7TILSI5ZvCABZvc75eq39MW/user-UBxc4iLk9UhBwSS9X8alkdlB/img-U58XBT1JCavGJqDL5d58qbBo.png?st=2025-06-30T18%3A06%3A20Z&se=2025-06-30T20%3A06%3A20Z&sp=r&sv=2024-08-04&sr=b&rscd=inline&rsct=image/png&skoid=cc612491-d948-4d2e-9821-2683df3719f5&sktid=a48cca56-e6da-484e-a814-9c849652bcb3&skt=2025-06-30T01%3A24%3A26Z&ske=2025-07-01T01%3A24%3A26Z&sks=b&skv=2024-08-04&sig=MLK5sHxyUoS0BiVoo8gaXLkJVoLwfca2GEbwYFkX4vk%3D
✅ Image downloaded as 'dalle3_test.png'
✅ API key is valid and DALL·E 3 image was saved.
```

The script will read the API key from the same `.env` file InkyPi uses, generates an image, then downloads it locally.

[check_dalle3_access.zip](https://github.com/user-attachments/files/21025993/check_dalle3_access.zip)

saulob, 2025-07-03:
Thanks for the test script

I ran the test script and got this error:

`❌ API responded with error: Error code: 429 – {'error': {'message': None, 'type': 'image_generation_user_error', 'param': None, 'code': None}}
❌ API key check failed or image could not be saved.
`

Looks like a 429 rate limit or image generation restriction, but it happens even on the first call. DALL·E 2 works fine with the same key, so I believe DALL·E 3 is being blocked on my account for some reason.

My API key is created with All permissions :(

[image]

phattmatt, 2025-07-12:
Hi @saulob,

Have you had any luck with this?

As you suspect this is related to the ChatGPT API rather than InkyPi could you consider closing this issue?

Thanks.

saulob, 2025-07-12:
Yes, my disappointment was indeed with OpenAI, not with you at all.

Thanks for your help and for the scripts. They were actually very helpful and allowed me to pinpoint the issue.

I really appreciate your time and support ⭐