fatihak/InkyPi #188: What is the device name for the Waveshare Spectra 6 (E6) Full Color 13.3 inch?

evanthomas, 2025-07-02:
I can't find anything in 
```
https://github.com/waveshareteam/e-Paper/tree/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd
```
corresponds to it.

evanthomas, 2025-07-04:
The installation appeared to run correctly:
```
pi@art-display:~/InkyPi $ sudo bash install/install.sh -W epd13in3E
Optional parameter WS is set for Waveshare support.  Screen type is: epd13in3E
Checking if inkypi.service is running
	inkypi.service not running [✔]
Fetching Waveshare driver for: epd13in3E
	Waveshare driver 'epd13in3E.py' already exists at /home/pi/InkyPi/install/../src/display/waveshare_epd/epd13in3E.py [✔]
	Waveshare epdconfig file already exists at /home/pi/InkyPi/install/../src/display/waveshare_epd/epdconfig.py [✔]
Enabling interfaces required for inkypi
	SPI Interface has been enabled. [✔]
	I2C Interface has been enabled. [✔]
Enabling both CS lines for SPI interface in config.txt
Extracting templates from packages: 100%
Installing system dependencies.  [✔] 
Installing inkypi to /usr/local/inkypi
	Creating symlink from /home/pi/InkyPi/install/../src to /usr/local/inkypi/src [✔] 
Creating python virtual environment. 
	Installing python dependencies.  [-] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pillow/
	Installing python dependencies.  [|] WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/openai/
	Installing python dependencies.  [✔] 
Adding additional dependencies for waveshare to the python virtual environment. 
	Installing additional Waveshare python dependencies.  [/]   DEPRECATION: Building 'RPi.GPIO' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'RPi.GPIO'. Discussion can be found at https://github.com/pypa/pip/issues/6334
	Installing additional Waveshare python dependencies.  [✔] 
Adding executable to /usr/local/bin/inkypi
Copying config files to /home/pi/InkyPi/install/../src/config
	Copying device.config to /home/pi/InkyPi/install/../src/config [✔] 
Added display_type: epd13in3E
Installing inkypi systemd service.
Created symlink /etc/systemd/system/multi-user.target.wants/inkypi.service → /etc/systemd/system/inkypi.service.
INKYPI Installation Complete! [✔]
[•] A reboot of your Raspberry Pi is required for the changes to take effect
[•] After your Pi is rebooted, you can access the web UI by going to 'art-display.local' or '192.168.0.171' in your browser.
[•] If you encounter any issues or have suggestions, please submit them here: https://github.com/fatihak/InkyPi/issues
Would you like to restart your Raspberry Pi now? [Y/N] Y
You entered 'Y', rebooting now... [✔]
```

However, the service failed to start:
```
pi@art-display:~ $ sudo systemctl status inkypi.service 
○ inkypi.service - InkyPi App
     Loaded: loaded (/etc/systemd/system/inkypi.service; enabled; preset: enabled)
     Active: inactive (dead) since Fri 2025-07-04 21:08:02 AEST; 2min 45s ago
   Duration: 5.749s
    Process: 534 ExecStart=/usr/local/bin/inkypi -d (code=exited, status=0/SUCCESS)
   Main PID: 534 (code=exited, status=0/SUCCESS)
        CPU: 3.923s

Jul 04 21:08:01 art-display inkypi[559]:   File "/home/pi/InkyPi/src/display/display_manager.py", line 41, in __init__
Jul 04 21:08:01 art-display inkypi[559]:     self.display = WaveshareDisplay(device_config)
Jul 04 21:08:01 art-display inkypi[559]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 04 21:08:01 art-display inkypi[559]:   File "/home/pi/InkyPi/src/display/abstract_display.py", line 19, in __init__
Jul 04 21:08:01 art-display inkypi[559]:     self.initialize_display()
Jul 04 21:08:01 art-display inkypi[559]:   File "/home/pi/InkyPi/src/display/waveshare_display.py", line 58, in initialize_display
Jul 04 21:08:01 art-display inkypi[559]:     raise ValueError(f"Unsupported Waveshare display type: {display_type}")
Jul 04 21:08:01 art-display inkypi[559]: ValueError: Unsupported Waveshare display type: epd13in3E
Jul 04 21:08:02 art-display systemd[1]: inkypi.service: Deactivated successfully.
Jul 04 21:08:02 art-display systemd[1]: inkypi.service: Consumed 3.923s CPU time.

```

evanthomas, 2025-07-04:
Firstly, the drivers in the `E-paper_Seperate_Program` do not seem to be the for new 13.3 Spectra display. I was able to source them using the Waveshare wiki [https://www.waveshare.com/wiki/13.3inch_e-Paper_HAT+_(E)_Manual#Raspberry_Pi](https://www.waveshare.com/wiki/13.3inch_e-Paper_HAT+_(E)_Manual#Raspberry_Pi)

Secondly, it looks like WS may changed the spelling of some of their methods. For example, `init` now seems to be `Init`.

I was able to get the service working by editing `epd13in3E.py` to pickup `epdconfig.py` from the same directory but this obviously not a fix.

evanthomas, 2025-07-11:
The remaining issue is not loading `epdconfig` which is in the `$HOME/InkyPi/src/display/waveshare_epd/` directory. I got it working by editing `epd13in3E.py` but that is not the correct fix. I think `$HOME/InkyPi/src/display/waveshare_epd/` needs to be in `sys.path`.

tkadauke, 2025-07-12:
@evanthomas, did you manage to make this display work? I followed the steps in this thread, but whenever InkyPi starts up, it instantly crashes my Raspberry Pi Zero 2W.

Here are the logs:

```
14:32:01 - INFO - __main__ - Starting web server
14:32:01 - DEBUG - config - Reading device config from /home/tkadauke/InkyPi/src/config/device.json
14:32:01 - DEBUG - config - Loaded config:
{
   "name": "InkyPi",
   "orientation": "horizontal",
   "inverted_image": false,
   "scheduler_sleep_time": 60,
   "startup": true,
   "display_type": "epd13in3E"
}
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/ai_image/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/ai_text/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/apod/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/calendar/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/clock/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/image_upload/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/image_url/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/newspaper/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/screenshot/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/weather/plugin-info.json
14:32:01 - INFO - config - Reading plugin info from /home/tkadauke/InkyPi/src/plugins/wpotd/plugin-info.json
14:32:01 - INFO - display.waveshare_display - Initializing Waveshare display
14:32:01 - INFO - display.waveshare_display - Loading EPD display for epd13in3E display
EPD init...
bcm2835 init success !!! 
e-Paper busy H
e-Paper busy H release
14:32:01 - DEBUG - config - Writing device config to /home/tkadauke/InkyPi/src/config/device.json
```

evanthomas, 2025-07-13:
I sort of have it working by hacking `$HOME/InkyPi/src/display/waveshare_epd/epd13in3E.py` and `$HOME/InkyPi/src/plugins/ai_image/ai_image.py` for issue #200

There doesn't seem to be an error message in your log, have you looked at `dmesg` output? Have tried running it directly from the command line as recommended  in the troubleshooting guide, there may be earlier messages that are not shown in `systemctl` output.

I do have a problem with my pi locking up completely. I have not tested with another pi yet so I don't have evidence that it is a problem with InkyPi. I will raise an issue for that if I can reproduce on different hardware.

tkadauke, 2025-07-13:
The logs I pasted is from running InkyPi straight from the command line. The dmesg output is clean. One thing I didn't try yet is connecting an external display to see if there's any kernel panic. I'll try that tomorrow.

evanthomas, 2025-07-26:
The problems I was having were due to a faulty pi. Once I replaced it, it has been stable.

Dgyepes1, 2025-08-04:
> The remaining issue is not loading `epdconfig` which is in the `$HOME/InkyPi/src/display/waveshare_epd/` directory. I got it working by editing `epd13in3E.py` but that is not the correct fix. I think `$HOME/InkyPi/src/display/waveshare_epd/` needs to be in `sys.path`.

I'm having the same issue with this display. How exactly are you importing `epdconfig` from `epd13in3E.py` ?

evanthomas, 2025-08-05:
I edited the import statement in `eps13in3E.py` to read
```
from . import epdconfig
```

fatihak, 2025-08-12:
@evanthomas You should use the WS provided epdconfig, which is also available in https://github.com/waveshareteam/e-Paper/blob/master/E-paper_Separate_Program/13.3inch_e-Paper_E/RaspberryPi/python/lib/epdconfig.py

I just pushed a change to the same branch so InkyPi can handle both init and Init(). Can you try again?

evanthomas, 2025-08-13:
All good now. Nice job!

fatihak, 2025-08-13:
@evanthomas thanks for testing! Closing this issue.

@giovanniborella Please checkout https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md#error-failed-to-download-waveshare-driver , I added the commands to pull the files for epd13in3E as an example. The rest of the install/setup should work as is without any changes. Feel free to open a new issue if you're still having trouble.

closed 2025-08-13 by fatihak

timarenz, 2025-08-15:
To make this actually work I also needed to download the correct DEV_Config file from same repo, in my case (Pi Zero 2 with RaspberryPi OS Lite 64bit)

```bash
cd InkyPi/src/display/waveshare_epd/
curl -L -O https://raw.githubusercontent.com/waveshareteam/e-Paper/refs/heads/master/E-paper_Separate_Program/13.3inch_e-Paper_E/RaspberryPi/python/lib/DEV_Config_64_b.so
```

Without doing so I got the following error:

```
06:06:28 - INFO - __main__ - Starting web server
06:06:28 - INFO - display.waveshare_display - Initializing Waveshare display
06:06:28 - INFO - display.waveshare_display - Loading EPD display for epd13in3E display
EPD init...
Traceback (most recent call last):
  File "/home/tim/InkyPi/src/display/waveshare_display.py", line 64, in initialize_display
    self.epd_display_init()
  File "/home/tim/InkyPi/src/display/waveshare_epd/epd13in3E.py", line 117, in Init
    epdconfig.module_init()
  File "/home/tim/InkyPi/src/display/waveshare_epd/epdconfig.py", line 91, in module_init
    spi.DEV_ModuleInit()
    ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'DEV_ModuleInit'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tim/InkyPi/src/inkypi.py", line 45, in <module>
    display_manager = DisplayManager(device_config)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tim/InkyPi/src/display/display_manager.py", line 41, in __init__
    self.display = WaveshareDisplay(device_config)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tim/InkyPi/src/display/abstract_display.py", line 19, in __init__
    self.initialize_display()
  File "/home/tim/InkyPi/src/display/waveshare_display.py", line 71, in initialize_display
    raise ValueError(f"Display does not support required methods: {display_type}")
ValueError: Display does not support required methods: epd13in3E
```