fatihak/InkyPi #200: AI Image is rotated 90deg and upscaled too much

evanthomas, 2025-07-07:
**Describe the bug**
Images generated through the AI plugin are:
- rotated, i.e. if horizontal orientation is selected the image is displayed vertically and vice-versa
- the image is upscaled so that only a small portion is displayed

The image is being generated in the correct format.

**Expected behavior**
Image is displayed in the orientation selected in the settings and the whole image is displayed.

**To Reproduce**
Generate an image through the AI image plugin.

**Logs**
```
Jul 07 17:35:44 art-display inkypi[841]: 17:35:44 - INFO - display.waveshare_display - Putting Waveshare display into sleep mode for power saving.
Jul 07 17:35:46 art-display inkypi[841]: 17:35:46 - DEBUG - config - Writing device config to /home/pi/InkyPi/src/config/device.json
Jul 07 17:35:46 art-display inkypi[841]: 17:35:46 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:35:46] "POST /update_now HTTP/1.1" 200 -
Jul 07 17:39:41 art-display inkypi[841]: 17:39:41 - DEBUG - httpcore.connection - close.started
Jul 07 17:39:41 art-display inkypi[841]: 17:39:41 - DEBUG - httpcore.connection - close.complete
Jul 07 17:39:41 art-display inkypi[841]: 17:39:41 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:41] "GET /settings HTTP/1.1" 200 -
Jul 07 17:39:41 art-display inkypi[841]: 17:39:41 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:41] "[36mGET /static/styles/main.css HTTP/1.1[0m" 304 -
Jul 07 17:39:41 art-display inkypi[841]: 17:39:41 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:41] "[36mGET /static/icons/settings.png HTTP/1.1[0m" 304 -
Jul 07 17:39:41 art-display inkypi[841]: 17:39:41 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:41] "[36mGET /static/scripts/response_modal.js HTTP/1.1[0m" 304 -
Jul 07 17:39:56 art-display inkypi[841]: 17:39:56 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:56] "GET /plugin/ai_image HTTP/1.1" 200 -
Jul 07 17:39:56 art-display inkypi[841]: 17:39:56 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:56] "[36mGET /static/styles/main.css HTTP/1.1[0m" 304 -
Jul 07 17:39:56 art-display inkypi[841]: 17:39:56 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:56] "[36mGET /images/ai_image/icon.png HTTP/1.1[0m" 304 -
Jul 07 17:39:56 art-display inkypi[841]: 17:39:56 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:39:56] "[36mGET /static/scripts/response_modal.js HTTP/1.1[0m" 304 -
Jul 07 17:40:15 art-display inkypi[841]: 17:40:15 - INFO - refresh_task - Manual update requested
Jul 07 17:40:15 art-display inkypi[841]: 17:40:15 - INFO - plugins.ai_image.ai_image - Getting random image prompt...
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.connection - close.started
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.connection - close.complete
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - openai._base_client - Request options: {'method': 'post', 'url': '/chat/completions', 'files': None, 'json_data': {'messages': [{'role': 'system', 'content': 'You are a creative assistant specializing in generating highly descriptive and unique prompts for creating images. When given a short or simple image description, your job is to rewrite it into a more detailed, imaginative, and descriptive version that captures the essence of the original while making it unique and vivid. Avoid adding irrelevant details but feel free to include creative and visual enhancements. Avoid common themes. Focus on unexpected, unconventional, and bizarre combinations of art style, medium, subjects, time periods, and moods. Do not provide any headers or repeat the request, just provide your updated prompt in the response. Prompts should be 20 words or less and specify random artist, movie, tv show or time period for the theme.'}, {'role': 'user', 'content': 'Original prompt: "In the style of Jacek Yerka"\nRewrite it to make it more detailed, imaginative, and unique while staying true to the original idea. Include vivid imagery and descriptive details. Avoid changing the subject of the prompt.'}], 'model': 'gpt-4o', 'temperature': 1}}
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - openai._base_client - Sending HTTP Request: POST https://api.openai.com/v1/chat/completions
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.connection - connect_tcp.started host='api.openai.com' port=443 local_address=None timeout=5.0 socket_options=None
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.connection - connect_tcp.complete return_value=<httpcore._backends.sync.SyncStream object at 0x7f930b7f10>
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.connection - start_tls.started ssl_context=<ssl.SSLContext object at 0x7f9356b2f0> server_hostname='api.openai.com' timeout=5.0
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.connection - start_tls.complete return_value=<httpcore._backends.sync.SyncStream object at 0x7f9308b910>
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.http11 - send_request_headers.started request=<Request [b'POST']>
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.http11 - send_request_headers.complete
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.http11 - send_request_body.started request=<Request [b'POST']>
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.http11 - send_request_body.complete
Jul 07 17:40:16 art-display inkypi[841]: 17:40:16 - DEBUG - httpcore.http11 - receive_response_headers.started request=<Request [b'POST']>
Jul 07 17:40:17 art-display inkypi[841]: 17:40:17 - DEBUG - httpcore.http11 - receive_response_headers.complete return_value=(b'HTTP/1.1', 200, b'OK', [(b'Date', b'Mon, 07 Jul 2025 07:40:17 GMT'), (b'Content-Type', b'application/json'), (b'Transfer-Encoding', b'chunked'), (b'Connection', b'keep-alive'), (b'access-control-expose-headers', b'X-Request-ID'), (b'openai-organization', b'user-ikil4br7olepwfvg76lcsijo'), (b'openai-processing-ms', b'653'), (b'openai-version', b'2020-10-01'), (b'x-envoy-upstream-service-time', b'655'), (b'x-ratelimit-limit-requests', b'500'), (b'x-ratelimit-limit-tokens', b'30000'), (b'x-ratelimit-remaining-requests', b'499'), (b'x-ratelimit-remaining-tokens', b'29747'), (b'x-ratelimit-reset-requests', b'120ms'), (b'x-ratelimit-reset-tokens', b'506ms'), (b'x-request-id', b'req_3e3c6751682ccf3149ca0c6573a46582'), (b'strict-transport-security', b'max-age=31536000; includeSubDomains; preload'), (b'cf-cache-status', b'DYNAMIC'), (b'Set-Cookie', b'__cf_bm=dLS0gOytJ5TtanFnnRWvcxAWkw.OIAsm_0X3ccJIxi0-1751874017-1.0.1.1-.q0mwbLbgoJbrW4PCZm80IlACzwLURxsri8v1FRcVT7haN89CizYxVL.bStD_d9WIak_zVQkGsUGdshBDyU3GSdoqJpgNp8w6O6eDItpdzE; path=/; expires=Mon, 07-Jul-25 08:10:17 GMT; domain=.api.openai.com; HttpOnly; Secure; SameSite=None'), (b'X-Content-Type-Options', b'nosniff'), (b'Set-Cookie', b'_cfuvid=9c82trk8A.yihGLnbeKsv7STR2kL6tvzMFAztM59c4Y-1751874017066-0.0.1.1-604800000; path=/; domain=.api.openai.com; HttpOnly; Secure; SameSite=None'), (b'Server', b'cloudflare'), (b'CF-RAY', b'95b5b158eaab77e2-MEL'), (b'Content-Encoding', b'gzip'), (b'alt-svc', b'h3=":443"; ma=86400')])
Jul 07 17:40:17 art-display inkypi[841]: 17:40:17 - INFO - httpx - HTTP Request: POST https://api.openai.com/v1/chat/completions "HTTP/1.1 200 OK"
Jul 07 17:40:17 art-display inkypi[841]: 17:40:17 - DEBUG - httpcore.http11 - receive_response_body.started request=<Request [b'POST']>
Jul 07 17:40:17 art-display inkypi[841]: 17:40:17 - DEBUG - httpcore.http11 - receive_response_body.complete
Jul 07 17:40:17 art-display inkypi[841]: 17:40:17 - DEBUG - httpcore.http11 - response_closed.started
Jul 07 17:40:17 art-display inkypi[841]: 17:40:17 - DEBUG - httpcore.http11 - response_closed.complete
Jul 07 17:40:17 art-display inkypi[841]: 17:40:17 - DEBUG - openai._base_client - HTTP Response: POST https://api.openai.com/v1/chat/completions "200 OK" Headers([('date', 'Mon, 07 Jul 2025 07:40:17 GMT'), ('content-type', 'application/json'), ('transfer-encoding', 'chunked'), ('connection', 'keep-alive'), ('access-control-expose-headers', 'X-Request-ID'), ('openai-organization', 'user-ikil4br7olepwfvg76lcsijo'), ('openai-processing-ms', '653'), ('openai-version', '2020-10-01'), ('x-envoy-upstream-service-time', '655'), ('x-ratelimit-limit-requests', '500'), ('x-ratelimit-limit-tokens', '30000'), ('x-ratelimit-remaining-requests', '499'), ('x-ratelimit-remaining-tokens', '29 [... cut ...]
Jul 07 17:43:41 art-display inkypi[841]: 17:43:41 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:43:41] "[36mGET /static/scripts/response_modal.js HTTP/1.1[0m" 304 -
Jul 07 17:43:41 art-display inkypi[841]: 17:43:41 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:43:41] "[36mGET /static/styles/main.css HTTP/1.1[0m" 304 -
Jul 07 17:43:41 art-display inkypi[841]: 17:43:41 - INFO - werkzeug - 192.168.1.61 - - [07/Jul/2025 17:43:41] "[36mGET /static/icons/settings.png HTTP/1.1[0m" 304 -
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Pi zero 2, Waveshare 13.3 inch (1200x1600)

phattmatt, 2025-07-07:
Hi @evanthomas,

I'm not seeing that behaviour here, but I am running a Pimoroni display not a Waveshare one. I've had a quick look at the code and cannot see anything special about the Waveshare code that would explain this behaviour. The following would help troubleshoot this:

Can you confirm you are running the latest InkyPi code, and have restarted since you last updated?

```bash
pi@inkypi-dev:~/InkyPi $ git show
commit b6aed9714010edf1d3249bcd81975bfc5e42cc97 (HEAD -> main, origin/main, origin/HEAD)
Merge: 6e63acb 5d4ed0e
Author: phattmatt <19799332+phattmatt@users.noreply.github.com>
Date:   Thu Jul 3 19:21:45 2025 +0100

    Merge pull request #189 from Jeppo/main

    Added 7.3" Inky Impression (2025 Edition) 3D Printable bezel for IKEA Rodalm Frame by Maxweeje to community.md
```
If not then follow the [update instructions](https://github.com/fatihak/InkyPi/blob/main/README.md#update).

<details><summary>Are you seeing the current image (display on the InkyPi home page) in the correct orientation?</summary>
<p>
[image]
</p>
</details> 

<details><summary>What does your './src/config/device.json' report for the orientation and resolution settings?</summary>
<p>

```bash
pi@inkypi-dev:~/InkyPi $ head -10 ./src/config/device.json
{
    "name": "InkyPi-Dev",
    "orientation": "horizontal",
    "inverted_image": null,
    "scheduler_sleep_time": 60,
    "startup": false,
    "resolution": [
        800,
        480
    ],
```

</p>
</details> 

What I've seen so far is there are two sections that deal with orientation:

This changes the resolution/dimensions, but not the orientation:
https://github.com/fatihak/InkyPi/blob/b6aed9714010edf1d3249bcd81975bfc5e42cc97/src/plugins/ai_image/ai_image.py#L76-L78

This rotates the image before displaying on the screen:
https://github.com/fatihak/InkyPi/blob/b6aed9714010edf1d3249bcd81975bfc5e42cc97/src/display/display_manager.py#L65-L69

This is the function that rotates the image if configured to do so:

https://github.com/fatihak/InkyPi/blob/b6aed9714010edf1d3249bcd81975bfc5e42cc97/src/utils/image_utils.py#L21-L30

I'm not seeing anything obvious yet.

evanthomas, 2025-07-09:
So, the problem seems to occur because of differences in opinion as to what "vertical" and "horizontal" mean. To me, and seems in the code, "horizontal" means landscape, i.e. the width is longer than the height. For the Waveshare device on the hand, the default "horizontal" means portrait orientation (for whatever reason).  When the AI image is generated the image sized requested is based on the convention that "horizontal" is the same as portrait. I think this is why it works fine for the Pimoroni display but not the WS.

Not sure what the best fix is but I think you need to dynamically determine what the device this is vertical. It might also be worth changing the working in the interface to "portrait" and landscape.

fatihak, 2025-08-13:
@evanthomas Did the changes on the branch also resolve the orientation issue for your display?

evanthomas, 2025-08-13:
It's upside down but I think that's because of the way I've mounted it. The vertical/horizontal orientation is fixed. The WS drivers are found without problem.

fatihak, 2025-08-13:
Nice! Closing this issue

closed 2025-08-13 by fatihak