fatihak/InkyPi #205: Sponsoring

Big-Brummer, 2025-07-10:
I want to sponsor the project owner but I do not want to type in my credit card numbers.
This is a real great project and I want to say thank you.
Is PayPal possible ?

SillyHats, 2025-07-11:
Patreon supports PayPal. Buymeacoffee supports Apple pay and Google pay. There are links for both on the main project page.

Big-Brummer, 2025-07-11:
@SillyHats would you pleas be so kind to add a link where to go.
May be I have potatoes on my eyes but I did not see it at all.

SillyHats, 2025-07-11:
https://github.com/fatihak/InkyPi?tab=readme-ov-file#sponsoring

Big-Brummer, 2025-07-11:
Guten Tag,
sind Sie Herr Adrian Immer ?
Und kennen wir uns ?
Es tut mir leid, ich würde dieses Inky Projekt gerne unterstützen es 
wurde tolle Arbeit geleistet.
Aber der Bezahlvorgang ist mir zu kompliziert. Und ich möchte auf mir 
unbekannten Seiten keine Bankdaten eintragen.

Sie dürfen mir garne eine PayPal Adresse nennen und ich überweise eine 
Anerkennung der Arbeit.

mit freundlichen Grüßen
Jürgen Steger

-------- Forwarded Message --------
Subject: 	Re: [fatihak/InkyPi] Sponsoring (Issue #205)
Date: 	Fri, 11 Jul 2025 06:04:17 -0700

https://github.com/fatihak/InkyPi?tab=readme-ov-file#sponsoring

fatihak, 2025-07-14:
Hi @Big-Brummer, thanks for considering sponsoring this project!

I believe Patreon supports Paypal, which you can find here: https://www.patreon.com/akzdev
I also created a PayPal donation page: https://www.paypal.com/donate/?hosted_button_id=QA23U2W96XWDG

Closing this issue

closed 2025-07-14 by fatihak