fatihak/InkyPi #216: problem with webserver and display

therealcoconut, 2025-07-14:
### I have a problem with the Web Server. 

Even when no display is connected, I can't access the web server. 
I used the troubleshooting information. The service starts normally.
During the installation, everything went smoothly except for one warning: https://github.com/pypa/pip/issues/6334

_I’m trying to use the Zero 2W with a Waveshare EPD7IN3E._

Thank you for the great project and your help!

fatihak, 2025-07-16:
@therealcoconut can you confirm if you have Raspberry OS Lite or Full version?

evanthomas, 2025-07-17:
There are several steps in setting up the WS device and I'm not sure if the InkyPi install does them all. You need to enable SPI and I2C with rasp-config and you need to add entries to `/boot/firmware/config.txt`. For example, I'm almost certain you need to do the last step yourself. Mine hang at the same point until I did that. Check the Waveshare wiki for your device.

h3ih3i, 2025-07-27:
I just encountered a funny issue. If your WS  display is not connected properly with the HAT or your PI(connected upside down), your `inkpi.service` will be stucked at `DEBUG - display.waveshare_epd.epd7in3e - e-Paper busy H`. And you won't be able to see the web setting page, neither your display won't refresh for the first time.

evanthomas, 2025-07-27:
Ok with me. Might be worth adding a note to the troubleshooting guide for WS screens.

knicefire, 2025-07-27:
Had a similar issue myself. 
It took me a while to figure out that I had connected the display ribbon upsidedown 😀😁

therealcoconut, 2025-07-28:
I tried various Pi OSs, but they didn't improve things.
I returned the display and am waiting for a new one to test.

thank you for your quick help. 🙂

closed 2025-07-28 by therealcoconut