fatihak/InkyPi #224: Clock doesn't update time

edrickwong, 2025-07-19:
**Describe the bug**
Clock doesn't update time

**Expected behavior**
Clock updates times

**To Reproduce**
Steps to reproduce the behavior:
* Got to InkyPi portal
* Go to Clock
* Select Clock Face
* Select Update Now

InkyPi updates with the clock face with the current time, but does not update. 

**Logs**
Provide the InkyPi service logs, see [troubleshooting.md](./docs/troubleshooting.md) for instructions on how to view logs.

```
Jul 19 16:43:21 breadpi inkypi[656]: 16:43:21 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Manual Update', 'plugin_id': 'clock', 'refresh_time': '2025-07-19T16:43:21.476556-04:00', 'image_hash': '36941fa4234b326bba18c626fb59e8bbf095c361c51f457a2f39a04928740e15'}
Jul 19 16:43:21 breadpi inkypi[656]: 16:43:21 - INFO - display.display_manager - Saving image to /home/bread/InkyPi/src/static/images/current_image.png
Jul 19 16:43:21 breadpi inkypi[656]: 16:43:21 - INFO - display.inky_display - Displaying image to Inky display.
Jul 19 16:43:49 breadpi inkypi[656]: 16:43:49 - DEBUG - config - Writing device config to /home/bread/InkyPi/src/config/device.json
Jul 19 16:43:49 breadpi inkypi[656]: 16:43:49 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:43:49] "POST /update_now HTTP/1.1" 200 -
Jul 19 16:47:41 breadpi inkypi[656]: 16:47:41 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:41] "GET / HTTP/1.1" 200 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /static/styles/main.css HTTP/1.1[0m" 304 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /static/icons/playlist.png HTTP/1.1[0m" 304 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /images/apod/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /images/calendar/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /images/clock/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /images/image_upload/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /images/image_url/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:42 breadpi inkypi[656]: 16:47:42 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:42] "[36mGET /images/newspaper/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:43 breadpi inkypi[656]: 16:47:43 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:43] "[36mGET /images/screenshot/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:43 breadpi inkypi[656]: 16:47:43 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:43] "[36mGET /images/wpotd/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:43 breadpi inkypi[656]: 16:47:43 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:43] "GET /static/images/current_image.png HTTP/1.1" 200 -
Jul 19 16:47:43 breadpi inkypi[656]: 16:47:43 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:43] "[36mGET /images/ai_text/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:43 breadpi inkypi[656]: 16:47:43 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:43] "[36mGET /static/icons/settings.png HTTP/1.1[0m" 304 -
Jul 19 16:47:43 breadpi inkypi[656]: 16:47:43 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:43] "[36mGET /images/ai_image/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:47:44 breadpi inkypi[656]: 16:47:44 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:47:44] "[36mGET /images/weather/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "GET /plugin/clock HTTP/1.1" 200 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "[36mGET /images/clock/faces/gradient.png HTTP/1.1[0m" 304 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "[36mGET /static/styles/main.css HTTP/1.1[0m" 304 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "[36mGET /images/clock/faces/divided.png HTTP/1.1[0m" 304 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "[36mGET /images/clock/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "[36mGET /images/clock/faces/digital.png HTTP/1.1[0m" 304 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "[36mGET /images/clock/faces/word.png HTTP/1.1[0m" 304 -
Jul 19 16:51:08 breadpi inkypi[656]: 16:51:08 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:51:08] "[36mGET /static/scripts/response_modal.js HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "GET / HTTP/1.1" 200 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /static/icons/settings.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /static/icons/playlist.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /static/styles/main.css HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/clock/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/ai_image/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/ai_text/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/image_upload/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/calendar/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/image_url/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/apod/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/newspaper/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/weather/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/wpotd/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "[36mGET /images/screenshot/icon.png HTTP/1.1[0m" 304 -
Jul 19 16:57:57 breadpi inkypi[656]: 16:57:57 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:57:57] "GET /static/images/current_image.png HTTP/1.1" 200 -
Jul 19 16:58:01 breadpi inkypi[656]: 16:58:01 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:58:01] "GET /settings HTTP/1.1" 200 -
Jul 19 16:58:01 breadpi inkypi[656]: 16:58:01 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:58:01] "[36mGET /static/styles/main.css HTTP/1.1[0m" 304 -
Jul 19 16:58:01 breadpi inkypi[656]: 16:58:01 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:58:01] "[36mGET /static/scripts/response_modal.js HTTP/1.1[0m" 304 -
Jul 19 16:58:01 breadpi inkypi[656]: 16:58:01 - INFO - werkzeug - 192.168.1.182 - - [19/Jul/2025 16:58:01] "[36mGET /static/icons/settings.png HTTP/1.1[0m" 304 -
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Provide the Raspberry Pi Model, Display Model & Size, and any other context.

* Raspberry Pi 2WH with Pimoroni Inky Impression Spectra 7.3"

phattmatt, 2025-07-20:
Hi @edrickwong,

Could you explain a bit more about what behaviour you are seeing and what you expect?

The way the plugins work (including the clock plugin) is that they update the screen once, and once only, when called. 

The display will then be static until another plugin is called.

If you want the time to be displayed, and updated, then you will need to add the clock plugin to a playlist, and set the plugin refresh time, and the playlist cycle time, to be something appropriate.

Hope this helps.

edrickwong, 2025-07-21:
Hi @phattmatt,

Thanks for clarifying the behavior. I didn't realize the plugin page only sets a static version of the clock face and I thought it would more set it and forget it. I also didn't realize the mechanism for updating it was actually using playlists, so thank you for clarifying that.

Is there anyway to make the clock plugin have the time syncing built-in, since I'd imagine that would be a pretty standard use case?

fatihak, 2025-07-23:
As @phattmatt stated you would need to add it into a playlist to have it update on a schedule. Closing this issue.

closed 2025-07-23 by fatihak