fatihak/InkyPi #226: Inky Impression Screen Not Updating

mcbonanno, 2025-07-20:
**Describe the bug**
Originally had issue similar to #159 where pi zero W 2 crashed yesterday morning.  Ever since, the Inky Impression 13.3 display has not been updating.  

**Expected behavior**
When running a refresh, the current_image.png file is updated and displayed on the InkyPi GUI homepage, but the image is never flashed to the display.  

**To Reproduce**
I have updated to the #220 Optimization, to try to address the display not updating and crashing issue, but it has not worked to update the display.  I have tried deleting the playlist and starting fresh with multiple different plugins, but none of them seem to complete the process of actually updating the display.

**Logs**
Jul 20 15:29:17 inkypi inkypi[702]: /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/__init__.py:18: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
Jul 20 15:29:17 inkypi inkypi[702]:   from pkg_resources import declare_namespace
Jul 20 15:29:22 inkypi inkypi[702]: 15:29:22 - INFO - __main__ - Starting web server
Jul 20 15:29:22 inkypi inkypi[702]: 15:29:22 - DEBUG - config - Reading device config from /home/mcb/InkyPi/src/config/device.json
Jul 20 15:29:22 inkypi inkypi[702]: 15:29:22 - DEBUG - config - Loaded config:
Jul 20 15:29:22 inkypi inkypi[702]: {
Jul 20 15:29:22 inkypi inkypi[702]:    "name": "InkyPi",
Jul 20 15:29:22 inkypi inkypi[702]:    "orientation": "horizontal",
Jul 20 15:29:22 inkypi inkypi[702]:    "inverted_image": null,
Jul 20 15:29:22 inkypi inkypi[702]:    "scheduler_sleep_time": 60,
Jul 20 15:29:22 inkypi inkypi[702]:    "startup": false,
Jul 20 15:29:22 inkypi inkypi[702]:    "resolution": [
Jul 20 15:29:22 inkypi inkypi[702]:       1600,
Jul 20 15:29:22 inkypi inkypi[702]:       1200
Jul 20 15:29:22 inkypi inkypi[702]:    ],
Jul 20 15:29:22 inkypi inkypi[702]:    "playlist_config": {
Jul 20 15:29:22 inkypi inkypi[702]:       "playlists": [
Jul 20 15:29:22 inkypi inkypi[702]:          {
Jul 20 15:29:22 inkypi inkypi[702]:             "name": "Weather",
Jul 20 15:29:22 inkypi inkypi[702]:             "start_time": "00:00",
Jul 20 15:29:22 inkypi inkypi[702]:             "end_time": "24:00",
Jul 20 15:29:22 inkypi inkypi[702]:             "plugins": [
Jul 20 15:29:22 inkypi inkypi[702]:                {
Jul 20 15:29:22 inkypi inkypi[702]:                   "plugin_id": "weather",
Jul 20 15:29:22 inkypi inkypi[702]:                   "name": "TO Weather",
Jul 20 15:29:22 inkypi inkypi[702]:                   "plugin_settings": {
Jul 20 15:29:22 inkypi inkypi[702]:                      "latitude": "43.65645893692825",
Jul 20 15:29:22 inkypi inkypi[702]:                      "longitude": "-79.44367128304685",
Jul 20 15:29:22 inkypi inkypi[702]:                      "units": "metric",
Jul 20 15:29:22 inkypi inkypi[702]:                      "displayRefreshTime": "true",
Jul 20 15:29:22 inkypi inkypi[702]:                      "displayMetrics": "true",
Jul 20 15:29:22 inkypi inkypi[702]:                      "displayGraph": "true",
Jul 20 15:29:22 inkypi inkypi[702]:                      "moonPhase": "true",
Jul 20 15:29:22 inkypi inkypi[702]:                      "displayForecast": "true",
Jul 20 15:29:22 inkypi inkypi[702]:                      "forecastDays": "7",
Jul 20 15:29:22 inkypi inkypi[702]:                      "selectedFrame": "None",
Jul 20 15:29:22 inkypi inkypi[702]:                      "topMargin": "",
Jul 20 15:29:22 inkypi inkypi[702]:                      "bottomMargin": "",
Jul 20 15:29:22 inkypi inkypi[702]:                      "leftMargin": "",
Jul 20 15:29:22 inkypi inkypi[702]:                      "rightMargin": "",
Jul 20 15:29:22 inkypi inkypi[702]:                      "backgroundOption": "color",
Jul 20 15:29:22 inkypi inkypi[702]:                      "backgroundColor": "#ffffff",
Jul 20 15:29:22 inkypi inkypi[702]:                      "textColor": "#000000"
Jul 20 15:29:22 inkypi inkypi[702]:                   },
Jul 20 15:29:22 inkypi inkypi[702]:                   "refresh": {
Jul 20 15:29:22 inkypi inkypi[702]:                      "interval": 300
Jul 20 15:29:22 inkypi inkypi[702]:                   },
Jul 20 15:29:22 inkypi inkypi[702]:                   "latest_refresh_time": "2025-07-20T15:21:29.302137-04:00"
Jul 20 15:29:22 inkypi inkypi[702]:                }
Jul 20 15:29:22 inkypi inkypi[702]:             ],
Jul 20 15:29:22 inkypi inkypi[702]:             "current_plugin_index": 0
Jul 20 15:29:22 inkypi inkypi[702]:          }
Jul 20 15:29:22 inkypi inkypi[702]:       ],
Jul 20 15:29:22 inkypi inkypi[702]:       "active_playlist": "Weather"
Jul 20 15:29:22 inkypi inkypi[702]:    },
Jul 20 15:29:22 inkypi inkypi[702]:    "refresh_info": {
Jul 20 15:29:22 inkypi inkypi[702]:       "refresh_time": "2025-07-20T15:21:29.302137-04:00",
Jul 20 15:29:22 inkypi inkypi[702]:       "image_hash": "adcba115aea7e8d2a5425f99e4dc02445c3e553ab6b547d8bc9d3794cb2e13d8",
Jul 20 15:29:22 inkypi inkypi[702]:       "refresh_type": "Playlist",
Jul 20 15:29:22 inkypi inkypi[702]:       "plugin_id": "weather",
[... 84 lines cut ...]
Jul 20 15:47:30 inkypi inkypi[712]: 15:47:30 - INFO - waitress - Serving on http://0.0.0.0:80
Jul 20 15:57:32 inkypi inkypi[712]: 15:57:32 - INFO - refresh_task - System Stats: {'cpu_percent': 0.0, 'memory_percent': 59.5, 'disk_percent': 20.9, 'load_avg_1_5_15': (0.0, 0.10498046875, 0.13525390625), 'swap_percent': 26.5, 'net_io': {'bytes_sent': 74942, 'bytes_recv': 206398}}
Jul 20 15:57:32 inkypi inkypi[712]: 15:57:32 - INFO - refresh_task - Running interval refresh check. | current_time: 2025-07-20 15:57:31
Jul 20 15:57:32 inkypi inkypi[712]: 15:57:32 - INFO - refresh_task - Determined next plugin. | active_playlist: Weather | plugin_instance: TO Weather
Jul 20 15:57:32 inkypi inkypi[712]: 15:57:32 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'TO Weather'
Jul 20 15:58:25 inkypi inkypi[712]: 15:58:25 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Playlist', 'playlist': 'Weather', 'plugin_id': 'weather', 'plugin_instance': 'TO Weather', 'refresh_time': '2025-07-20T15:57:31.029780-04:00', 'image_hash': 'f499bcb5c9c8913867096355c2e52b6e98b508edc3ee8b71d101bc60cb4642ef'}
Jul 20 15:58:25 inkypi inkypi[712]: 15:58:25 - INFO - display.display_manager - Saving image to /home/mcb/InkyPi/src/static/images/current_image.png
Jul 20 15:58:26 inkypi inkypi[712]: 15:58:26 - INFO - display.inky_display - Displaying image to Inky display.
Jul 20 15:59:13 inkypi inkypi[712]: 15:59:13 - INFO - refresh_task - Manual update requested
Jul 20 15:59:13 inkypi inkypi[712]: 15:59:13 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'TO Weather'
Jul 20 16:00:00 inkypi inkypi[712]: 16:00:00 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Playlist', 'playlist': 'Weather', 'plugin_id': 'weather', 'plugin_instance': 'TO Weather', 'refresh_time': '2025-07-20T15:59:13.665863-04:00', 'image_hash': 'b5839ca58fd4c3d848580d9049cc8cdce8594307e4f6a452735d1232e39055c8'}
Jul 20 16:00:00 inkypi inkypi[712]: 16:00:00 - INFO - display.display_manager - Saving image to /home/mcb/InkyPi/src/static/images/current_image.png
Jul 20 16:00:02 inkypi inkypi[712]: 16:00:02 - INFO - display.inky_display - Displaying image to Inky display.
Jul 20 16:04:16 inkypi inkypi[712]: 16:04:16 - INFO - refresh_task - Manual update requested
Jul 20 16:04:16 inkypi inkypi[712]: 16:04:16 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'Wiki pic'
Jul 20 16:04:16 inkypi inkypi[712]: 16:04:16 - INFO - plugins.wpotd.wpotd - WPOTD plugin settings: {'customDate': '', 'shrinkToFitWpotd': 'true'}
Jul 20 16:04:16 inkypi inkypi[712]: 16:04:16 - INFO - plugins.wpotd.wpotd - WPOTD plugin datetofetch: 2025-07-20
Jul 20 16:04:17 inkypi inkypi[712]: 16:04:17 - INFO - plugins.wpotd.wpotd - WPOTD plugin Picture URL: https://upload.wikimedia.org/wikipedia/commons/3/32/C2022_E3_%28ZTF%29-_Alessandro_Bianconi.jpg
Jul 20 16:04:19 inkypi inkypi[712]: 16:04:19 - INFO - plugins.wpotd.wpotd - Image resized to fit device dimensions: 1600,1200
Jul 20 16:04:23 inkypi inkypi[712]: 16:04:23 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Playlist', 'playlist': 'Test', 'plugin_id': 'wpotd', 'plugin_instance': 'Wiki pic', 'refresh_time': '2025-07-20T16:04:16.344760-04:00', 'image_hash': 'da3f1f6f6c3bc0c4a0fde9d379328f1d3924eb57846ecaa4075a6c893b5136a0'}
Jul 20 16:04:23 inkypi inkypi[712]: 16:04:23 - INFO - display.display_manager - Saving image to /home/mcb/InkyPi/src/static/images/current_image.png
Jul 20 16:04:27 inkypi inkypi[712]: 16:04:27 - INFO - display.inky_display - Displaying image to Inky display.

**Additional context**
Raspberry Pi Zero W 2 with Inky Impression 13.3 display.  Using official pi power supply

fatihak, 2025-07-21:
This happened on my 13.3 Inky Impression once, not sure why it happens. Can you disconnect and reconnect the pi from the display (after shutting down the pi and unplugging the power cable), and retry?

mcbonanno, 2025-07-21:
That worked! I had tried many resets, but disconnecting the pins did the trick.  Thanks for all your work on a great project!

fatihak, 2025-07-21:
Nice, glad you're enjoying the project! Closing this issue