fatihak/InkyPi #232: Image preview

dzhang-heihei, 2025-07-24:
Hello,
I would llike to know if there is a way to preview the image/screenshot that I want to send to my display before I click on the button "Update Now".

Thank you for this great projet!

phattmatt, 2025-07-24:
Hi @h3ih3i,

There is currently no way to preview the image before it is sent to the display.

h3ih3i, 2025-07-29:
I modified in the code source to have a more custom style and I wish to view the result quickly.

fatihak, 2025-08-06:
Yes, you can call the generate_image function but you'll need to provide the expected plugin and device settings as arguments. You can find details on this in the building_plugins documentation: https://github.com/fatihak/InkyPi/blob/main/docs/building_plugins.md

Closing this as not planned since the 'Update Now' functionality is generally used to preview plugins before adding it into playlists/scheduling automatic refreshes, and I don't see the need for additional preview functionality.

closed 2025-08-07 by fatihak